grbl/grbl #566: steps not proportional

OldETC, 2015-01-12:
Grbl 0.9 Arduino R3.  Steps are not proportional... i.e.  running 1/2 stepping algorithm in the stepper driver.  
Code I am running is G91 x100 with the 100 varying upto 1000 to see if I can see any change.

GRBL Settings:
$0=100 (step pulse, usec)
$1=100 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=7 (dir port invert mask:00000111)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.020 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$14=1 (auto start, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=100.000 (homing feed, mm/min)
$25=150.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=5.000 (homing pull-off, mm)
$100=157.840 (x, step/mm)
$101=157.840 (y, step/mm)
$102=157.840 (z, step/mm)
$110=152.400 (x max rate, mm/min)
$111=152.400 (y max rate, mm/min)
$112=152.400 (z max rate, mm/min)
$120=100.000 (x accel, mm/sec^2)
$121=100.000 (y accel, mm/sec^2)
$122=100.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

On the scope, the positive side of the steps is always 5.45ms and the negative side is always 7.25ms.  Running open loop, no load on motors, starting from position with both coils energized 1 and 3 positive.  Using SEED motor driver feeding the steppers from Arduino UNO R3s as drivers.  
Code for the stepper driver:
//  Demo function:The application method to drive the stepper motor.

//  Hareware:Stepper motor - 24BYJ48,Seeed's Motor Shield v2.0

//  Author:Frankie.Chu

//  Date:20 November, 2012
//  Modified by HLHowell for direct control of GRBL using pins 2,3

#define MOTOR_CLOCKWISE      1

#define MOTOR_COUNTER_CLOCKWISE  -1

/******Pins definitions*************/

#define MOTORSHIELD_IN1	8
#define MOTORSHIELD_IN2	11
#define MOTORSHIELD_IN3	12
#define MOTORSHIELD_IN4	13

#define CTRLPIN_A		9
#define CTRLPIN_B		10

#define STEP_IN                 2
#define DIR_IN                  3
#define Ampmeter                A0 // reads the top of Ir
#define Imax                    2 // 2A maximum
#define Ir                      0.62 // the current monitor resistance
const unsigned char stepper_ctrl[]={0x27,0x17,0x1e,0x2e};

int stepperMaxSpeed=300; /* max RPM */
int StepperDirection=1;  /* 1=CLOCKWISE, -1=CounterClockwise */
int number_of_steps = 200; /* steps per rotation */
int Idrive=255;
int steptblct; /* set by the size of the steptable. */
unsigned int micros_delay=100; /* the time to keep step high */
int step_number;
/************************************************************/
/*Function: Get the stepper motor rotate                    */
/*Parameter:int steps,the total steps and the direction the */
/*motor rotates, when viewed perpendicular to the drive end */
/* of the shaft                                             */
/*			if DIR > 0,rotates clockwise, 	    */
/*			if DIR < 0,rotates anticlockwise.   */
/************************************************************/
void step(void)
{
/* when here, a step input has been received.  Step the 
   motor one step in the designated direction. */
   step_number=step_number+StepperDirection;
   if (step_number<0) step_number=4;
   if (step_number>4) step_number=0;
  // Serial.println(step_number);
   PORTB = stepper_ctrl[step_number];
}

void initialize()

{
  //interrupts();
  //Serial.begin(9600);
  //delay(20000);
  
  noInterrupts();
  
  pinMode(MOTORSHIELD_IN1,OUTPUT); /* sets the driver bits direction */
  pinMode(MOTORSHIELD_IN2,OUTPUT);
  pinMode(MOTORSHIELD_IN3,OUTPUT);
  pinMode(MOTORSHIELD_IN4,OUTPUT);

  pinMode(CTRLPIN_A,OUTPUT);       /* sets the control bit direction */
  pinMode(CTRLPIN_B,OUTPUT);
  digitalWrite(CTRLPIN_A,HIGH);
  digitalWrite(CTRLPIN_B,HIGH);
  
  pinMode(STEP_IN,INPUT); /* GRBL input control pins */
  pinMode(DIR_IN,INPUT);

  stop();                  /* stop the motor by disabling the driver */

  StepperDirection = MOTOR_CLOCKWISE;
  steptblct=sizeof(stepper_ctrl);
}

/*******************************************/

void stop()
{
  /*connect all motor pins to ground , to stop the motor. */

  PORTB=PORTB&0x06; /* force all inputs to ground */
  
}

void setup()
{
  initialize();//Initialization for the stepper motor.
  StepperDirection=1;
}

void loop()
{

// debug check
  digitalWrite(CTRLPIN_A,HIGH);
  if (digitalRead(STEP_IN))
  {
    while (digitalRead(STEP_IN)); // wait for step to release
    if (digitalRead(DIR_IN))
    {
      StepperDirection=MOTOR_CLOCKWISE;
    }
    else
    {
          StepperDirection=MOTOR_COUNTER_CLOCKWISE;
    }
    step();//Stepper motor moves 1 step
  }
}

Driving in clockwise direction.

Motors capable of 394 steps per second max.  These settings get 324.64, so should work OK.  Tried faster and slower, but the relation ship of the pulses doesn't change, although the frequency does.

If I put this on the machine this way, it will give resonances, and additional drive noise into the mechanics.

Checked and only one artifact I cannot explain, PIN 9 of the stepper driver UNO shows a PWM negative pulse 10us wide.  I have been chasing this for a couple of weeks.  Can't find the source.
No scope camera, but I could try to get an image with the cellphone if you want it.

I'm going through the GRBL code now to see if I can isolate the issue.

chamnit, 2015-01-12:
@OldETC : I'm having a hard time following what the problem is. Step size over time itself should remain fixed, which is set by the step pulse setting, and pulses should increase in frequency when increasing in motor speed. 

The code in the latter part of your post doesn't exist in the Grbl source code and I'm unfamiliar with what its used for. Guessing, it looks like its for the Seeed Motor Shield v2.0. Grbl does not support this board.

OldETC, 2015-01-13:
Found that bug.  It was my stepper driver going one index too far.  I'll take another look at the SEED software.  Thanks.
Sorry for the interruption.

closed 2015-01-13 by OldETC