grbl/grbl #567: Soft limit hit with weird behavior on 0.9g

faultylee, 2015-01-12:
I've just added limit switches and started using Hard/Soft limit + Homing on my Shapeoko2. It works great after sorting out some issues with my setup. It works fine for the first small job after that, but when I have a bigger job that I didn't realized that it hit the soft limit, it doesn't show the usual alarm message, UGS just shows `Queue` on the status. Not realizing what happened, I click `Pause` then `Resume`, grbl actually continue on, until the code that actually exceed the limit was executed, the alarm state was entered, still no alarm message. I only realized the alarm state when no command works, and I keep seeing the `>>>`. I'm running Grbl 0.9g on Uno, stock firmware downloaded from here for Shapeoko. I use UGS 1.0.8-nightly to stream. 
My config: (not the latest version of x and y max travel, but close)
```
$0=10 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=7 (dir port invert mask:00000111)
$4=0 (step enable invert, bool)
$5=1 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.050 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$14=1 (auto start, bool)
$20=1 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=3 (homing dir invert mask:00000011)
$24=25.000 (homing feed, mm/min)
$25=1000.000 (homing seek, mm/min)
$26=25 (homing debounce, msec)
$27=2.000 (homing pull-off, mm)
$100=40.000 (x, step/mm)
$101=40.000 (y, step/mm)
$102=320.000 (z, step/mm)
$110=5000.000 (x max rate, mm/min)
$111=5000.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=250.000 (x accel, mm/sec^2)
$121=250.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=320.000 (x max travel, mm)
$131=320.000 (y max travel, mm)
$132=60.000 (z max travel, mm)
```

The gcode: (I only include up to the point that causes the problem)
```
(Created with grbl post processor (based upon emc2b) 2015/01/11 18:45)
M3
G17 G90 G21
G00 Z2 
(Profile 1)
G90
(Sketch 3)
X19 Y29.749 S20000 M03
Z1 
G01 Z-0.966 F149 
G02 X19 Y30.749 I0 J0.5 F400 
G01 X32.102 Y30.749 Z-0.966  
G03 X33.089 Y31.12 Z-0.966 I0.003 J1.488  
G02 X34.212 Y30.265 Z-0.966 I0.559 J-0.429  
G03 X33.716 Y29.135 Z-0.966 I1.091 J-1.152  
G01 X33.716 Y23.809 Z-0.966  
G03 X34.087 Y22.821 Z-0.966 I1.488 J-0.004  
G02 X34.295 Y21.969 Z-0.966 I-0.464 J-0.564  
X34.209 Y21.903 Z-0.966 I-0.094 J0.034  
G01 X33.972 Y21.666 Z-0.966  
X33.948 Y21.627 Z-0.966  
G02 X33.232 Y21.698 Z-0.966 I-0.302 J0.598  
G03 X32.102 Y22.195 Z-0.966 I-1.155 J-1.095  
G01 X21.466 Y22.195 Z-0.966  
G03 X19.966 Y20.695 Z-0.966 I0 J-1.5  
G01 X19.966 Y13.543 Z-0.966  
G03 X24.218 Y9.3 Z-0.966 I4.242 J0  
G01 X33.645 Y9.3 Z-0.966  
X33.645 Y2.748 Z-0.966  
G03 X37.897 Y-1.503 Z-0.966 I4.244 J-0.006  
G01 X73.695 Y-1.503 Z-0.966  
G03 X77.946 Y2.748 Z-0.966 I0.006 J4.244  
G01 X77.946 Y13.197 Z-0.966  
G03 X77.964 Y13.543 Z-0.966 I-3.706 J0.366  
G01 X77.964 Y30.749 Z-0.966  
X169.606 Y30.749 Z-0.966  
G03 X170.593 Y31.12 Z-0.966 I0.003 J1.488  
G02 X171.707 Y30.265 Z-0.966 I0.555 J-0.429  
G03 X171.211 Y29.135 Z-0.966 I1.091 J-1.152  
G01 X171.211 Y23.809 Z-0.966  
G03 X171.582 Y22.821 Z-0.966 I1.488 J-0.004  
G02 X171.835 Y22.122 Z-0.966 I-0.466 J-0.564  
X171.722 Y21.903 Z-0.966 I-0.25 J-0.009  
G01 X171.485 Y21.666 Z-0.966  
G02 X170.736 Y21.698 Z-0.966 I-0.353 J0.507  
G03 X169.606 Y22.195 Z-0.966 I-1.155 J-1.095  
G01 X158.961 Y22.195 Z-0.966  
G03 X157.461 Y20.695 Z-0.966 I0 J-1.5  
G01 X157.461 Y13.543 Z-0.966  
G03 X161.713 Y9.3 Z-0.966 I4.258 J0.015  
G01 X171.158 Y9.3 Z-0.966  
X171.158 Y2.765 Z-0.966  
G03 X175.409 Y-1.486 Z-0.966 I4.242 J-0.008  
G01 X211.208 Y-1.486 Z-0.966  
G03 X215.459 Y2.765 Z-0.966 I0.006 J4.244  
G01 X215.459 Y13.342 Z-0.966  
G03 X215.468 Y13.543 Z-0.966 I-4.269 J0.291  
G01 X215.468 Y30.749 Z-0.966  
X247.5 Y30.749 Z-0.966  
G03 X249 Y32.249 Z-0.966 I0 J1.499  
G01 X249 Y274.245 Z-0.966  
G03 X247.5 Y275.745 Z-0.966 I-1.501 J-0.001  
G01 X215.468 Y275.745 Z-0.966  					  <- grbl stops around here before this was executed
X215.468 Y292.899 Z-0.966  
G03 X215.459 Y293.087 Z-0.966 I-8.085 J-0.292     
G01 X215.459 Y303.72 Z-0.966  
G03 X211.208 Y307.972 Z-0.966 I-4.246 J0.005  
G01 X175.409 Y307.972 Z-0.966                     <- Soft limit hit
```
My workspace origin for Y is about -305. The 2 places where Grbl stop were guessed, based on the message and response shown in UGS and the current tool position, both after G03.

I've tried entering command manually and if the command exceed the soft limit, a proper alarm message is displayed, so I'm assuming the above is not as expected.

doppelhub, 2015-01-20:
Are you using GrblController?  If so, that program doesn't gracefully return errors supplied by Grbl... it just hangs.  If this is the case, you should submit this feature request to the GrblController team (not Grbl).

faultylee, 2015-01-21:
I'm using UGS 1.0.8-nightly (Universal Gcode Sender), if I type in a few command manually, it will trigger the alarm message, but not in mid job.

chamnit, 2015-02-06:
@faultylee : This seems like a problem with UGS. They haven't integrated all of the new capabilities of v0.9, as far I understand. I suppose it's possible that UGS does some type of pre-filtering with soft limits, which would make it stop the first time, but I can't say for sure. Try using GrblPanel, which is more feature complete, or our officially supported Python streaming script in our repo. The latter will tell you the raw feedback Grbl is sending.

faultylee, 2015-02-07:
@chamnit Thanks for the suggestion. I'm running UGS on linux and no spare Windows machine, so I can't try out GrblPanel for now. It's still weird that if UGS is doing pre-filtering, why won't it happen to manually input command. Anyway, I'll try out the Python streaming script.