grbl/grbl #568: Homing failed when X and Y homing cycles are combined ?

jules, 2015-01-13:
I had a hard time understanding why my homing cycle was failing. 
I'm using open switches at both end of X and Y, and one on Z+.
I manually test them and they work perfectly.

if I keep the default (in config.h) : 

#define HOMING_CYCLE_0 (1<<Z_AXIS)
#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))

Z homing works (but not always), X and Y always failed.

But when I update the config.h with separate cycles for each axis everything works perfectly !

#define HOMING_CYCLE_0 (1<<Z_AXIS)
#define HOMING_CYCLE_1 (1<<Y_AXIS) 
#define HOMING_CYCLE_2 (1<<X_AXIS)

doppelhub, 2015-01-20:
Define 'failing'.  Did the machine drive through the limit switch and crash?  I'm thinking this is a hardware noise issue.  If you have an oscilloscope, I'd inspect the limit switches at the 328p (pins 9,10 and either 11 or 12).  I suspect you've got noise in your system from running the XY steppers at the same time.  Do you find that with hard limits enabled your code sometimes stops due to a nonexistent 'hard limit' error?

jules, 2015-01-20:
Here is what is happening : 
-1 : The Y limit switch gets hit first, the Y axis stop moving (no search for a 'home' position, I mean going back and forth etc)
-2 : Then the X axis starts moving, find the limit switch then starts the 'homing' point search sequence: go in the negative direction, then the positive direction until it re-hit the switch and does that in an endless loop.

jules, 2015-01-20:
Do you find that with hard limits enabled your code sometimes stops due to a nonexistent 'hard limit' error? : I can really tell as I haven't use the machine in production yet. But I run some simple drawing examples and everything is fine.

chamnit, 2015-01-20:
@jules : I have a feeling that your wiring is incorrect. The behavior you describe makes no sense otherwise. Please draw out and post the wiring of your limit switches.

closed 2015-02-02 by chamnit