grbl/grbl #569: Pull-up on Limit X isn't working - Atmega 328p

metropt, 2015-01-13:
Hi,
I have a custom board and all is working except the pull up on the limit pin x. I checked that there is no short between it and gnd and there is continuity between the uC pin and the header where I'm connecting the switch. What could cause it?

Thank you

doppelhub, 2015-01-20:
If the issue were the Z limit switch, I'd guess you were running the incorrect (PWM vs. non-PWM) grbl build.  However, the X limit switch pin has not changed, so if you've reproduced the issue on several custom boards, then the problem is almost certainly your custom PCB.  A simple check to verify this isn't a Grbl issue (99% sure it isn't) is to program the 328p to pull pin 9 high and see what happens... if there's a short to ground, you've likely damaged the 328p's drive circuitry, so you should probably start with a fresh custom PCB.  If you feel this is a SW issue, please let us know which version of Grbl you're using, but I can't see any way this is a SW issue.  Another troubleshooting step is to attempt to drive a 20mA current-limited 5V test signal into pin 9 on an unpowered PCB... if you hit the current limit (20mA, 0V), then your PCB has a short.  If you don't hit the current limit (0mA, 5V), then the issue is SW.  You can make a 20mA current limit test source by connecting a 250 Ohm resistor in series with a 5V power supply, then connecting the other end of that resistor to pin 9.  In all cases above, 'pin 9' is the X limit pin, assuming your custom PCB isn't using an Arduino-compatible footprint.

metropt, 2015-01-21:
I'll try to drive the pin but I could damaged the pin during setup the limits.

closed 2015-01-21 by metropt