grbl/grbl #570: Power supply control pin

cjd000, 2015-01-14:
Hi,
This is an addition request - not an issue, but let me start by saying what a fantastic job you guys are doing at making this kind of tech available to joe public. If it wasn't for you guys doing this, the general hobbyist like me wouldn't stand a chance!

Currently i am using a set of TB6560 single axis stepper controllers linked to a Atmel 328P board being powered by a computer ATX style Power supply.

The problem i have is that, as some of you may know, the TB6560 drivers require a particular power on sequence to ensure that the 5V supply is on before the motor power is applied and that the reverse is followed for switch off, otherwise these controllers can blow (from experience!). I've accomplished this by using the Coolant Mist pin as a PSU (power supply) control signal and hook up the 5v from the PSU USB "always on" output.

The advantage is that i can shut down the power at the end of a machining cycle (lights, vacuum table, spindle, fans, motors etc) all with one command - Coolant off.

What i would like to see is this feature embedded so that 'feed hold' or soft resets etc control the power rails correctly. Also at initial power on i need to ensure that the PSU control pin is set correctly. This pin on all ATX style PSU's is active low so i also need to figure out how to invert the pin. Currently i'm using a transistor circuit to invert it (its probably very easy through the code but i'm struggling to find it)

Is this something you would be interested in adding as a feature? I'm sure there are others using these PSU's for their CNC systems as they provide high current and a stable 12v & 5v output.

Thanks
Chris

doppelhub, 2015-01-20:
When you issue a reset, the Arduino pins are unprogrammed, and will float until grbl reloads.  Unfortunately, there's nothing that can be done in software.  There same condition occurs whenever the 328p is unplugged, or plugged in and has not yet loaded grbl?  Or when your computer goes to sleep.

If I understand your requirements correctly, you need a hardware solution, such as the transistor circuit you've already mentioned.  In all above cases, the pins are floating by default, so you'll need a pull down (if your transistor circuitry doesn't already handle this).  Another solution is to gate the external power supply with a 5 V saturation N channel FET, instead of the logic input.  With the proper (e.g. 2 kOhm) pull down resistor, you can guarantee the power supply never hits the TB6560 until the 5V rail is up and stable (e.g. grbl has loaded and pulled the FET gate high).  

Please let me know if I've misunderstood your deliverables.  Ultimately, this is an Arduino HW limitation... no discrete pulldowns onboard.

cjd000, 2015-01-20:
Thanks for the response. 

I have the hardware side fixed using the transistor circuit so the PSU is only enabled when arduino outputs the high state for mist control.

What I need the help with is the software side as my solution is a bit of a temporary bodge!
What I'd like to get to is utilising the last unused pin on the 328 board for this and then being able to use a command to turn the power on and off without some of the built in limitations of using the mist control pin.

Eg, automatically turn the power on a second or two before starting the homing procedure, adding a power on/off button to the grbl interface software and having the power auto off after the gcode is finished etc.

All this functionality can be added by the interface designers once the firmware has that extra output pin setup. Kind of like a 'universal' control pit that can be enabled from the firmware configure.h and pins files with invertable output and configurable command to turn on/off the pin

Any help you could give on this would be great as I'm sure others would appreciate the flexibility of a controllable extra output pin, even if it's just to enable the stepper controller fans!

Regards,
Chris

blinkenlight, 2015-01-25:
@cjd000 : On the practical side of things, most cheap TB6560 driver boards implement no power sequencing whatsoever and keep working fine nevertheless, but I do appreciate you're trying to do The Right Thing as mandated by the datasheet. However, proper power sequencing at turn-on should be your power supply's job, not GRBL's. Another thing is that power should be turned on exactly once when you sit down at your mill, and left on until you're done - considering everyone in their right mind uses some amount of micro-stepping these days, and that micro-steps are not a natural state of the motors, your inter-step position going "poof" the second you stop applying power, so don't do that. If you want to do it properly, build a short delay circuit onto the power rail that should come up first and just use that to switch on the rest...

langwadt, 2015-01-29:
The single channel drivers I've seen use a linear regulator to make 5V from the motor supply, that pretty much guarantees that the 5V  will be up at the same time as the motor voltage

AnHardt, 2015-01-29:
How about feeding the 5V part of the stepper drivers with the 5V standby
output of the ATX powersupply?

cjd000, 2015-01-29:
Thanks Andreas that's exactly what I'm doing!

The tb6560 boards have a 5v regulator running from the motor supply line so the motor power is present before the 5v has chance to energise. For this driver it results in a lot of smoke! I talk from experience with two being killed befor I hacked the boards to run from the atx 5v supply. By using the usb always on supply and switching the 12v on and off from the arduino I always ensure the control lines are active first. 

The setup I described at the start of this thread works great. I control the board through the mist coolant command. I think some of the responses here are getting off topic as I have no issue with the electronics!

I'd just like to have one output from the ardunio that's usable for whatever purpose rather that using the mist coolant function. This way I can add a custom button to the interface software and merrily control the PSU from the computer interface or even write it into the gcode so I can shut down the power to everything when the cnc job is complete. 

Chris

timryder, 2015-02-04:
If you're writing your own GUI for this machine/software.... just don't call it "Coolant Mist On", call it "Enable Motors" or whatever.  It technically is doing it and you'll be the only one that knows it's actually just coolant mist. You can get crazy and modify the firmware to ignore the M8 M9 gCodes and you can write your own M code which execute the same functions GRBL already has written.  

I just don't see the advantage here for all that work and limited pins on the 328p. 

Think about it from a programmers side of things, what would it be called? what M/G code would execute it. Does it need to happen real time on an interrupt or added to the planner to make sure it empties the motion buffer before executing?  Way too many possibilities to consider to just make this change. In my humble opinion.