grbl/grbl #571: Define accel on edge branch

metropt, 2015-01-15:
How will be accel defined on the new version?

Thank you

metropt, 2015-01-15:
It doesn't show on the options but it is possible to change.

chamnit, 2015-01-15:
@metropt : Not sure what you mean by it doesn't show on options. It prints out fine when issuing a '$$' command.

metropt, 2015-01-15:
that strange, on my system it jumps from $112 to $130 :s

chamnit, 2015-01-15:
@metropt : That is strange. Try wiping your EEPROM with the Arduino IDE clear eeprom library. THere's instructions on the wiki on how to modify it to do it. (Note to self, add this to the Grbl source).

chamnit, 2015-01-15:
@metropt : If that doesn't work, try reflashing the optiboot boot loader onto your board. I had a weird problem once, where nothing made sense. When I did reflashed boot loader, it worked fine. It happened when the firmware was getting close to about 30k, which is about the size of it with the CoreXY additions.

closed 2015-01-17 by chamnit

metropt, 2015-01-17:
clear the eeprom doesn't seem to work. BTW I didn't flash the bootloader. I directly load the grbl.hex file using the avr programmer.