grbl/grbl #573: VARIABLE SPINDLE stop working on 0.9h

GabeHC, 2015-01-16:
update from 0.9g variable spindle function stop working, D11 stay 0V all time.
I am using UNO + CNC Shield, Is there something I miss?

EliteEng, 2015-01-17:
@chamnit @GabeHC I can confirm it gets stuck at line 116 of spindle_control.c and does not enter spindle_set_state to set either the Spindle enable or direction pins.

I will have a look into it ( I need to get my head around the new states anyway )

EliteEng, 2015-01-17:
@chamnit I believe it is an error with the logic.
Am I right that line 116 of spindle_control.c
```if (sys.state != STATE_CHECK_MODE) { return; }```
should be
```if (sys.state == STATE_CHECK_MODE) { return; }```

chamnit, 2015-01-17:
@EliteEng : Nice catch Rob. Fixed and pushed update this morning.

closed 2015-01-17 by chamnit