grbl/grbl #575: Check Limit Switch State Before Probing

doppelhub, 2015-01-20:
Existing behavior:
If the probe pin is already tripped when then probe command is issued, Grbl returns an error and will not accept future commands until reset.

Proposed behavior:
Grbl should verify the limit switch is in the correct state before probing.  For example, if the probe is pulled high and trips when pulled low, Grbl should verify the pin is high before initiating the probe sequence.  If the pin isn't in the correct state, Grbl should return an error, but should also accept future commands without requiring a reset.

chamnit, 2015-01-20:
@doppelhub : The proposed behavior is exactly what Grbl does, except the error is an alarm that requires a reset. There are some issues with syncing a probe pin state check. When the g-code command for probe is parsed, it doesn't correlate in time with what Grbl is doing at that moment. This is due to the queue of motion commands in the planner buffer. There are some other issues along with this, if Grbl were to reply with just an `error:`. It's doable, but I've got other pressing things on my plate. 

However, I'll add it to my every increasing list of things to do.

closed 2015-01-20 by chamnit

doppelhub, 2015-01-20:
Thanks for the reply, Sonny!  This would explain another error we've been working through over the past week: uncorrelated errors when stuffing a probe command into the buffer.  Our workaround was to ensure the buffer was empty before issuing probe commands, and your explanation explains why.  Thanks!