grbl/grbl #577: Acceleration control of axis on reversal of direction

JohannesGross, 2015-01-21:
grbl: 9g

1. The accelertion / decelartion control does well if operated manually or by program one axis at a time - including axis reversal.
G91 Y10
G91 Y-10
G91 Y10
G91 Y-10
Aceleration / decelaration is performed between any step

2. But if a batch of g-codes from a program operate two axis at same including direction reversal grbl behaves somewhat unexpected:
G91 Y10 X10
G91 Y-10 X-10
G91 Y10 X10
G91 Y-10 X-10

The acceleration / decelaration cycle relates to the whole batch of these 4 commands and does not care of direction reversal. That is not that what a mill wants to have.
(As I am new in CNC matters I might have perfformed errors may self. In this case please point me to - else it migh tbe an error in grbl)
Johannes

timryder, 2015-02-04:
Can you record this behavior on video and upload it?

JohannesGross, 2015-02-04:
Here you are. Sorry for that lousy quality. You see the intital aceleration - then several sudden direction changes and - then at and a proper decelaration.
Program:
G90 G21 G1
X100Y100
X150Y-150
X100Y100
X150Y-150

Replacng G1 with G0 is cathastrophic. Severe and total step losses, partly missing direction hange totally.
http://youtu.be/YtO0F-c3jLk

chamnit, 2015-02-04:
@JohannesGross : Thanks for the video. That helps a lot. I think I know what's causing this problem. I'm going to push this up to the top of my priority list. Hang tight.

chamnit, 2015-02-04:
@JohannesGross : Can you post a printout of your '$$' Grbl settings? I have a feeling this is numerical in nature and could depend on settings.

villamany, 2015-02-04:
http://3dpburner.blogspot.com.es/p/files.html

-------- Mensaje original --------

Fecha:2015/02/04  22:01  (GMT+01:00)
Para: grbl/grbl <grbl@noreply.github.com>
Asunto: Re: [grbl] Acceleration control of axis on reversal of direction (#577)

https://github.com/grbl/grbl/issues/577#issuecomment-72938136

chamnit, 2015-02-04:
@JohannesGross : Thanks. I think I've located the problem. I'll push the update later tonight. It'll be in the edge branch, rather than the master. The edge branch v0.9h will be merged into master later this week. Please give it a test when I post it.

chamnit, 2015-02-05:
@JohannesGross : Ok pushed the bug fix. Can you try it now? You'll need to change over the edge branch, download the source, and compile/upload it.

JohannesGross, 2015-02-05:
Hi,
1. Please find my current settings below.
2. I just tested same sumultanous procdure with X/Z and YZ. The probelm does not occure.
3. Retested X/Y - it happens again and again
4. Found the repository you mentioned but as I am a bloody redneck in programming matters I cannot compile it. Else please guide me to the file I need for testing.
4. I feel this kind of motion is not rare. Any G0 command will move axes simultaneously from working area to working area.
Thanks for your effort :-) 
John 

$$
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=7 (step port invert mask:00000111)
$3=7 (dir port invert mask:00000111)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.020 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$14=1 (auto start, bool)
$20=0 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=1 (homing dir invert mask:00000001)
$24=20.000 (homing feed, mm/min)
$25=600.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=800.000 (x, step/mm)
$101=800.000 (y, step/mm)
$102=800.000 (z, step/mm)
$110=2000.000 (x max rate, mm/min)
$111=2000.000 (y max rate, mm/min)
$112=2000.000 (z max rate, mm/min)
$120=100.000 (x accel, mm/sec^2)
$121=100.000 (y accel, mm/sec^2)
$122=100.000 (z accel, mm/sec^2)
$130=510.000 (x max travel, mm)
$131=476.000 (y max travel, mm)
$132=101.000 (z max travel, mm)
ok

chamnit, 2015-02-05:
@JohannesGross : Ok. I uploaded a compiled test-firmware for you. [link](https://raw.githubusercontent.com/grbl/grbl-builds/master/builds/grbl_v0_9h_edge_328p_16mhz_115200_build20150205.hex). Right click and save-as. Hopefully you can flash a hex file. Otherwise, you'll need to read the Wiki pages on how to compile. It's fairly easy.

On the rarity point, it's rare from the stand point that this is the first time this has been reported. Most motions in a CNC program do not move like this. I also think it's a numerical round-off issue that can be settings dependent. If I'm right, this bug fix should resolve this once and for all.

kfoltman, 2015-02-05:
Could this also manifest in an opposite behaviour, i.e. missing acceleration control on certain parameter combinations, resulting in motor trying to start at full speed? I'm sure I've seen this behaviour with one file on my machine, unfortunately it was too large/complicated to try and isolate the fragment that triggered it

chamnit, 2015-02-05:
@kfoltman : Possibly. I suppose this problem could have been misinterpreted as causation of something else. The bug fix I issued is more having to do with intermittent diagonal motions that return on the path they just traveled. This doesn't explain a starting from rest to immediately going to full speed. Maybe unless it was in the middle of a program. I would need solid test case to reproduce the problem to isolate the cause of the problem.

JohannesGross, 2015-02-05:
Works grrreat! :-)
I just tested: xy, xz, yz, xyz, xy along a triangle. All like expected.
THANKS :-)

closed 2015-02-05 by chamnit