grbl/grbl #581: Disable homing of arbitrary axes

bigdantheplumber, 2015-01-30:
I'm building a CNC drill press and I'm having trouble homing the machine. Because it's a drill press it only has the Z axis and when I try to home the machine it homes Z just fine but then it does nothing. I don't get an "ok" response and it seems to ignore any input. My guess is it is searching for the X and Y limit switches and not receiving any input so, it just keeps on searching. Here is the section from config.h that I modified. I commented out the second line for HOMING_CYCLE_1 thinking this would work because it says it's optional

```c
#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
//#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
// #define HOMING_CYCLE_2                         // OPTIONAL: Uncomment and add axes mask to enable
```

I also tried changing the values to 0

```c
#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
#define HOMING_CYCLE_1 ((0<<X_AXIS)|(0<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
// #define HOMING_CYCLE_2                         // OPTIONAL: Uncomment and add axes mask to enable
```

I will be building a CNC lathe soon which doesn't have a Y axis. What I don't understand is I can't be the first person to have this problem but I can't find any information regarding this problem anywhere. So, bottom line, my question is: how do I disable homing axes?

chamnit, 2015-01-30:
@bigdantheplumber : You're doing it right, at least in the first part, but I haven't checked the state of this optional homing cycle in a while. I'll try to look into it tomorrow morning quickly. But first, just to get the easy stuff out of the way, make sure you are editing the correct config.h file in your Arduino library/grbl folder, if you are using the IDE uploading method. You have to re-compile Grbl and upload the new firmware with these changes.

bigdantheplumber, 2015-01-30:
Yes, I'm editing the correct file and recompiling/uploading via the arduino
ide. Lol. Man, I've been banging my head against the wall all day on this
problem. It seems like it should work but something just isn't right.

chamnit, 2015-01-30:
@bigdantheplumber : Tested it this morning. It works fine when you comment out the `HOMING_CYCLE_1` line. Make sure your homing switch is wired in correctly too. Can you describe what exactly happens when you run the cycle?

closed 2015-02-02 by chamnit