grbl/grbl #582: question status report Buf

villamany, 2015-01-30:
Hello, i'm creating a windows application for control my laser engraver. I am trying to mantain the 127 bytes rx buffer as high as possible as reccomended on wiki page. When i get the status report message i can only see Buf:0 and Buf:1 messages, Is this ok or it should be more high values?

chamnit, 2015-01-31:
'Buf:' is the number of motions queued in grbl's planner. You are looking for 'RX:'. I think it's in the wiki pages, if not, I'll try to post what they are soon.

closed 2015-01-31 by chamnit

villamany, 2015-01-31:
Yes i read it, but with full rx buffer the planner buffer should be also about full?

Also read as recomended trying to mantain the rx buffer full but also said for 0.9+ the reccomended is simple send and wait response one by one? Whatis the best way for smooth transitions on v0.9?