grbl/grbl #583: Source code of GRBL v0.51

YarikYar, 2015-02-02:
Pls,send me source code of grbl 0.51,i would run it on ATMega 168=)an link in description is wrong.
Or send pls an pinout for v0.51 on ATMega168)
Thanks)
My email is yaryarikyar@ya.ru

Norwegian-Gadgetman, 2015-02-02:
This link?

http://bit.ly/VXyrYu
It should translate to:
https://raw.githubusercontent.com/grbl/grbl-builds/master/builds/grbl_0_51_atmega168_16mhz_9600.hex

Just use 'Save As' to save it as a .hex file.
And yes, it IS a text file with lines like these:
:100000000C9440010C945D010C945D010C945D0115

YarikYar, 2015-02-02:
No,i need uncompiled source,pls:)

chamnit, 2015-02-02:
@YarikYar : We no longer support it and the branch was deleted. If you're lucky, you might find someone with an old fork that still has it.

closed 2015-02-02 by chamnit