grbl/grbl #585: Software Crash @1400 steps/mm

JohannesGross, 2015-02-05:
grbl: 9g

1. The accelertion / decelartion control does well if operated manually or by program one axis at a time - including axis reversal.
G91 Y10
G91 Y-10
G91 Y10
G91 Y-10
Aceleration / decelaration is performed between any step

2. But if a batch of g-codes from a program operate two axis at same including direction reversal grbl behaves somewhat unexpected:
G91 Y10 X10
G91 Y-10 X-10
G91 Y10 X10
G91 Y-10 X-10

The acceleration / decelaration cycle relates to the whole batch of these 4 commands and does not care of direction reversal. That is not that what a mill wants to have.
(As I am new in CNC matters I might have perfformed errors may self. In this case please point me to - else it migh tbe an error in grbl)
Johannes

chamnit, 2015-02-05:
@JohannesGross : Did we not fix this problem with v0.9h? Or is this another issue altogether?

Norwegian-Gadgetman, 2015-02-05:
Maybe I'm missing something here, but why use the G91 command?
(G91 = Set to relative coordinate system)
As this command doesn't have any positional parameters, I'm guessing that that GRBL is running the actual moves as G0 (Rapid positioning) instead of G1 (Linear interpolation, running at no more than current Feed Rate setting)

JohannesGross, 2015-02-05:
Oh sorry. I am confused. Plese delete this part.
John

closed 2015-02-05 by JohannesGross