grbl/grbl #586: Software Crash @1400 steps/mm

JohannesGross, 2015-02-05:
V: 0.9h / 0.9g

I experienced some firmware crashs at resolution of more than 1000 steps/mm (is there a known limit?) I feel there might be some serious problem hidden or limits of arduino power.

1. 1000 steps/mm program runs fine
2. 1200 steps/mm same program runs (log) but mill suffers on unstable stepping. Reducing acceleration makes things a bit better but not essentially.
3. 1400 steps/mm same program stops with different and changing errors (soft limit stop, EEprom failure .....) Most times a power reset of Arduino is suffitient for recovering. Some times it needs new programming.

BTW: This issue is not essential for me just now. I just check the limits of the mechanics of my mill and it seems to be better than those limits. 

Screen shots:
[image]
[image]

Settings:
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=7 (step port invert mask:00000111)
$3=7 (dir port invert mask:00000111)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.020 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$14=1 (auto start, bool)
$20=0 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=1 (homing dir invert mask:00000001)
$24=20.000 (homing feed, mm/min)
$25=600.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=800.000 (x, step/mm)
$101=800.000 (y, step/mm)
$102=800.000 (z, step/mm)
$110=2000.000 (x max rate, mm/min)
$111=2000.000 (y max rate, mm/min)
$112=2000.000 (z max rate, mm/min)
$120=100.000 (x accel, mm/sec^2)
$121=100.000 (y accel, mm/sec^2)
$122=100.000 (z accel, mm/sec^2)
$130=510.000 (x max travel, mm)
$131=476.000 (y max travel, mm)
$132=101.000 (z max travel, mm)

chamnit, 2015-02-05:
@JohannesGross : I have a feeling it might be the GUI program you are using. Please try GrblPanel or UGS to see if that makes difference. Also, I would check for electrical noise problems. It looks like it crashes when you spindle turns on.

chamnit, 2015-02-05:
@JohannesGross : It also looks like your hard limits are triggering, which is causing your problem. Please try disabling them with the `$21=0` command. Then re-run your tests. If the machine works, which it should, you really need to remove any sources of electrical noise and do some signal conditioning on your limit switches.

JohannesGross, 2015-02-05:
Thanks for reply. Tomerrow I will continue investigation.
I do not believe the limit switches are a problem. They have capacitors for debouncing, have no GND connection to the mill case and cables are completley shielded (CAT5). There is only ONE cental GND point at PSU and the PC is optically separated.
The limit warnigns occure while all axis are in the center of their walk. Even soft limit warnings occure at that location.
At 1000 steps / mm the mill run for hours without any issue.

But I understand your hints completely and of course I will double check all your hints.

chamnit, 2015-02-05:
@JohannesGross : There shouldn't be a speed issue with Grbl. People have run complex toolpath jobs at 15,000mm/min, including myself. This leads me to believe it's something to do with your setup or GUI. Go through the process of elimination first and make sure everything is adequately grounded.

GabeHC, 2015-02-06:
It happen to me twice when setting up GRBL 0.9g & 0.9h
both X and Y steps/mm set to 800 and drive it to 15000mm/min 
system hanged and screw up EEPROM 
I believe it hit the step rate limits some where near 40K/sec
so I have to lower to 14000mm/sec to have a stable system

chamnit, 2015-02-06:
Ah ok. This is NOT a problem with Grbl nor a critical issue. This is simply a problem with hitting the max CPU limits of the Arduino. The Grbl project clearly states on our wiki pages and front page that we support up to 30kHz step rates. Your tests you are exceeding that 30kHz and crashing your Arduino. 

Simple solution. Don't do that. Realistically your step/mm shouldn't be larger than 200. This is because you can't realistically cut anything that accurately on a CNCS machine. That's 5 microns if you're wondering.

closed 2015-02-06 by chamnit

JohannesGross, 2015-02-06:
Yep! GabelHC you are right. I calculated the frequency to be 26 KHz. But I forgot the G0 speed. It was twice. And that is definitely ovarload for that little Arduino. 
chamnit: sorry for derangement. One problem more sitting between the chair and the screen :-(

BTW: I do not increase the steps/mm because of accuracy but because of smoothness of motions. In lower regions I have some serious resonances.

OK. Thanks for your support! :-)