grbl/grbl #588: Allocating new settings

electrokean, 2015-02-11:
Over at https://github.com/electrokean/grbl/tree/6-AXIS I am working on a 6-axis version of Grbl.  This will initially be used to run a 5 axis non destructive testing robot.
The X, Y, & Z axis all have homing switches, and soft limits - so pretty standard there.
The B axis will operate a tilt axis, which will have a homing switch, and a soft limit.
But the A axis will be a rotary table without homing switch or soft limit as it handles continuous rotation.
I want to add a new setting for a soft limit enable mask - i.e. one bit per axis - and was wondering if there was any interest in making this a standard setting - e.g. $28.
If not, is there any convention for custom user settings?
Obviously the EEPROM settings for the 6-axis firmware aren't going to be completely compatible, but it would be nice to use a standard setting parameter.

cri-s, 2015-02-11:
Not really, whay don't use the standard, softlimit are only checked if max

cri-s, 2015-02-11:
Not really, whay don't use the standard, softlimit are only checked if
max travel is not 0.
This don't mean that the flag for enabling/disabling softlimits is not
necessary, it could be
useful too.

2015-02-11 8:05 GMT, Cri S <phone.cri@gmail.com>:

electrokean, 2015-02-11:
Are you sure about that?
I'm running from code on the edge branch from only a few days ago, and setting max travel to 0 on an axis causes an immediate limit alarm on any movement in that axis.

Mind you, I just noticed I'm also getting "ALARM: Hard limit", when I though the newer versions differentiated soft and hard limits...

cri-s, 2015-02-11:
3 Axis don't need this, but for more axis, change inside limit.c on
limits_soft_check

  for (idx=0; idx<N_AXIS; idx++) {

    #ifdef HOMING_FORCE_SET_ORIGIN

to

  for (idx=0; idx<N_AXIS; idx++) {
    if (settings.max_travel[idx])
    #ifdef HOMING_FORCE_SET_ORIGIN

electrokean, 2015-02-11:
Agreed, this is probably a better solution than adding another parameter.
Thanks.

closed 2015-02-11 by electrokean