grbl/grbl #589: Disable stepper drivers? (like RepRap M18 "Motors off" effect)

natevw, 2015-02-11:
I have a CNC 3040T ("Chinese CNC" with JP-382A stepper controller and JP-1482 spindle board, should be same controller as perhaps more common 3020T-DJ would have if Googling). This is a parallel port controller intended to be used with Mach3.

Now, IIRC when I used Mach3, it would allow the stepper motors to run free when idle, without any pin settings. With Grbl, the steppers are always "taut" so it is difficult to adjust them with the handwheels when that is necessary [e.g. to back away from a limit switch].

Is there a way to have Grbl do the same thing? I don't know that there is any pinout dedicated for enabling the steppers, it seemed somehow Mach3 was able to do it with just the DIR/STEP pins configured.

rusirius76, 2015-02-11:
In the $$ settings you'll find the "stepper idle delay" set it in the number of mS you want the motors to go idle. If it's 255 it is disabled and the steppers will always be energized.

closed 2015-02-11 by chamnit

natevw, 2015-02-11:
I have these settings:

```
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=1 (dir port invert mask:00000001)
$4=0 (step enable invert, bool)
```

…so I would expect it to idle the motors after a practically imperceptible delay? Or will cycle auto-start interfere with this. I may need to get out my old PC again and figure out what Mach3 was doing I guess.

rusirius76, 2015-02-11:
Yes, that should have them idle after 25mS, and the auto-start as far as I'm aware shouldn't affect it in any way.  Could it be something in the drivers themselves?

chamnit, 2015-02-11:
@natevw : Pin D8 controls the stepper enable pin. Make sure that your stepper driver enables are connect to this pin.

natevw, 2015-02-20:
Yeah, it's clear now from reading the stepper driver datasheets that a separate pin is needed. So either a) this line is brought to a pin on my driver box and I need to figure out which one or b) they are currently pulled high whenever power is on  (which according to schematic for other somewhat-related stepper boards I've found may be the case) and I would have to add in a new optoisolator and do some PCB layout revisions myself.

I think I was misremembering the Mach3 behavior, I booted that machine back up inside and checked the settings — no enable pin was configured there. For now I must just need to turn off to manually turn.