grbl/grbl #590: How to do a tool change?

109JB, 2015-02-12:
I have been following the development of GRBL for quite some time now. I currently use LinuxCNC on a desktop controlling a 3-axis milling machine via parallel port.  I would like to get away from the desktop and move to something like GRBL. I like the fact that the Arduino is doing nothing more than controlling the machine movements.  On occasion with the desktop setup, I receive real time warnings and rarely errors. Even though this is rare, it has prompted me to keep an eye out for solutions. As a hobbyist, I would like something inexpensive which is exactly why GRBL is appealing.  Having said all of that, there is one thing that has prevented me from making the leap to GRBL. That is tool changes.  Right now I use LinuxCNC's manual tool change routine that uses an M06 T# command that pulls the tool length offset from the tool table.  I see that GRBL does not currently support M06, but it does support program pause, probing, and tool length offset (G43.1). I am thinking that this would allow writing into the NC code a list of commands to perform a tool change. For example, performing these steps:

1 - Spindle stop
2 - Move to a designated spot for tool change while retracting spindle
3 - Pause
4 - Physically change the tool
5 - Perform a Z-axis probe of tool length using a tool setting plate
6 - Code that enters the offset using G43.1
7 - Retract spindle
8 - resume program

The thing I don't know is how to enter the new tool length offset into G43.1.  For those that do this, can you explain, and maybe give example NC code for this?  Or, is there another way to do this?

I have reached the point where I am going to give it a shot and have ordered the Arduino, and some other items to try this.  Any help is appreciated.

Also, another thing that I would like to have is drilling G-codes, but this is not a deal breaker.

Thanks,

John B.

chamnit, 2015-02-12:
@109JB : Hi John! Agreed, parallel port solutions are getting old and breaking down and Grbl is finally getting to the point for people to make the leap, as you are about to. Tool changes has been something that a lot people have asked about. So the current thinking of tool changes is that it should be the job of the GUI to handle it. The GUI would perform all of the offset measurements, keep/maintain them, and send the correct offsets to Grbl when there is a tool change. To have Grbl do that would take a lot of precious resources, plus it would be difficult to make it completely generalized for all types of machines. 

This really comes down to a Grbl GUI integrating and supporting tool changes. As far I know, there aren't any that do that are out there yet, but I think there are couple that will be releasing soon. You can always bug the GUI authors about it though! Probably the most advanced version (that supports the most of the new Grbl features) is GrblPanel. I'd start there.

As for drilling g-codes, I investigated installing them once. They are pretty simple but the g-code is weird with their "sticky" variables. If Grbl has the space, they'll be installed, but I think most people can get by with translating them to straight G0/G1 g-codes.

cri-s, 2015-02-12:
For example this SW handle M6 code for Grbl, and a lot more.
http://www.mikrocontroller.net/attachment/247373/SerialComCNC_0.7f.zip
It works only on windows because of the webcam lib.

cri-s, 2015-02-12:
http://www.mikrocontroller.net/attachment/246745/SerialComCNC_v0.7d.jpg
screenshoot.

2015-02-12 16:16 GMT, Cri S <phone.cri@gmail.com>:

109JB, 2015-02-12:
Chamnit - Thank you for the rapid response. I have been looking at the GUI's that work with GRBL and that sounds like a great way to implement many of the features that are missing in GRBL.  I completely understand the need to preserve resources within the Arduino. It seems as though things like the drilling codes could also be implemented in the GUI. For example, could a GUI, when it sees for a properly formatted G81, G82, G83.. code, transform the individual code into appropriate G0/G1 moves? I am not a programmer per se, but have done a little over the years. Many of the GUI's are open source and this would be something that I maybe could attempt.

cri-s - That looks like a great program.  Is there a website that details the program in English?  I have searched but have not found anything. I would be interested in seeing what NC codes are available in the program.  It looks great though.

chamnit, 2015-02-14:
@109JB : Yes, a GUI should be able to convert it, if it had a good parser. Long ago, I wrote a python script that would pre-process g-code programs to convert arcs into G1 commands (Grbl couldn't handle arc early on). That same concept could be done for the full g-code standard. I've been meaning to revisit it and write an open g-code parser library for GUIs and pre-processors, but haven't had the time to do it.

gerritv, 2015-02-14:
In my TODO list (not high up though) for GrblPanel but here is a temp solution: https://pico-systems.com/gcode.html.
The canned cycles are very complex if fully implemented from my reading. Lots of permutations. Generally the code generators can be convince to not output the canned cycle, instead doing G0/G1's. Eg. CamBam's Grbl post processor does this for one of the drilling cycles.

109JB, 2015-02-16:
Thanks again guys.  I did find that some of the CAM software have available postprocessors that are GRBL compatible. CamBam was one that I saw, and also HSM Express has a tinyG postprocessor. As I understand it, tinyG and GRBL have the same set of available G/M codes, so the tinyG postprocessor should work for GRBL as well. I unfortunately don't use either due to financial restrictions on my end. HSM Express is free, but requires Solidworks which is far from free. CamBam is of much lower cost, but still out of reach to me at this point in time.. I currently use pyCam or hand code everything.  I am planning to look intot he possibility of making pyCam so that it is GRBL compatible, or creating my own program to convert the canned cycles that I want to use into G00/G01/G02/G03 code.  Gives me an excuse to dust off my weak programming skills.

Thanks,

John B.

gerritv, 2015-02-16:
CamBam gives you 40 tries, and if you think about how that works it can do a lot of work for you.
Does pyCam output drill cycles???

This is an example of  a script to take one or more of the drilling cycles and convert it to G0/G1. https://github.com/garciasa/grbl-drills-cambam 

A good article on canned cycles: 
http://webs.purduecal.edu/et/files/2012/09/lecture15.pdf
http://webs.purduecal.edu/et/files/2012/09/lecture16.pdf

Not trivial for the generic case

109JB, 2015-02-16:
I have only used pyCAM for a few 3D milling projects that didn't have holes. I guess I never thought about whether it has drilling modes.  I have thought about downloading the CamBam demo, but am really holding out for a 3d CAM. I have had some experience using MasterCam and have access to a seat of that program when I want to drive over to a buddy's commercial shop. He lets me use it when he isn't. Anything more complex that I need to do I generally head over there.

As for the canned cycles, i understand how they work. I was a CNC machinist for many years before I went back to college, and the shop I worked at we were required to do quite a bit of hand programming. We weren't just loading stock and pushing buttons. We actually had to understand how the coding worked. Whether they are trivial or not is dependent on your definition of trivial. The most common canned cycles have a definite flow that can pretty easily be broken down into the individual moves with a few checks for what the current machine modes are. I'm not talking about canned cycles that would require spindle sync like a tapping cycle, but rather the more basic ones like G81,82,83. If there is good understanding of these canned cycles they are pretty basic in my view. I actually am planning to write a standalone program to convert these canned cycles to basic moves as a way to put that Java programming book I asked for at Christmas to use.

alpharesearch, 2015-02-16:
If you start with Java and your create the canned cycles module you could have a look at https://github.com/winder/Universal-G-Code-Sender and it would be nice to have this directly integreted. If you want work with UGS source I would suggest to use NetBeans.

109JB, 2015-02-16:
@alpharesearch , Thanks.  I actually downloaded Netbeans and went through the exercise of making a simple program to get used to the interface back between Christmas and New Year. I haven't been able to find the time to get back to the Java though. Hopefully things will slow down for me a bit and I can dive in.

winder, 2015-02-16:
UGS has a spot for expanding Gcode commands inside the gcode parser, I used
it for adding code to expand small arc's into line segments back when that
was a potential issue with GRBL.

If anyone has an algorithm they want to try out, feel free to experiment:
https://github.com/winder/Universal-G-Code-Sender/blob/1d6a6adfda2d2a1aa030cad4b0288839267eb6b7/src/com/willwinder/universalgcodesender/gcode/GcodeParser.java#L349

109JB, 2015-02-16:
@winder I will definitely have a look.  Of all of the GUI's for GRBL, I really like the look  and feel of UGS.  Some CNC GUI's are to crowded.  UGS has a great balance from the look of it. I will know more once I start using it, but it looks great from my perspective.

Thanks again to all of you.

vlachoudis, 2015-09-15:
Just for information, bCNC https://github.com/vlachoudis/bCNC was recently enhanced with manual till change capability and auto proving for the new tool

zachari043, 2019-11-06:
I use this:

G00 G53 Z-1 //Z home
G00 G53 X-9.6 Y-1.6 // probe position
T1 M00 // changing tool
G91G38.2 Z-30 F100 // rapid first touch
G0 G91 Z1 //retract
G38.2 Z-10 F10 // precis touch
G10 L20 P1 Z-3.4 // distance from part Z0 and touch Z active position
G00 G53 Z-1 //Z home

tcurdt, 2023-08-29:
> I use this:

Could you expand on this? @zachari043 

I get the initial sequence:

```
  G0 G53 (move to probe position)
  T1 M0 (wait for tool change)
  G91 G38.2 Z-30 F100 (rapid probing)
  G91 G0 Z1 (retract)
  G91 G38.2 Z-10 F10 (precise probing)
```
but how is

```
G10 L20 P1 Z-3.4 // distance from part Z0 and touch Z active position
G00 G53 Z-1 // Z home
```
restoring the original position?

zachari043, 2023-09-15:
Better version (not tested)

G00 G53 Z-1 //Z home
G00 G53 G30 // go to probe position set by G30.1 X-9.6 Y-1.6
T1 M00 // stop to manualy changing tool, start to continue
G91G38.2 Z-30 F200 // rapid first touch
G0 G91 Z1 //retract
G38.2 Z-1.5 F20 // precis touch
G4 P1
G10 L20 P1 Z[posz] 
G00 G53 Z-1 //Z home