grbl/grbl #595: Probe Bugs 0.9g

jarretluft, 2015-02-14:
Hey Sonny,
I may have stumbled upon a couple bugs with probing - i'm using 0.9g, not sure if either of these are fixed in the upcoming 0.9h, if so let me know:

**One:**
If I start at 0,0,0 and probe G38.2Z-10F10, and the probe contacts 2 mm into that run, my Z would be at -2, z-motion stops and the probe response returns correctly.  However, if I then issue any G0 command that doesn't specify a Z value, GRBL attempts to move the rest of the way to Z-10 while moving along the other axis.
So, if I ran G0X10 after the probe, the machine would attempt to move 10mm in the X direction, as well as the remaining -8mm in the Z direction to get to the Z-10 specified in the probe...

**Two:**
Prior to a probe command, the $G produces the expected output: 
[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0.]

After a probe command, running $G produces the same output but missing the opening bracket and the first parameter like:
 G54 G17 G21 G90 G94 M0 M5 M9 T0 F10.]

I would have expected this to be [G38.2 G54.....] as G38.2 was the last executed movement command and is listed in the wiki as one of the commands to expect in this position of the $G report.

Thoughts on either of these?

chamnit, 2015-02-14:
@jarretluft : Thanks Jarret. I'll take a look at these issues this weekend. It's likely an oversight by me and doesn't look to be difficult to fix.

electrokean, 2015-02-14:
Probing is important for my current project, so I just tried to reproduce these two issues.
Both worked as expected for me.
@jarretluft what build are you running from ($I), any customisations, and what Arduino hardware & software?

jarretluft, 2015-02-14:
@electrokean thanks for following up:
I didn't realize there were 'builds' within each letter version, just assumed a 0.9g release was fixed, but apparently not.
I used the $I on both my arduino's, one is running 0.9g 20140801 and one is running 0.9g 20140905.
On the older firmware which was my main controller on my mill, I get both issues, but on the slightly newer firmware when I hooked that up to the mill, the issue with sending a G0X## command after the probe works properly, it no longer tries to move the remainder of the distance in the Z direction, so this issue must have been corrected between aug 1st and sept 5th releases.

The second issue with not having an opening [ or G38.2 in the $G output is still there in 20140905 but I will update again, assuming there is an even more current version of the 0.9g hex posted, and see how that works.

Hardware: Uno = 0.9g20140801, duemilanove = 0.9g20140905
Software: both UniversalGcodeSender and Chilipeppr produce the same issues.
No customizations. Stock hex file downloaded from the grbl github page -- obviously downloaded at different times, given the different builds.

electrokean, 2015-02-14:
Possibly this commit fixed the first issue: https://github.com/grbl/grbl/commit/d55f6b938cd54a95e14d94e2ad7730d1aeb772a4
Not sure about your second issue, but I think it may have been fixed in this commit updating report.c: https://github.com/grbl/grbl/commit/7e67395463b7fedbd0ac2e03aef2e84875d8dcbc#diff-cf560e4ac2d02a0306ba5da42c18c5a2
There doesn't seem to be a hex build since that later one you're using, and the second commit above isn't in the master branch.  So you may want to have a go at downloading the latest edge branch, and compiling it yourself.  Instructions in the Wiki.

chamnit, 2015-02-15:
@jarretluft : If the edge version indeed fixes it, I'll wait until the newest version gets vetted for any major problem. I'll then push that to master.

jarretluft, 2015-03-06:
Sorry for the delay, Edge does fix both issues I was seeing.  I will close this off.

Side question: 
I notice that probe now has a :1 at the end of the X/Y/Z coordinates, what does that represent?
[PRB:-30.000,30.000,-3.959**:1**]

closed 2015-03-06 by jarretluft

electrokean, 2015-03-06:
@jarretluft the ":1" at the end of the probe result indicates a successful probe.  A ":0" would inidicate unsuccesful.  This is used with the non-erroring versions of the probe commands (G38.3 & G38.5).

jarretluft, 2015-03-06:
Awesome! Great to hear 38.3 is now supported!
Thanks! :)