grbl/grbl #596: Consider M17/M18 for stepper power on/off

richardjm, 2015-02-14:
When running a job I'd like the motors to remain enabled always to prevent unexpected movement. But  when a job is finished I'd like to give the motors time to rest by turning them off.

With 3D printers this is supported via the M17/M18 commands as described http://reprap.org/wiki/G-code#M17:_Enable.2FPower_all_stepper_motors - I was wondering if this would be a nice addition to the MCode features.

chamnit, 2015-02-14:
@richardjm : Probably not. I've been generally hesitant to add non-standard g-codes to Grbl because it only makes the haphazardness of g-code "standard" worse. Grbl tries to stick to LinuxCNC's definitions as closely as it can. But, I think we can come up with some other solution, like a sleep mode triggered by a '$' grbl command.

richardjm, 2015-02-14:
Fair enough. The $ commands were new to me coming from Marlin but I've got
used to them so another $ for a sleep mode seems ideal.

closed 2015-02-14 by chamnit