grbl/grbl #597: QUERY: $1 = Step idle delay usage?

chamnit, 2015-02-14:
All: 

A question about installing a sleep mode into Grbl has been brought up. I'm thinking about if we should remove the `$1=255 (step idle delay,msec)` setting, keep it, or move it into a config.h define.

How many of you:
- Keep your steppers always enabled by setting `$1=255`?
- Let the steppers power down when Grbl IDLEs by setting `$1` less than 255?
- Think a sleep mode via a `$S` command will be something you want?

timryder, 2015-02-14:
I like the idea of using an $S command to disable the motors a Ton.

alpharesearch, 2015-02-14:
I think everybody with a ShapeOko or any other system that uses GT2 belting needs the steppers always on. So I would think it would be good to have this as a default setting and have a $ sleep command to keep the steppers cool over long idle times.
In my case I power the machine only on if I need it, I don't have it just sitting idle around for long. But if there would be a sleep command I think I would add it to my g code file on the start and end.

electrokean, 2015-02-14:
I'd like to be able to disable the motors between jobs, and I think a Grbl specific sleep command that the GUI can use is a good idea.
The existing $1 != 255 default isn't ideal for use with micro-stepping drivers anyway, and they're pretty common.

gerritv, 2015-02-14:
I leave that setting =0, don’t find that my motors get hot just sitting there.

 

Perhaps using “%” would be a good alternative? One at the top of the file, then if there is one at the end turn off the motors. One at the end only is an error.

Or use a setting to indicate the M30 should turn off the motors.

 

Gerrit

cri-s, 2015-02-14:
I use it (with little dodification). I have reserved 251-255 for special op.
Not for stepper enable, but principally for reducing current.
If not using it, it is handy for led to show movement.
255 is actual code.
If setting it to 254, on mc_reset (M2, M30) after 254 ms, the stepper
driver are deactivated.
This is nice feature, but not always wanted, because on microstepping
the position is lost.
It works with microstepping to, if before M2 code G28 is used.
252/253 is manual control, 253 activate it and 252 deactivate it.
If after powerup the setting is 253 it enter alarm state.
251 is code to output M10/11 on this pin.

Other modification , start/homing and pause/stop on the same pin using
ADC input.
mit licence, works on gpl too.

paulkaplan, 2015-02-14:
Easel uses it to lock / unlock motors, but only because there is no energize / sleep setting!

Some Shapeoko I've seen have it set to less than 255 so they can manually move the gantry while the machine is on for convenience. It seems like if the idle timeout was changed so that the motor went idle only when all axes were idle, it would avoid the Shapeoko belt problem and be a more intuitive setting. It is strange in context of a CNC that the timeout only works on one axis at a time (i.e. if the Y axis is moving, the X axis steppers should remain engaged). The "%" suggestion by @gerritv seems like a good way to do this also.

Questions about how that would work though:
- Would the machine power up in sleep mode?
- What happens if gcode is sent while in sleep mode? Does it "wake up"?
- Would sleep mode be part of the real-time reporting?
- Would sleep be global on all axes, or independent per axis?

chamnit, 2015-02-14:
@alpharesearch : I agree. GT2 belts and disabling the steppers isn't compatible. You have to keep everything enabled so you don't slowly lose position. I think it has to do with stepper drivers losing microstep count when its disabled and then reenabled or GT2 belt drives are really sensitive. Probably the latter.

@gerritv : I considered the use of `%` as a program start, but this doesn't help distinguish jogging commands. For GT2 systems, you really have to have the steppers enabled from power-on, homing, jogging, to throughout program run. Using `%` doesn't capture these cases. For lead-screw CNCs, I don't see a problem with disabling my steppers, since there is so much friction holding things in place. 

Using M2/30 could be done. But, I still worry about loss of steps in GT2 systems, if a user needs to run another job immediately after. The user would have to re-run a homing cycle to ensure location is preserved.

@paulkaplan : Grbl shouldn't disable individual axes. There is only one pin that enables and disables all of the stepper drivers, so if any one axis needs to move, it enables all of them. It only disables when there are no motions in all axes.

I propose the following operation:
- New SLEEP state. Reported in status reports. To be used by GUIs to know it needs to account for issues pertaining to losing steps if drivers are disabled and re-enabled. Grbl will not enforce homing after exiting SLEEP to keep things simple.
- During SLEEP, any motion command that re-energizes the steppers will exit sleep mode and revert to IDLE when doing nothing.
- IDLE state signifies doing nothing, but steppers are always enabled.
- Any ALARM state automatically disables steppers, like sleep, but when exiting via homing or `$X`, Grbl will enter IDLE, rather than SLEEP. (Not sure about the latter, but seems to make sense to me.)
- Upon power-up and homing cycle disabled, Grbl enters IDLE. Thus, enables steppers persistently.
- Upon power-up and homing cycle enabled, Grbl boot-up enters ALARM as always. Steppers are disabled.

chamnit, 2015-02-14:
Another proposition:
- It should be the job of the GUI to tell Grbl to go to sleep after a job. It could be as simple as a checked box in the GUI that would send a `$S` sleep command after the last line of the g-code program is sent.

alpharesearch, 2015-02-14:
@chamnit @gerritv : why not add wake up from sleep to the existing ~ cycle start command, what does a new % wake up command give you?

@chamnit : do you plan to have one command ($S) to toggle between sleep and and idle and use the status report to determine the status or have one command to set to IDLE (~ or %) and one command to set to SLEEP ($s)?

timryder, 2015-02-14:
I personally like total control over the state of my motors. I'd prefer $S command

chamnit, 2015-02-14:
@alpharesearch : I'd rather not confuse the user with a sleep toggle. $S should only make Grbl go to sleep. 

I like the idea of using the cycle start '~' key to exit sleep without having to use a motion command. I don't like '%' as it  mucks up the gcode standard.

paulkaplan, 2015-02-14:
I feel like this is just confusing the state machine even farther. What is the use case for sleep?

alpharesearch, 2015-02-14:
If I think about it a little more, waking up should not require a specific command, just any command that requires the steppers should wake them up. This turns from SLEEP to IDLE and it only goes back to SLEEP if the sleep command is sent. If someone wants to lock out the machine this could be done by putting GRBL in a soft alarm state via a new command, this would enable sleep and keep the machine from moving until out of alarm state. If you clear the alarm the machine starts in sleep until any command is sent that needs the steppers. After power up the machine is in SLEEP, after homing the machine would be in IDLE.
@paulkaplan : my use case, I would like the sleep mode to keep my steppers cool when not needed

Norwegian-Gadgetman, 2015-02-14:
I also keep my steppers always energised. I did it when I ran with MXL belts and I do it now with GT2 belts. And no, I don't think MXL belts are in any way immune to losing position when the steppers are disabled. Steppers have a slight tendency to 'click onto' a whole step position if not energised. Whether this is exaggerated with GT2 belting or not, I don't know, and have no plan to test.

If I want to move the steppers manually, I hit  the big, red button...

rusirius76, 2015-02-14:
Agreed. My z axis uses gt2 belts and two motors. Steppers going idle results in the x carriage crashing into the table. So it's imperative that my steppers remain enabled. I don't have issues with heat, they barely get warm,  but if I did, I think a sleep command would be great.

chamnit, 2015-02-14:
@paulkaplan : The main use case is to disable the machine when a job is done. The best feature about CNC is its automated and you can leave the machine. If you have a really long job that doesn't need a lot of tending, you can leave and have Grbl go to sleep when it's done. 

I think you're right @alpharesearch, any command should wake Grbl from sleep. To keep people from accidentally using $S by fat fingering the keyboard, maybe the command should be $SLP.

poelstra, 2015-02-14:
Given discussions about potentially adding a SLEEP state to the state machine, I was wondering whether I might add the following input to the discussion:

On my setup, I have an electrical Start/Stop circuit (a relay is latched on when Start is pressed, and keeps latched until Stop is pressed or power is removed). I've recently switched from LinuxCNC to Grbl, and on LinuxCNC I had the state of that circuit wired to an EStop input.

When the Start/Stop circuit was in stop state, LinuxCNC would prevent pressing the green "On" button in its Axis UI, thereby preventing any movement (like homing). Only if I press Start on my hardware (which also energizes the steppers), LinuxCNC allows me to press "On" in Axis, thereby enabling functions like homing. If I press Stop on the hardware, it automatically also turns its software 'version' of it to off.

I kind of expected similar behaviour in Grbl, wiring my EStop to Reset/Abort, but apparently any toggle of this pin (on or off) soft-resets Grbl.
Now, when I press Stop, I think it makes sense to soft-reset Grbl, and lock out everything.
However, it's then possible to e.g. do $H, which won't actually move my hardware because the drivers are physically disabled.
Additionally, when I do press Start, Grbl again soft-resets, which I didn't expect.

So I guess my use-case is in a way pretty close to a SLEEP mode, but I'd like to propose a mode where this SLEEP(ish) mode is NOT automatically woken up by a movement command, but instead prevents the movement. I.e. it is entered as long as the external hardware EStop is active, keeps active when I release the external EStop, and can only be manually unlocked by a Grbl command ('~' or a $-command). (Maybe the mode should then be called DISABLED, or ESTOP, but anyway...)

As an aside, my driver board has the option btw, to reduce (i.e. not completely turn off) the motor current. I've currently wired the driver enable pin to that, so I do like the existing $1 setting to automatically lower driver current after some time.

Hope I'm making any sense here...

chamnit, 2015-02-14:
@poelstra : IMO E-stop should not be handled at all within Grbl. Too many things could go wrong. I always recommend that estop simply cuts power to the entire machine or disables it completely through hard wiring. 

What you're asking for is something along the lines of a safety door switch, which was just released this week in the edge branch. It'll feed hold and suspend Grbl and de power everything except the stepper drivers. Queues incoming data and waits until a cycle start to resume. The switch must be closed to work. 

Sleep mode is intended simply as a way to disable the steppers through software for whatever reason a user may need it for. Most would be to reduce power consumption when it's not being used.

twforeman, 2015-02-14:
I keep my steppers energized all the time that the machine is on. If there was a sleep mode I would probably not use it, but that doesn't mean it shouldn't be implemented.

The reason I would not use it is because after sleeping you would need to home the machine again to re-pick up your zero.

I tend to turn on my machine, do a few jobs, and then power it down again. I don't leave it sitting on and idle for long periods of time.

GabeHC, 2015-02-15:
Keep the function there! and change to int32. So we can set it from few milliseconds to few second.

chamnit, 2015-02-15:
@GabeHC : For what purpose do you need to have the stepper idle powered for a few seconds before disabling? I'm having trouble imagining a special scenario that this would be needed.

Idling while powered is generally what a G4 dwell command is for. If you need to disable after, the proposed $SLP command would turn it off.

alpharesearch, 2015-02-15:
@chamnit : I think it would be a good if the proposed $SLP command can work just by being on the end of the g code file. In other words not a real time command like feed hold for example. I think in this case people can just put this command on the end of some of there g code files without having to wait for sender program support.

extrafastlight, 2015-02-16:
Since this is a DYI project with non-standard hardware, a $S would be useful but should not be mixed up by the way, energy can be saved / or heat dissipation could be reduced.

The current 0.9g just provides one 'enable' sum signal for all axis. This makes it useless for external current decay but has more the functionality of a sleep mode for the hardware since all axis are 'ored' together.
There are stepper driver out like the older TA85xxHQ who has a currecnt decay input signal to reduce the current when not moving while other driver do that automatically 2M9xx.
A further item to note is the capability of the stepper driver to keep the microstep position (state) while not enabled.

So, before talking about a sleep mode, I would like to see a function for separate 'enable' outputs to control the current of an axis. Using this signal for a 'full energy switch off' or just for a current decay is then to the 'project manager' :-)

So, I vote for $S but not as a realtime maintained state. A GUI or control programm could send this command on demand.

1) separate enable for each axis
2) $S0 as a command bit not as a realtime state
3) axis sync to the spindle :-)

BTW: I think the stepper driver and current charge should be in a range where the motor stays almost cool. Continuous load would heat the motor anyway, so a sleep mode would not help to reduce the heating.

chamnit, 2015-02-16:
@extrafastlight : I had a hard time following your post, but Grbl only controls the stepper enable via one pin. It doesn't wire in with the sleep features on stepper drivers. Although named similarly, Grbl 'sleep' mode only disables the steppers through that enable pin. 

Anyhow, all `$` commands are not realtime and will be processed according to the order the commands are sent. So, yes, you can place a `$SLP` command at the end of every g-code program, or have the GUI append one for you, to disable the steppers when you're done with a job.

As for 1) and 3), Grbl on the Arduino Uno does not have enough available pins to do separate enables per axis, and spindle-sync is a whole different topic.

gerritv, 2015-02-16:
Perhaps we should revisit why this came up in the first place, the solution is starting to sound complex.

 If your motors get warm, don't worry.
 If your motors get truly hot to the touch then perhaps you have other problems? E.g if your motors are buzzing when idle then the decay or current is wrong.

Were I to use $SLP then I would Home before starting another program. Any time the steppers are 'released' means you no longer can be confident of their position.

chamnit, 2015-02-16:
@gerritv: The original reason is to get rid of the $1 step idle delay setting. This in itself is confusing. And from the sounds of it, not used at all. 

My intent was to remove $1 and just keep the steppers enabled all of the time, except during certain alarms. 

This sleep command is nothing more than a way to turn off the steppers by the user for whatever reason. For example, if someone does use homing and want to manually move their axes by hand, they can use sleep to get the roughly into position (super easy on belt type CNCs). Then re enable to locate their part zero. Not everyone uses homing.

tmpvar, 2015-02-16:
@chamnit I'm a +1 for sleep mode via a `$S` command (Preferably with gcode / `~` input to resume)

We will need some clear documentation to explain the potential loss of (micro)steps

ashelly, 2015-02-17:
My use case and code base have diverged fairly far from stock grbl, but for what it's worth, I use 2 different step_idle_delay settings.

I have an encoder on one axis, and use 0 during calibration to detect the limit of travel where the motor starts slipping.
The rest of the time, the machine runs 24-7, and I send it gcode in small batches. I set $1=250  so that  the motors hold position during sequences of moves, and shut down while idle to reduce power consumption and heating.    For one operation, I found that 250ms  wasn't long enough, so I added a compile time option for a long delay - axes with the bit set have their shutdown delay multiplied by 250. (I also made the disable call asynchronous to avoid stalling `st_go_idle()`)

If you replaced the delay with always on plus a sleep command, I could rework my sender to use that. But I'd also add back a way to run like the current delay 0 setting for my calibration mode.

chamnit, 2015-02-18:
All: After some thought, here's a revised proposal.

- `$1=50 (step idle delay, msec)` setting will be removed.
- Grbl will always have the steppers enabled.
- There will be a compile-time option to have the steppers disable when motion completes to support users that still use the `$1` setting.
- A new stepper sleep/disable command that is not real-time. Enacted by `$SLP` or `$DISABLE`. Most likely something like the latter to avoid confusion of the SLP pin common on stepper drivers. This only disables the 'enable' pin.
- Grbl will enter a suspend state, where only real-time commands can interact with Grbl. Only a cycle start command will cause Grbl to exit and re-enable the steppers.

Sound good to everyone?

ashelly, 2015-02-18:
By real-time commands, you mean the single-character ones in config.h?   
So that means you can't download  a new bunch of gcode while it is disabled, you need to send the cycle start first, even though there may not be code to start executing?   Seems OK, just want to be sure I am understanding.

Also, It would be nice if the compile-time option is a number like the current EEPROM  setting.  If I do want to disable motors, it is nice to have some control over the timing.

gerritv, 2015-02-19:
Sounds good to me for my scenarios.

electrokean, 2015-02-19:
Sounds good to me too.
I'm assuming motors will be enabled by default on Grbl startup?

poelstra, 2015-02-19:
This sounds like a building block for the EStop workflow that I mentioned before.
Don't want to hijack this thread though, so I've created #604 to check your opinion on that.

chamnit, 2015-02-19:
@ashelly : Yes. Real-time commands are the single character ones in config.h. Status reports should still work during a suspend, so that the GUI and user knows what's going on. The suspend would just queue incoming serial RX data and not parse it, nor will Grbl send back any 'ok'/'error's for the GUI. Safety door and feed hold now use this new suspend state in Grbl v0.9h already. A compile-time option would work as $1 does now, but you'll have to recompile if you want to change it.

@electrokean : Yes, Grbl will always keep things powered on, if everything is running fine. An alarm may be an exception. I'll have to think about this a bit. It probably could be argued both ways that an alarm should disable the motors, or keep them enabled. But, most likely, Grbl will keep them enabled by default and users could configure this behavior in a config.h define.

@poelstra : See my Estop response on this, but this does bring up considerations of how the stepper enable should behave upon a reset and a GUI-driven e-stop,.

powercycle, 2015-06-04:
Can't we already achieve "sleep" mode at the end of the job by issuing $1=0?
I like the idea of enabling the motors as a default ($1=255).
Is this allowed in the Nx strings?

sebbra, 2015-06-04:
I have my stepper-enabel hardwired to enable for drivers. I use the stepper-enable output as a moving-LED.

109JB, 2015-06-04:
I know this is an old issue, but I will give my spin on it.

My machine has ballscrews and the Z axis is heavy enough that it can back-drive the ballscrew and drift down if the stepper is not enabled. For this reason, I have the steppers drivers enabled all the time. The particular stepper drivers I am using have a built in 1/2 idle current setting where the current to that motor is dropped. This is all handled in the stepper driver though and I don't need GRBL to do anything here.

For other purposes it would be nice to have the ability to enable or disable the motors via user command.  I am not sure that it needs to be anything in EEPROM settings though. Perhaps an EEPROM setting so the user can pick his default for startup (enabled or disabled), and then a command that can be used to switch them on or off.

For someone that ha no limit switches installed it would definitely be nice to switch them off to manually position the machine to their "home" position.

All that said, for my setup with homing enabled, I can't think of a reason that I need anything different than what it has now.

chamnit, 2015-06-04:
@109JB : Understood. This feature is in my development path for v1.0. I ended up deciding not to add it to v0.9 since it'll require some refactoring of how things work a little too much.

blinkenlight, 2015-06-05:
It seems I missed this discussion somehow back when it started, sorry. Just to add my $0.02, even though it probably won't change anything this late: yes, I am very much relying on the "stepper disable" signal, it's absolutely a must-have for me. As others have mentioned, I'm using it to reduce current any time the motors are not moving because the motors are more or less melting down otherwise (yes, they're set for the right current - they just heat up way too much for my liking). Considering I'm keeping the mill idle for extended periods while I'm fumbling to rectify the latest mistakes with the CAM software before the next pass, having the mill cool down instead of catching fire is fairly essential for me. Notably, since the drivers are not actually disabled during this time, position down to micro-step is not lost this way - something the next pass fully relies upon.

The other thing to note is that, no offense, but based on my existing experience with GUIs, I'd **NEVER** trust this being handled by a GUI - they do enough damage as it is; this absolutely needs to be a hardware feature in my opinion. I'll be certainly interested in any option to have this working as it does now in 0.9.

mschorer, 2015-06-05:
Just to add another 2ct:
I'm using threaded rods - also set current right - it seems (and definitely
feels right when touching the motors) obvious to power down the motors.
Also wouldn't want to rely on a host software ...
To me, the current state of sw with configurable timeout seems the perfect
solution.

109JB, 2015-06-05:
Just a few comments, not about GRBL but steppers in general.  In terms of operating temperature, steppers naturally run hot. It is the nature of the beast. Having said that, they are also designed to take it. Many stepper motors are designed for a maximum case temp of up to 100 deg Celsius. A temperature that will cause a 3rd degree burn on contact in less than 1 second. Even a moderate temperature of 50 deg Celsius (about 120 deg F) is too "hot" to touch for more than a few seconds and will burn you with prolonged contact, but is well within the norm for a stepper motor.  I think a lot of people that think their steppers are running too hot are using way too conservative an estimate of their stepper motor's temperature in this regard. Mine run a maximum of about 60 deg Celsius measured with a non-contact IR thermometer. I can't touch them and I don't really care because they are still well within what they are designed to take.

In regard to the above, if someone is really worried about it, there are stepper drivers that do idle current reduction to prevent heating. Most of these limit the current to the motor when idle to a 1/2 of the set current. I know that a lot of people are using less sophisticated drivers that don't have this option, and what I would say there is get an IR thermometer and check your motor's REAL temperature. If not over say 60-70 deg C then I wouldn't even worry about it.

109JB, 2015-06-05:
Oh. the 60 deg C number for my steppers was with my old drivers that did not have idle current reduction, but I didn't switch drivers because of heat, but rather to get better motor performance (Less resonance, higher speed, etc.). The old drivers are going on a new project soon that doesn't require uber-performance.

blinkenlight, 2015-06-05:
I appreciate the comments. To be fair, I'm not actually expecting my motors to catch fire, and I am fairly sure they're well within their operating range; it's just that they do, as I have put it, "heat up way too much for my liking". Sufficiently so for me to have hacked an add-on piggyback board over the driver PCB, to enable it to cut down current when "disabled" (something the driver *can* do, but not automatically). Considering I have spent considerable effort to fix the drivers (unrelated issues) and that now they work fine otherwise, I obviously don't intend to replace them. I guess what I'm saying is that I'm aware I don't necessarily **need** this feature to use my mill, but I will do **whatever it takes** to keep using it.

mschorer, 2015-06-05:
Same here.
Maybe not a necessarity, but power down timeouts definitely sound like "the
right thing to do".
Also, from an implementation p'o'v, adding one more state should require
reasonable effort ( memory, code and implementation wise).

dguerizec, 2016-05-29:
I tried to implement sleep mode, and well, it's working pretty good with very few changes.
Let me know if my approach is flawed or if I've missed something, but it does not look as complex as I read here.
Let me know also if I have to port it on the edge branch (didn't use/test that branch yet).