grbl/grbl #598: Serial Baud Rate

EliteEng, 2015-02-15:
I would like to start a conversation about baud rates.

I have been setting my baud rate to 250000 for a while now and I am getting very good results.

If you look at Table 20-7 of the datasheet http://www.atmel.com/images/Atmel-8271-8-bit-AVR-Microcontroller-ATmega48A-48PA-88A-88PA-168A-168PA-328-328P_datasheet_Complete.pdf it shows the commonly used baud rates and error%.

The only reason I see not to change would be so the Arduino Serial Monitor can be used. ( I don't use this anyway ).

I use GRBL-Panel (for general use) and Realterm (for quick testing) both of these you can adjust the baud rate.

mentyl, 2015-02-15:
I tested and use 2000000 winth arduino nano clone board with CH340G (USB<->serial chip) for with 2000000 is max. Working fine error% on Atmega is 0% (for 115.2k error is 2.1%). As CH340G datasheat says, error must be no more then 2% :).
For 250000 error is 0% on Atmega. I testet this with my board (CH340G), in CH340G app note this speed is not suported but working with no problem :)

darkfibre-nl, 2015-02-15:
@mentyl the max. baudrate isn't specified. The datasheet puts it this way:
' Hardware full duplex serial interface, set transceiver buffer, supports
communication baud rate varies
from 50bps to 2Mbps. '
and

'CH340 supports common baud rate:
50,75,100,110,134.5,150,300,600,900,1200,1800,2400,3600,4800,9600,14400,19200,28800,
33600,38400,
56000,57600,76800,115200,128000,153600,230400,460800,921600,1500000,2000000
and so on.' (note: the last bit.) Google translates the last character from
the chinese datasheet ( 等 ) as ' etc ', and 'and other '. So although
250000 isn't mentioned specifically, I feel safe to assume it's supported.

mentyl, 2015-02-15:
Yes, you have right, but i think error is not specified for not listed speed, and may be greater then this 0,3% in datasheet.

chamnit, 2015-02-15:
I did a throughput calculation once. As I recall, 250000 baud shouldn't give you any benefit. Grbl can't process gcode faster than it can be sent at 115200 baud.

electrokean, 2015-02-15:
Yes, unless you're sending a lot of really short movement commands (like an arc broken into lines) with a high feed rate, you're not likely to starve Grbl of work at 115200 baud.  In addition, you're more likely to have a communication error at high baud rates.  There is no checksum or such in the protocol to detect something like losing a digit in a coordinate, so no point using a higher baud rate if it doesn't benefit you.
With regards the error at 115200 baud - with a 16MHz crystal it is about 2.1%.  With asynchronous communications the timing is restarted on every start bit, so it is only cumulative across the 10 bits (start, 8 bits, and stop), giving a max of about 19% on the stop bit (9 x 2.1%).  You can theoretically get away with as much as 5% error per bit (+/-45% on the stop bit), but you can have errors on both ends of the communcation, so best to stay below 2.5%, and the lower the error the better.  Cheap Arduinos with resonators instead of crystal oscillators will introduce further error, and shouldn't be used at high baud rates.

chamnit, 2015-02-15:
Grbl can process and execute a g-code block every 4-5ms at worse case. This translates to roughly about an average 30 characters per block at the highest rate. There are experiments by @tbfleming (JScut) on YouTube showing the high speed machining performance of Grbl. It's limited by g-code parsing, planning, and execution, rather than the 115200 baud.

I suppose it's possible that you'll get slightly better high end performance by bumping up to 250000 baud for certain scenarios, but since 115200 baud is a standard baud rate supported by just about everything, I think it's safe to stick to that for the general release.

mentyl, 2015-02-16:
I tested three boundrates 115200,250000,20000000 with my arduino nano and CH340 USB chip. With simple C# code that sends simple G-code
1 Test,
start position X0Y0Z0
G0X100
G0X0
Avarage time off all boundrates is almost the same about 6.5Sek
2 Test, same distanse, but gcode every one milimeter - results as above for all boundrates.
3 Test, same distanse, but gcode every 0.1 milimeter
Resoult:
 115200 - 1/10 ok other fail due to comunication error
 250000 100% fail due to comunication error
2000000 100% OK with Avarage time of 10 pass - 7.15Sek
For me with no CRC or other check I have good reason to use 2MB speed :)

mentyl, 2015-02-21:
More tests, more errors on all bauds. I simply add an hash using DJB2 algorithm on command end after ';'  ($H;21900). But i add this to protocol.c . Real-Time commands: Reset,Feed hold stile sometimes occour, because they are interpreted in serial.c without eny check.
I think is't safe to use grbl withaut any command transmission check. I plan to move DJB2 hash check to serial.c Wchich will impact Real-Time commands too.
I dont see this on TODO list but i think this is very usefull.

tbfleming, 2015-02-21:
Is there any reason to support Arduino clones which don't have reliable communications? Why not just use the real thing?

mentyl, 2015-02-21:
Price :), and i think "original" arduino board and FTDI chip not guaranty 100% safe comunication with no  errors.

Norwegian-Gadgetman, 2015-02-21:
The original is what? $25
And we know that it hasn't been assembled with factory reject or fake chips...

Remember the FTDI driver debacle not long ago?

closed 2015-02-22 by chamnit