grbl/grbl #599: DVD drive stepper not moving

mtmeyer, 2015-02-15:
I pulled apart a dvd drive for the Z axis of my CNC plotter and i cant seem to get the stepper to move at all.  I changed all of the settings to match these: http://www.g7smy.co.uk/?p=391 to try and see if the settings were the issue but it still didnt work.  Any ideas?

electrokean, 2015-02-15:
As a starting point I'd I suggest trying a lower Z axis step rate ($2) and seeing if that helps.  The page you link to doesn't actually use the Z axis, so their settings may not be appropriate.
Other than that it is almost certainly a hardware issue - you might want to post on a forum related to your stepper driver (EasyDriver?)

mtmeyer, 2015-02-15:
I mainly took the step/mm setting from that website as i assumed it would be the same for my stepper/leadscrew dvd drive assembly.  I will try and lower the step rate and see if anything happens.  I am using an A4988 stepper motor driver so if lowering the step rate doesnt work i'll find a forum relevant to the A4988.  Cheers

closed 2015-02-15 by chamnit