grbl/grbl #600: QUERY: no stop on homing switch

njrl, 2015-02-15:
hello,..
when i start the homing cycle, grbl wont stop at the end switch. the switches are NO and working fine because the activated hard limit feature react like expected. 
i'm running [0.9g.20140905:]
my config: http://pastebin.com/Wd0LnG8Q
any suggestions? 
thx in advance

chamnit, 2015-02-15:
@njrl : Hard limits trigger upon a pin change. It's agnostic to type of switch. It sounds like you have your switches wired as NC or incorrectly wired. Homing will behave this way if you have.

njrl, 2015-02-15:
@chamnit : ahrg ok ... my fault ... the switches are ok i'm using an protoneer cnc shield V3.03 and i did not set the jumper for selecting VCC or GND as opposite side and the wire capacity was enougth to trigger the hard limit. thx for the hint to figure out.

my followup question is:
what is the simplest way to disable the homing of one axes (X), because I dont have/need these? 
I mention is not possible with the standard build.

chamnit, 2015-02-15:
Look in the config.h file. There are a number of advanced options including homing only one axis.

closed 2015-02-15 by chamnit