grbl/grbl #602: Any plans for spindle synchronization

109JB, 2015-02-19:
I am preparing to get GRBL running for my milling machine and think it will work great, but the lathe is another story. Since threading on a lathe is a relatively common operation, I don't see being able to go to GRBL without having threading ability. This means spindle synchronization would be needed. Is there any plan to implement this in the future?  I searched and saw an old topic about this, but was just checking to see if there has been any more thought into it. 

Thanks

chamnit, 2015-02-19:
@109JB : Spindle sync is a personal goal of mine, since I have a micro lathe and want to CNC it as well. But this feature buried down my priority list, as Grbl primarily supports 3-axis cartesian machines. 

Spindle sync itself isn't that difficult to implement, as long as you have a consistent expected input from the spindle encoder. What really needs to be defined is what that signal should look like, such that a typical hobbyist could wire one onto their lathe and into Grbl easily. It should also be very robust. 

If you have suggestions on a simple circuit and the input that Grbl should expect, that will go a very long way to helping closer to getting this installed. Here's one idea (of many) that I liked. [Link](http://www.davehylands.com/Machinist/CNC/CNC-Spindle-Encoder/)

109JB, 2015-02-19:
There are commercial boards that are available that aren't expensive at all.  here is one that is $25 assembled.

http://www.cnc4pc.com/Store/osc/product_info.php?cPath=25&products_id=129

it is the same circuit that DAK engineering uses on turbocnc, so if you wanted to build it yourself here are some links.

http://www.dakeng.com/threading.html
http://www.wrathall.com/Interests/CNC_conversion/spindle_encoder.htm

The first link has a link to a several pdf's with instructions and hookup.  

I realize that the main audience right now for GRBL is 3-axis cartesian machines, but I think you are selling yourself short. One year ago, I would not have considered GRBL for either of my machines and now I am convinced that something like this is the way to go. Where you can have rock solid motor control from a microprocessor that is dedicated to do only that task. On a milling machine, not having spindle sync just means that you can't do rigid tapping. A nuisance, but with tapping you can just pull the part off the machine and tap it using a cordless drill, or similar. Not a big deal.  On a lathe however, the lack of spindle sync means you can't thread which is a deal killer for just about every CNC lathe out there. So, it is not surprising that there isn't much use for GRBL on a lathe. It is simply because it isn't practical on the lathe with the no-threading limitation.

I really hope that you can implement this in the future.

Thanks 

John B.

chamnit, 2015-02-19:
@109JB : Thanks John. It's been a very interesting year. One year ago, I wouldn't have recommended using Grbl as replacement for traditional parallel port-based CNCs. Now, it's come into its own, mainly due to the recent performance and stability improvements of Grbl v0.9. It's very very useable in its current form, as you found out, but there are some higher priority items than the spindle sync.

I would argue most people wouldn't know what to do with spindle sync, if Grbl had it. Only a relatively small group of experienced CNC machinists would. But, everyone would benefit with real-time feedrate overrides and spindle overrides. That's next on the list of priorities. 

IMHO, lathes are strange in that they are both simpler to learn on than a mill, as they aren't complicated machines, but they are much more dangerous. The market that Grbl fosters to is the CNC novice, which small scale mills are much safer to learn on with their low power, high RPM spindles. Crashes on a mill will likely only involve a small broken bit and maybe a bandaid. On a lathe, a workpiece could be flung across the room and a hole in hopefully the wall, rather than you. 

That said, if the motion controller isn't rock solid and has weird race conditions, unlike some other embedded CNC controllers, this doesn't matter that much on a smaller mill, but has huge implications on lathes. I think you can use Grbl on lathes now due to its stability, but it's more a matter of time before I can get to installing spindle sync. And trust me, I want this as badly as you do.

109JB, 2015-02-20:
I completely understand the priorities.  Really was just checking to see if it had been though about.  All of the developers on GRBL are doing a great job and I want to commend you with how far you have brought GRBL.

cri-s, 2015-02-20:
The sensor for lathe index wheel should go to input pin that can be used to externally increment timer
without needing interrupt. In this way it could be used as speed sensor too for mill in order to
do closed loop on pwm output.

closed 2015-02-22 by chamnit

tklus, 2017-05-19:
has this been thought about anymore? I wonder if we can implement it in the GRBL MEGA version. I just picked up a little lathe and would love to convert it to CNC but not being able to do threads is a big down side of converting to CNC at this point.

Thanks!
Tim

109JB, 2017-05-19:
I keep hoping.. Would love to get rid of the giant desktop computer that is hooked to my lathe right now.

tklus, 2017-05-20:
Is this something that should be suggested on the grbl mega git hub?

tbfleming, 2017-07-10:
I'm starting to look into this, but in a different direction. Instead of mucking with sensors I'm thinking of driving a spindle with a ClearPath SDHP. This would allow it to also function as a C axis for live tooling.

chamnit, 2017-07-10:
@tbfleming : The right way to add spindle sync is to have it closed-loop controlled, like your ClearPath SDHP solution. Driving it like an axis would be ideal, since you would have all of the sync data generated automatically with the planner. It would be the job of the spindle controller to make sure it can keep up with what the machine axes are doing. It's probably how industrial machines do it.

There are a couple of scenarios that would need to be addressed for something like this. Often spindle sync is much slower than normal non-sync because the spindle can react much faster than the axes, or vice versa. There will need to be separate modes to accommodate this. Also, you'll need some kind of feedback from the spindle controller to let Grbl know that it failed to sync to immediately throw an alarm. For example, a tap jammed during a tapping cycle.

chamnit, 2017-07-10:
That said, having an rpm sensor on an existing lathe spindle without a speed controller and having the axis motion control react instead is a lot cheaper and simpler to implement for most DIYers.

HuubBuis, 2017-09-14:
I did a CNC conversion of my 300 mm Lathe. I put a nema 23 stepper (57BYGH627) on the spindle using a 3:1 gearbox. The X and Z axis are done using a small nema 17 stepper (42BYGHW811). 3 TB6600 drivers and a GRBL controller. Works great Threading, parting, knurling (cut) radius turning etc.

GfreshUK, 2018-01-22:
@HuubBuis
Could you share any further information on this? I would love to convert my lathe to CNC but can't get my head round how I could cut threads using GRBL.

HuubBuis, 2018-01-22:
I put a stepper (nema 23) to drive the spindle. Using a lever i can chooze between drive by stepper or drive by the original motor. The stepper drives the spindle using 2 gears and the the original gear on the spindle. At first i made the gears of aluminum, but they make more noise than the original plastic (delrin) gears, so now is use the original lathe delrin gears. Even after 2 years of abuse, i haven't broken a gear and none is worn out. Using small stepper has some advantages.
Of course the torque is limited, but i do threading in brass, aluminum and steel without any problems. Its just a matter of selecting the right (low) speed and (small) cutting depths. However, the next time i would select a smaller gear at the spindle stepper te get more torque.
To reduce noice the Y-axis stepper runs at 16 micro steps, the X and Z at 8 micro steps. 
The spindle is controlled by the GRBL Y-axis set at 20 steps per mm. My software treats 1 mm as 1 °. For threading it is just a matter of calculating the amount of degrees to turn and let GRBL do the job.
In stead of ball spindles i implemented back lash compensation so i still can do conventional turning. 
For the records, i have a small chinese lathe HBM300 without a gearbox (Norton). That means spindle speed may vary and "low" torque at low speeds. On the other hand, when things go wrong, de damage is less.

I included my drawing for the spindel (C-axis made using progecad) and a svg file. I hope it will help you.
Any questions, please ask.
[SpindleMotor.zip](https://github.com/grbl/grbl/files/1653784/SpindleMotor.zip)

X3msnake, 2018-01-23:
What did you use for backlash compensation?

HuubBuis, 2018-01-23:
The back lash compensation is implemented in my software (Windows 10 app).
When I set (adjust) the start position (X,Z,C) by moving from the "Home Position" to the "Start Position", the backlash compensation is set to zero in that direction of movement. From then on, at every direction **CHANGE**,  first a move to compensate for back lash is executed, after that the actual move is executed. Be aware when using G02 or G03 (Arc move), Then a change in direction takes plase at every 90° angle so you may have to split that move.

X3msnake, 2018-01-23:
What's your app written in? C?

Did you share the source code on Github?

HuubBuis, 2018-01-23:
The app (CNCL) is written in C# using visual studio 2017.
The code is not shared but you can download the app from the Microsoft Store and try it out for a month without any limitations.
I hope i do not violate any rules by showing a link to the app! 
https://www.microsoft.com/store/apps/9P42TB5T697H

X3msnake, 2018-01-23:
It's a pitty that you do not share the source code tho.

For software related to GRBL I'm only interested in open source code.

Thanks for the info anyways.