grbl/grbl #603: Strange square

dorucazan, 2015-02-19:
Hi,
I have used this test gcode to 'draw' a rectangle:
<i>
G21
F5
G90 G94 G54
M03
G01 X10 Y0 Z0
G01 X10 Y10 Z0
G01 X0 Y10 Z0
G01 X0 Y0 Z0
M05
</i>
After moving in XY plan for first and second sides, on the approximately 1/3 of 3rd side the tool started to descent (see attached picture). I am using Arduino Duemilanove with 328 processor and grbl 9. Is it something I did wrong?
Thanks a lot,
Doru Cazan

[image]

chamnit, 2015-02-19:
@dorucazan : That seems strange. Is Grbl actually moving the machine like this or this is a problem with the GUI that you are using? GUI's sometimes use their own g-code parsers and it may not resemble what Grbl actually does. Secondly, try downloading the source from the edge branch, Grbl v0.9h. There was a critical bug fix for particular motions, but this isn't one of them. It doesn't hurt to try though,.

dorucazan, 2015-02-20:
I will close this until testing new version. Thanks, Doru Cazan

closed 2015-02-20 by dorucazan