grbl/grbl #604: Suspended state and EStop handling

poelstra, 2015-02-19:
On my machine, I have an electrical circuit with a relay and physical buttons to disable my stepper drivers. The state of that circuit is also fed to an input pin on my controller (previously LinuxCNC, now Grbl). On LinuxCNC that pin was EStop, on Grbl it is reset/abort.

While EStop is active, LinuxCNC knows that it cannot do anything (and should abort any pending operation).
If the EStop pin becomes inactive, it still won't do anything, until I press the cycle-start button in (green button in Axis). This button is disabled as long as EStop is active.

I'd really like to have this functionality in Grbl too, and I think the Suspended state as proposed in https://github.com/grbl/grbl/issues/597#issuecomment-74885187 adds the foundation for it.

A few additional (small) things would be necessary (which I'll happily implement myself, if people agree):
- Add (compile-time?) option to always enter Suspended state on startup/reset
- Add (compile-time?) option to prevent cycle-start while reset/abort is active
- Add (compile-time?) option to not reset Grbl when reset/abort goes from active to inactive (only from inactive to active)

This way, if EStop (wired to reset/abort) becomes active while running, then Grbl will reset, thus re-entering suspended state. Cycle start is prevented until EStop is cleared again. Exactly the workflow I had, and really like :)

Please let me know what you think!

ashelly, 2015-02-19:
How is this EStop feature different from the Safety Door feature on the edge branch?

chamnit, 2015-02-19:
@poelstra : I get this question about once a month and more frequently recently. This is my stance.
- Best setup: E-stop should be handled by hard wiring to the Arduino reset pin (not Grbl soft reset). This guarantees power is stop and is not subject to coding errors/bugs.
- Alternate setup: E-stop should be handled by the GUI itself. Should a GUI implement an E-Stop-style feature, it should send a soft-reset and immediately cease all communication with Grbl. The soft-reset will kill all power and motion, while it will also clear the planner and serial read buffers. If it ceases communication, Grbl will just sit happily idle.  (And perhaps send the newly proposed $DIS stepper disable after as well.) 
- Safety door switch is different from an e-stop. Door switch will do a controlled feed hold and then allow you to continue the queued motions. E-stop should kill power, kill the job, and prevent anything from happening until the user has acknowledged it. It's assumed that the e-stop will force you to re-setup the machine because of the harsh nature of it.

poelstra, 2015-02-19:
@ashelly : @chamnit explains it perfectly (third bullet) :)

@chamnit : I agree that EStop should primarily be handled in hardware (which is indeed what I have done). I disagree though, that wiring it to the Arduino's reset pin is the best setup.

I actually think that wiring EStop to Arduino's reset is even a bad idea, because while it is in reset, I cannot verify whether it actually booted correctly, has the correct configuration, communication with the GUI is as I expect it to be, etc.

The only way to verify that, is to release my hardware EStop signal, which will then power up my drivers and simultaneously boot the Arduino, which will have a (brief) period of unitialized output pins, and may even appear to have been flashed with the wrong firmware/settings...

I do agree that it's very useful if a GUI also knows about this behaviour, and can e.g. keep buttons like Homing disabled until the $DIS mode has been released, but I wouldn't mind having to enter a $EN(ABLE) on its 'commandline' for the time being ;)

chamnit, 2015-02-19:
@poelstra : I think one thing that is evident is that there is a number of ways to implement an E-stop, based on their interpretations of what an e-stop should be. 

With the physical button problem you mention, I think this is a non-issue. This is the standard behavior of Grbl when it powers-up. You'll already have a procedure existing that will handle this. It not unrealistic for an e-stop to exhibit the same behavior.

While not everyone will wire in (or know how to) a physical button, the best I can do is provide the tools to implement one on the GUI side. Let the GUIs interpret their own version of e-stop behavior. Each machine is different and may require this type of flexibility.

In the end, Grbl isn't an all-in-one replacement for LinuxCNC on its own. It requires a GUI, and writing one is fairly simple. IMO, Grbl should do as much as it can to stay flexible and let the GUIs determine specific behaviors like e-stops and even tool changes.

cri-s, 2015-02-19:
For HW Estop on Reset pin, pull-downs are needed , or pull-up in case
of inverted step pins.
for better E-Stop Hw should be used. You could switch the pulse line
from microcontroller
to pull-downs where Ne555 is connected at low frequence and momentany
toggle switches
that are used  to move cnc out of switch if you don't want use hobby
type bypass e-limit
switches that disconnects the limit switches from the e-stop buttons.

poelstra, 2015-02-19:
@chamnit Yes, there certainly is a wide range of machines and functionality already supported by Grbl, and I think you're doing a great job striking the balance between all of these!

I've been pondering some more, and I agree that most of what I'm looking for could indeed probably be done by a GUI (e.g. simply not sending motion commands as long as 'EStop' is active, and always sending a $DIS when a Grbl reset is detected), although firmware support from Grbl would certainly help for those cases for GUIs that don't support that, and to make it more reliable.

As said, I'm happy to provide a PR with a proposal for the changes I suggested. All three compile-time options only, and I think all of them not being much more than a different initialisation value or an extra 'if'.
If you won't even consider merging it, it's no use to do it of course, otherwise I'll give it a shot.

@cri-s I don't have my limit switches connected to my EStop in any way, and I'm not trying to use hard limits as EStop.So I'm afraid I can't follow you?

chamnit, 2015-02-19:
@poelstra : I'm not sure if I agree about the firmware support from Grbl would make GUIs more reliable. If we install some of your optional suggestions, this means that GUIs would have to account for these optional conditions to work correctly. There would need to be some way to tell the GUIs what the build configuration is that its dealing with. 

On your three suggestions:
- Suspend upon startup-reset: It's probably a good idea, but this would require all GUIs to account for this state. It's probably better to try to use an existing state or mechanism to be backward compatible.
- Prevent cycle start during reset/abort: Can't do that. Grbl always executes everything its sent if it can. This was changed recently in v0.9h. It depends on the GUI to manage the 'cycle start' by streaming. This was done to greatly simplify Grbl's state machine and to be able to be agnostic to streaming and jogging commands.
- Not reset Grbl when reset goes from active to inactive: I'm not sure what you mean by this. IMO reset should ALWAYS reset Grbl, so that there is no confusion on it's operation.

I think we could use one of the exist ALARM mechanisms to enact a software-based E-stop. Like @cri-s mentioned, hard limits is a way to do what you want the e-stop. When triggered, it completely stops Grbl from doing anything until you send a reset command. Perhaps, there can be a real-time command like a 'Ctrl-E' or 'Ctrl-Esc' to trigger this same stop/reset mechanism.

cri-s, 2015-02-20:
If you only have E-Stop, and want that gui should recognize E-Stop, just connect DTR or any
other serial free status line to E-Stop.

For firmware support, it is possible to go into alarm when reset is detected as boot reason for avr.
If you have homing, it  is pretty useless, as already in alarm mode after powerup.

poelstra, 2015-02-20:
@chamnit Thanks for your thoughts! My reasoning on the reliability was that GUIs already 'need' to handle states like the initial homing alarm on startup, safety door, and the $DIS state you proposed. The only thing the optional condition does is change when such modes become active, which is something the GUIs can then automatically handle. So I think everything can be implemented in a way that is transparent to any GUI, given that the GUIs correctly handle the state (idle, alarm, disabled, ...).

* Suspend upon startup/reset: I definitely agree that we should use an existing state for this, where 'existing' might be 'new' if I understood you correctly that $DIS would become a new state?
* Prevent cycle-start during reset/abort: Hmm, I thought that you mentioned you wanted to prevent any motion while in $DIS state, which is why I thought this would be easy (once $DIS is implemented)
* Not reset when reset goes from active to inactive: currently, if the reset/abort pin (so not Arduino's reset, but Grbl's reset) is 'triggered', it always performs a soft-reset. I.e., if I make the pin low, it soft-resets, if I make it high, it soft-resets again. I think it would be more useful if it would only soft-reset on one of the 'edges' (configurable, like invert-logic for limit pins), such that I can 'prepare' it using the GUI while EStop is active (I still cannot leave $DIS mode though), and be sure that it won't 'glitch' or reset any of the settings I just made when I then release EStop. For example, Chilipeppr pops up an "Grbl has been reset" message, which 'feels' like something went wrong, whereas they actually went better :)

And yes, the current hard-limit behaviour in a way works a bit like what I envision, but it differs in that it completely 'locks up' Grbl: I cannot ask for settings, etc. I guess that if the hard limit alarm mechanism would still allow 'normal' interaction (but prevent movement, of course) this would probably also do what I'm aiming for.

Adding a Ctrl-E or Ctrl-Esc is certainly a good idea too, as it would indeed differ from the existing Ctrl-X in that Ctrl-X could maybe leave steppers enabled, and Ctrl-Esc could also disable them, just like a 'hard reboot' would do. I'd probably always use my hardware EStop button for that though, but it could be useful for people that want EStop, but don't have a hardware circuit for it.

@cri-s Well, I have both hard limits and EStop. I guess it's debatable whether triggering a hard limit should also force EStop 'mode', but the way I have it wired, they cannot be exactly the same: when my EStop is active, I don't have power to my drivers, but I 'need' power to them to drive them off the hard limit ('force limit override mode'), if the limits were hit before.
And I don't think it's useless if I have homing (which I do have), because one of the reasons for all of this discussion, is that it's currently possible to issue $H to get out of that initial alarm, while my EStop is still active. Grbl will happily start sending pulses (which do nothing because the drivers are physically disabled). It'd be better if Grbl would simply refuse any motion, directly indicating what the problem is. It could be done with 'just' a GUI, but having support in Grbl will automatically make it work (well, prevent it from working :)) in all GUIs.

Please note that I really appreciate all of your input on this! So please excuse me if some of my remarks appear harsh due to e.g. bad wording, they are in no way intended to be so :)

chamnit, 2015-02-20:
@poelstra : During an e-stop (or hard limit), what is the purpose of interacting with Grbl? It's an emergency state, there shouldn't be anything else occurring.

As for $DIS, true, it may be an additional state, but it's something that is fairly transparent to the overall operation of Grbl. I'm still thinking it through on how to implement it, keep things backward compatible for GUIs, and making sure it operates correctly through all scenarios. Right now, it's only a proposal.

natevw, 2015-02-20:
In my situation (xref: https://github.com/grbl/grbl/issues/580) the parallel port driver box that came with my machine includes an E-Stop switch that does not cut power to anything. I understood your position on this, @chamnit's position on this, and would like to some day figure out how to make such a mod to the box…but for now I am trying to leave its circuits as-is while still having some measure of "panic button" enabled.

My problem is still that I can resume Grbl while the button is still latched, which often creates a bad situation *the next time* I may need it. Perhaps my use case here is simply inexperienced/naïve, but so far the times I've used it were not such emergency situations where losing steps was fully warranted — just wanted to revise part of the clamping or double-check this or that before proceeding.

Adding a physical feed hold button may really be more of what I'm after, but even if that takes care of my use case, it still bothers me there is no way the software (either Grbl or the PC) can help me remember to pop the panic button back out for next use.

As I am writing my own interface for Grbl control of this machine (mostly for the experience), I wonder if (still barring any other solution) you could simply make the pin state exposed — then my UI could refuse to send any $X until that is resolved, without any changes you see as safety related?

poelstra, 2015-02-20:
@chamnit True, at the moment the EStop happens, there's an emergency state. So I then fix the problem, and (soft-)reset Grbl (Ctrl-X). No real trouble there in the current situation, although I think it'd be nice that a GUI can always request what the current status is. E.g. if I decided that the GUI also got in a bad state (could be the reason for me hitting my EStop), I first want to reload that, and it somehow needs a way to ask for the current state (like it currently also can when it is e.g. waiting for homing to be performed).

But then, after Grbl is soft-resetted, my physical EStop is still active. At this point in time it's probably no longer really an 'emergency', so we could call it a normal 'stop/disabled' state. I then would like to be able to communicate with Grbl (e.g. to verify that the GUI indeed sees the correct state, i.e.: disabled/EStop'ish), so it is now safe for me to remove the EStop by pressing my physical Start button, thereby giving Grbl control of my steppers.

Note that at that time, the hard limit switch may still be active (even though I released EStop), and I would like to have the possibility to jog the steppers out of the hard limit. This should of course only be possible by a manual override, like entering $X. LinuxCNC has an 'Override limits' checkbox that allows you to do one jog operation, if the hard limit is still present, it enters its own hard limit mode again.

I know $DIS is still a proposal, which is why I'm adding this scenario as input to the discussion, especially for the case when it does need to become an additional state anyway :)

chamnit, 2015-02-20:
@poelstra : Wiring your physical E-stop button to the Arduino reset pin will accomplish everything you want. When you release, Grbl reboots to it's power up state. There is no potential issue of Grbl operating while your physical E-stop js active as you have brought up as a potential problem.

A software E-stop is much trickier, as your long posts illustrate. There isn't a clean simple solution and it will likely involve a lot of changes in how Grbl operates. You can still have GUIs do exactly what you need on their side. The only time Grbl doesn't report it's status is when there is an alarm, and Grbl clearly states that there is an ALARM and what caused it. I find no issue there.

I think you're overthinking this, and I really don't see much of a problem. Please keep in mind the Grbl is not LinuxCNC and vice-versa. Grbl has to make careful choices since it's only half of the CNC controller. The other half is the GUI, and like I said many times before, a software E-stop should really be a GUI protocol. To help clear things up, perhaps a Wiki description on the recommended way to write a software E-stop should be posted.

As for the hard limit issue, you can disable hard limits via the '$' setting and then jog the machine. Reenable them afterwards. Personally, I think it's dangerous to give users the power to move the machine when a hard limit is engaged. It's better to have them manually deal with the problem, since it's impossible for Grbl to know the situation on that particular machine. If someone constantly hits hard limits, then they should really re-assess how they use their machine. 

@natevw : I hear you about the feedback issue with a physical E-stop switch, but an E-stop really shouldn't happen often and you can always wire in flashing lights to indicate that emergency situation. Having Grbl provide feedback when it detects an e-stop would be nice, but it's really not necessary, especially considering we'd have to dedicate an input pin for it.

skrutt, 2015-02-20:
@poelstra I'm not sure I understand why you aren't just using the feed hold and resume inputs that we already have, why not make a grbl pendant (maybe like this one https://lunaticcharade.wordpress.com/2014/11/15/grbl-pendant/ )

And then wire the estop to spindle & motor power? This way it seems to me you could do everything you want, pause the machine, resume, estop, change settings with power off?

poelstra, 2015-02-20:
@natevw Didn't see your post, sorry. It seems you have (almost) the same use-case as I do.

@chamnit Yes, way too many words for what I thought was a small change.

@skrutt Because feed hold is a controlled stop (maintaining position), whereas my EStop physically removes power, and thus position is lost, and because a GUI shouldn't be able to issue resume as long as the 'hold' (EStop) is active.

I think that most (all?) of what I want can be achieved with a few very small changes, and likely won't even require changes to existing GUIs (though user experience may be better if they actively understand the behaviour).

A PR probably says more than a thousand words, so less talking and more coding (from my part) ;)

chamnit, 2015-02-20:
@poelstra : All is not lost. I'm still mulling possible solutions.

- Perhaps all we need is an option to always boot Grbl into an Alarm state upon power-cycle. This covers hard-wired e-stops, if you don't have homing enabled (as this triggers an alarm already). 
- And hard/soft limits don't halt Grbl, but just place them into Alarm mode that suspends rather than locks out. So you have feedback control. 
- E-stop real-time control character that GUIs can trigger this Alarm with suspend mode.

Overall, I don't think this will go into v0.9, but rather v1.0. Let keep this conversation going and try to come up with a simple elegant solution that can be either backward-supported or easily updated for current GUIs.

natevw, 2015-02-20:
> I hear you about the feedback issue with a physical E-stop switch, but an E-stop really shouldn't happen often and you can always wire in flashing lights to indicate that emergency situation.

At that point, if I'm opening the box back up and rewiring I should probably just analyze if I can somehow cut at least spindle power while the E-Stop button is latched. (And while I'm at it [see what my stepper enable situation really is](https://github.com/grbl/grbl/issues/589)…)

What I want is the ability to do this as "software flashing lights".

> Having Grbl provide feedback when it detects an e-stop would be nice, but it's really not necessary, especially considering we'd have to dedicate an input pin for it.

I'm a little confused what you are saying here? I wired my E-Stop to pin A0 for `PIN_RESET`, and it behaves as I would expect…when I mash the button (the *first* time ;-) then I get an alarm in Grbl etc. Could that existing pin's state be exposed via something like `BITFLAG_RT_STATUS_RESET_PIN`?

poelstra, 2015-02-21:
@chamnit Yup, that's exactly what I was thinking! (Plus only triggering abort/reset and limits when they go from inactive-to-active, and preventing exit of the alarm state as long as abort/reset is active.)

skrutt, 2015-02-21:
@poelstra Okay, i understand more what you want to do now and why. And i can see that i would also like something on those lines where pressing hold will inhibit all motions until released. For my machine   though i would still want this to be split into two buttons, where one could be hold, and the other be only for emergencies and cutting power. 
I can see why you feel reluctant to rewire anything because I have not completed all the wiring myself either, where getting and wiring in an estop is actually on my todolist still..

I also now realise that i thought you also wrote @natevw s post,  where wanting to stop the machine to check if everything is alright is a more typical case for using the feed hold :)

poelstra, 2015-02-21:
@skrutt Hehe, yeah, @natevw's and my use-case are similar, although Nathan also described more of a feed-hold use-case. That's not really what I'm after though, as I think that's already handled by the existing feed hold/cycle start and safety-door feature (though I must admit I haven't used that myself yet).

chamnit, 2015-02-21:
@poelstra : Sort of like what you suggest. :) E-stops are still wired to the Arduino hard-reset line, not Grbl's soft-reset.

- A compile-time option to force an alarm upon power-cycle or hard reset, but not a soft-reset. This is already the behavior for Grbl when a user has Grbl's homing enabled. So this shouldn't be a problem. The thought of NOT having Grbl automatically alarm (without homing) when a user powers-up was to make it absolutely simple for new users that are trying out Grbl for the first time. If you don't have a machine with a hard-wired E-stop switch or homing limit switches, you'll never need this auto-alarm. Adding a compile-time option for this for certain machine and types of users would probably be a good idea.

- I specifically stated NOT soft-reset with this auto-alarm upon power-up. Soft-reset is not intended to be used as an E-stop, and I would argue it shouldn't be, because it's over-complicates its usage.

- No triggering only from inactive to active. There are long issues posts on this topic, related to hard limits. In short, you can't guarantee Grbl will read the switch correctly, if the switch is bouncing or there is line noise. There have been tests to verify this behavior. Only if you have a 100%, robust, filtered input is when you could do something like this. Nearly all users will not. Hard wiring the e-stop switch to the Arduino hard reset solves the problem with reset, but hard limits are still a little annoying. However, there are work arounds.

- On preventing alarm exit upon reset is active, this is solved by hard wiring the Estop to the Arduino hard reset pin, not the Grbl soft-reset. Again, soft-reset should only reset Grbl, not manage an e-stop. In addition, soft-reset isn't always wired to a physical switch and works with real-time commands. If you don't have the switch, the operation of this prevent alarm while reset is active is undefined. Again, the solution here is to have the GUIs manage their versions of what an e-stop should be with the available tools. 
- It may be a good idea to add a force alarm that behaves like the hard limit alarm for GUIs to add to their toolbox. In this case, a soft-reset command would exit this, just it would in the hard limit case.

poelstra, 2015-02-21:
For reference, this is what [LinuxCNC's Integrator's Manual](http://www.linuxcnc.org/docs/devel/pdf/LinuxCNC_Integrator_Manual.pdf) (page 54) has to say about it, which was the inspiration for wiring it that way on my machine:

> This assumes an ESTOP switch connected to pin 01 on the parport. As long as the switch will stay pushed, LinuxCNC will be in the ESTOP state. When the external button gets released LinuxCNC will immediately switch to the ESTOP-RESET state, and all you need to do is switch to Machine On and you’ll be able to continue your work with LinuxCNC.

chamnit, 2015-02-21:
@poelstra : Yes. First, Grbl is not LinuxCNC. Second, you get this exact behavior with wiring the e-stop into the Arduino hard reset pin.

chamnit, 2015-02-21:
@poelstra : In addition, if Grbl is not responding to status report real-time commands and has not notified you of an ALARM and the causation, then the GUI can assume Grbl disabled for some reason. Like an E-stop. The GUI simply needs to state, "Check Grbl" or "Check E-stop".

poelstra, 2015-02-21:
@chamnit Yes, Grbl is not LinuxCNC, but you did a very good job at being compatible with a LOT of concepts of it, and you're attracting quite a few ex-LinuxCNC'ers nowadays :)

I see your point about keeping things simple for users with 'simple' machines.
I also like the simplicity of wiring EStop to Arduino's reset, but I really disagree that it's the exact same behaviour as what LinuxCNC does.

Now, I do start to see why our opinion on this may be different. I guess I see Grbl as more than 'just' a step-pulse generator. To me, it's also managing the software-part of the overall state of the machine. I see Grbl GUI's more as 'views' onto that state. So that would include knowledge about power up -> enable steppers -> oops, EStop -> EStop released -> re-enable steppers -> etc.
Maybe that's indeed because I have that LinuxCNC picture in my mind. To make an analogy, I see Grbl as the realtime-kernel-driver-part of LinuxCNC, and a Grbl GUI as e.g. Axis (one of LinuxCNC's most popular GUI's).

To my feeling, wiring EStop to Arduino's reset, would be similar to wiring EStop to my LinuxCNC PC's reset button (if PC's still had such a thing ;)). That is, if I hit EStop, the complete PC goes blank, until I release EStop, after which it will slowly boot up, emitting a few glitch pulses on its parallel port, and then presents its GUI again.

I don't think that a soft-reset should ever clear a 'hard limit' type of alarm, even if it was only activated by a GUI command. Reason is that the soft-reset is also necessary to 'resynchronize' Grbl's "line -> ok" protocol. So every GUI should always issue a soft-reset in order to flush any pending lines that were present due to previous GUI communication. I didn't really think of this before, and it indeed makes it debatable whether disabled-mode (Alarm state) should be entered upon a soft-reset like this.

Does it make sense to have things like this?
* Ctrl-X soft-resets, but never changes states, it's basically necessary for GUI's to start with a known state of the line-based protocol
* Hard limit causes a soft-reset into Hard Limit Alarm state (as you proposed)
* Reset/Abort pin becoming active causes Grbl to soft-reset into Abort Alarm state, and cannot be left as long as Reset/Abort is 'active'
* Reset/Abort pin becoming inactive causes transition (not soft-reset, to not disrupt GUI protocol etc) into Disabled Alarm state. This is the same state as we currently have, where we need to enter $H / $X
* Entering $H first homes, then enters Idle state, and so does $X
* Power-up boots into Abort Alarm state, or Disabled Alarm state, depending on state of Reset/Abort pin
* Ctrl-Esc soft-resets into Disabled Alarm state

Of course, the Abort and Disabled states only make sense when people have limit switches and/or EStop/Abort/Reset lines. We already have a define for the "$H / $X needed"-mode, so we'd need another one for whether people have Abort, and whether it's active low or high.

Concerning not missing e.g. a hard limit on a pin-change: that's a good point, and I indeed read the discussions on it in https://github.com/grbl/grbl/issues/542#issuecomment-72870986.

I think that that race-condition is not really a problem in practice though, because as @matthijskooijman also noted, if people have bouncing lines the switch will 'eventually' permanently enter its active state, and pin-change interrupts will keep occurring for each bounce, so it will always eventually be picked up.
Not really a problem, given that people have an electrical problem anyway. They have noisy lines, so they also have 'noisy' delays until the limit is detected.
Besides, even in the current scenario, the problem is technically still there: if the spike is short enough for the pin-change to trigger, but to have been cleared by the time we read it in the Arduino, it can also be short enough to never even trigger the pin-change interrupt of the Arduino at all.

Having no unexpected resets of Grbl when leaving an Abort (and clearing a Hard Limit) easily wins over having Grbl detect a noisy limit a few ticks earlier, IMO.

I hope to have some time tomorrow to experiment with the code a little, to see if my ideas would actually work.

chamnit, 2015-02-21:
@poelstra : I think you have the purpose of soft-reset a bit confused. It's main purpose is to provide a way to reset grbl to it's default condition without having to power-cycle. This also happens to clear the buffers so streaming can be sync'd.

Soft-resets NEVER clear a hard limit alarm, it only exits the suspension of tasks to force the user to acknowledge the problem. Afterward Grbl resumes into an ALARM state. This blocks all g-code commands, but allows users to send a small set of '$' commands to do what they need to, should they need access to it. The '$X' is used to abort ALARMs only. Soft-resets should never release an alarm. Only '$X' and '$H' homing do.

In addition, if you soft-reset Grbl while it's in motion, it will ALARM when you do this, because the user forced Grbl to immediately stop and Grbl can no longer guarantee position (potential lost steps). This is the only time a reset will cause an alarm, and it will most definitely stay that way.

I also think you are missing the consideration of users will not always use the soft-reset pin and that there is a real-time character command that can be sent through the serial protocol to enact it. If you try to force Grbl to not release if the switch is active, what do you do when there is no switch at all? You have to contend with two different scenarios and this makes things needlessly complicated.

I disagree with your PC analogy, because you aren't turning off the PC and the Grbl GUI. That is a bit extreme. Rather you are disabling the LinuxCNC realtime kernel per your analogy. Again, I see no problem with using the hard reset, since users will already have a power-up sequence that deals with the situation.

I highly recommend that you review the ALARM functionality and how the soft-reset interacts with it in the source code over the remainder of the weekend. And then come back and talk about this more. We've wasted too much energy and time on this already.

For now, I still think Grbl has an adequate E-stop solution with the hard reset pin for those that actually wire in a physical e-stop, and have GUIs simulate one. We cannot ask GUIs to suddenly change everything for Grbl v0.9.

Later for Grbl v1.0, in a year, I think we can come up with an agreeable solution to make it easier for GUIs to know what Grbl is doing during an e-stop. This will give them time to catchup and buy-in. But for now, I don't think your proposals will work under the current conditions, especially when you have to deal with the marriage of a physical e-stop and a GUI-based e-stop and how they work together, while keeping it simple as possible. 

My gut feeling is that Grbl needs to have a dedicated e-stop pin to handle situations like yours, but we'll have to decide if it's really important to use up valuable pins for this. ALARM handling could be slightly tweaked to provide better event insight for GUIs. 

The bottom line is where do you draw the line in the sand? What does Grbl do and the GUI do? It's clear we disagree on this. My overall philosophy is the more Grbl has to manage, the less flexible it becomes. If we can't come to any agreement, you can always fork and do what you want, as many many others have done. Grbl will keep going and stay to its core competency of concentrating on the motion control.

poelstra, 2015-02-21:
I know what soft-reset currently does, and my response to not let it change alarm states was to your remark:
> In this case, a soft-reset command would exit this, just it would in the hard limit case.

I already said I'd stop talking and start coding, but then you said
> Let keep this conversation going and try to come up with a simple elegant solution that can be either backward-supported or easily updated for current GUI

and actually came up with almost exactly what I wished for!

I'm confused, and I'm sorry if I've wasted your time.

gerritv, 2015-02-21:
@poelstra 
Let me know what I need to add to GrblPanel once this settles down :-) Might even try a Dutch translation as well.

I look at Grbl and Gui in same way as you.
Gerrit

chamnit, 2015-02-21:
@poelstra : I'd like to figure this out, but we have to tread carefully. I disagree with the exact implementation of your proposal and some of the specifics. But, I think there is a way to do it with some tweaking of the ALARM handling, rather than using the soft-reset. I think we still need to keep this conversation going, but it was getting a bit intense for a relaxing weekend. I'm merely suggesting, at this point, taking a step back. Mull this over. Look at/hack the code. Let's revisit this after we've had a chance to let things sink in. Like I said before, we can't change v0.9, but will need to implement something for v1.0. We've got time.

chamnit, 2015-02-23:
@poelstra : Ok. I've had some time to let this all sink in and talked in detail with my machinist friend about all this.

- Grbl isn't LinuxCNC, but it does use the g-code descriptions, based on the NIST standard. The machine controller and behavior was modeled (in general) on production stand-alone CNC machines, like Haas and historical input from professionals I work with. Some of the decisions on how it behaves were based on that and how to maintain GUI control during certain states through only one serial communication line to Grbl. Everything has to go through it and still has to function. So, it may seem like Grbl is a LinuxCNC facsimile because it uses the same g-codes and definitions (which are also ISO standard), but it's not due to some of these restrictions.
- That said, it's clear that the ALARM handling in Grbl is bit convoluted and could be cleaned up, but there is a method to the madness in how it ended up this way.
 - Grbl has two ALARM states. A critical event ALARM and a normal ALARM. 
 - The critical event ALARM is something that freezes Grbl, like hard limits, with an event that is not synced with the GUI. When this occurs, the GUI may still be streaming g-code. If Grbl returns to its boot-up prompt, this streaming will cause Grbl to send back a whole bunch of "error:Alarm Lock"s to the GUI. Not exactly clean. So, instead, this freezing is there to force the GUI to acknowledge the critical event, dump all of incoming stream data (USB is not 100% synchronous), and let the GUI go into alarm handling. A RESET exit this critical event ALARM and brings Grbl into a normal ALARM state.
 - The normal ALARM doesn't freeze Grbl, but it allows prompt interaction, while locking out all g-code commands and most of the Grbl '$' commands. This allows the GUI or user some minimal interaction with Grbl to temporarily change some settings, like disabling the hard limit, to cancel the ALARM and manually move the machine via g-code commands, or run the homing cycle. A RESET does not cancel this alarm. It only re-initializes Grbl to its default state.
 - To clear a normal ALARM, Grbl uses the `$X` kill alarm or `$H` homing cycle commands. In retrospect, these commands shouldn't exist, but there needed to be a way for GUIs/users to interact with Grbl during a normal ALARM. So, `$X` was introduced. On a Haas, RESET should technically clear the ALARM state, if it's able to. I suppose, the `$X` could be a candidate to clean things up, but this would require a lot of refactoring of the Grbl interface, as I have mentioned several times already.

- This is what I can do now:
 - Install a compile-time option to always ALARM upon a power-cycle. This is default behavior when Grbl has homing enabled, so GUIs should be ok with this. This option covers machines or users that want this behavior, but, by default, this option will not be enabled to keep it absolutely simple for new users to get started.
 - Install a compile-time option to only trigger the hard limit ALARM when it is active. True that a bouncing switch may keep triggering the hard limit ISR and may eventually read the pin correctly. However, the same could be said when the hard limit switch is released, and Grbl will eventually read the pin as triggered again. I still see no real benefit for this, and highly recommend that users use soft limits instead. In practice, you should never hit hard limits. IMO, hard limits should behave harshly and with 100% guarantee that the machine will stop. I will write a comment about this in the config.h with this option.

- This is what I consider as potential refactoring candidates:
 - Soft limits shouldn't bring up a critical event ALARM, as the v0.9h edge branch does now. It should probably go into a normal ALARM, but the way Grbl's g-code parser is written and how it checks soft-limits makes this change a bit difficult. The problem is how Grbl should notify the GUI when this ALARM occurs so that it can cease the streaming with 100% certainty. It seems easier to keep it as is.
 - Since the RESET pin and the RESET real-time command enact the same command, I think it's not correct to use the RESET pin as the e-stop pin. This convolutes the RESET real-time command. It would be better to have a dedicated e-stop input pin to keep things separated and simple. The question is, whether it's really necessary to use up precious input pins for this, when using the Arduino hard reset pin does the same thing.
 - Critical event ALARM and normal ALARM handling: This definitely could use some work, but I can't think of a better way to do it at this point. This is something we'll have to mull over in detail. If we change this in any significant way, we'll have to give GUIs time to update their protocols to maintain compatibility. That said, this would be something that would be installed on a later version of Grbl, like v1.0.

Thoughts?

poelstra, 2015-02-23:
@gerritv Thanks, will do :)

@chamnit Thanks for your detailed reply! I've been playing with the code myself this weekend, and indeed ran into some of the issues you've mentioned, especially the implementation of alarm state handling differs from what I formed in my head but by looking at the behavior from the outside. I really wouldn't call it 'madness' though :) And I saw that you already did quite some refactoring to make things future-proof.

* Install option to always power up in alarm: I have implemented a compile-time option (which I've currently dubbed #define ABORT_LOCK) to prevent exiting alarm state ($X) and homing ($H) as long as the abort-pin is 'active'. It should actually be completely transparent to users if they haven't wired anything to the abort pin, and I think GUI's will also handle it without change. I'm going to do some more testing, and will commit it on my fork for you to review whether you think it makes sense.

* Install option to only trigger hard limit when active: implemented that too yesterday, and quick test seems to work fine. Appeared that I do have some bouncing on my limit switch, which sometimes is just enough to still trigger an extra hard limit when its actually moving off of it, so I'll install some filtering and do some more testing before committing.

Btw: scratch my remark about wanting active-only triggering of the abort/reset pin. It already works that way... Looking at the hard-limits code, I thought the same would hold for the abort/reset pin logic, especially considering that I also saw that happening on my machine (and never saw that before on LinuxCNC). Appears I have some bouncing on that relais-circuit :-/ So sorry for the noise on that!
I measured the bouncing to be ~1ms on my scope, so probably I never saw it because LinuxCNC has a 1kHz thread running for that logic (IIRC).

* Soft limit and critical event: yeah, I understand what you're saying there. It took me quite some thinking to (hopefully) make sure that the alarm mode is never exited as long as abort-lock is active. For abort-lock it actually appears to be simple (still need some testing ;)), but things will likely get more complex for other cases.

* Stopping streaming on error: I've thought about this too, have some ideas, but need to think it through a little better before I make a complete fool out of myself ;) I think that sending back a stream of errors wouldn't actually be such a bad idea though (after sending a status message indicating the alarm). The GUI could stop streaming as soon as it sees the alarm status message (and if it doesn't, no worries because Grbl will keep sending errors back). This will ensure that the streaming protocol stays in sync, without further resets etc.

* Reset pin behaviour: I've actually called it 'abort pin' in my comments in the code, to lessen potential confusion with Arduino's reset pin. I was indeed also wondering whether having a dedicated e-stop input would make more sense, and I think it kind-of does, in the sense that it's functionality-wise more similar to a hard-limit, and could maybe be better handled by the same code / pin-change interrupt etc. You mention that the reset pin and reset real-time command (ctrl-x) enact the same command. In my implementation, I've changed that, that if ABORT_LOCK is defined, the reset/abort pin now always (i.e. not only when motion was ongoing) triggers the CYCLE_ABORT alarm. So these resets then no longer are equal, but this seems not so bad?

* Critical event vs 'normal' alarms: currently, critical event shouldn't have to be a separate state, but could be OR of hard and soft-limit. Saves a state :) Other than that, I agree that to make this more robust, quite substantial changes would have to be necessary. It could very well be that not much will actually change from a GUI's view, but it'd probably need some serious testing, so probably not 0.9 indeed.

Some things that popped up when I was playing with the code (which is very well commented and structured, btw, thanks a million for that :))
* It may be a good idea to explicitly mark whether axes need homing. Would for example help to not have to home after a soft-limit alarm. Maybe even have a separate 'homed' bit per axis.
* I think there's a race-condition (two actually) in protocol_execute_realtime(), as more bits may have been set to one between copying e.g. sys.rt_exec_alarm and resetting its bits. It may be better to remove the bit-resetting at the end of the if(), and replace the copy with an atomic 'read and clear'.
* The handling of the control bits in the pin-change interrupt may be a bit tricky at the moment, because triggering a change in any of those bits, will actually only look at the 'first' active bit. E.g. changing the state of the door-pin, will trigger a reset instead if that pin is still high. It may help to install the same code there as what I'm using for detecting active-only hard limits.
* Having 'active-only' triggering logic can also help in making some of the other code a little bit more simple (and in some cases even more reliable), because it's then no longer necessary to 'manually' read the physical pins and applying the optional pin-invert, and prevents a race-condition between reading physical pins and whether it has been picked up by the pin-change interrupt. I guess it's more easily shown in code than words, so I'll see if I can make an example of it later.

Hmm, too many words again.
Anyway, I have some more testing to do :)
Will post back when I've committed some of my ideas.

natevw, 2015-02-23:
> The problem is how Grbl should notify the GUI when [soft limit] ALARM occurs so that it can cease the streaming with 100% certainty. It seems easier to keep it as is.

Could soft-limit alarm be a simple "error: command exceeds soft limit" similar to if I send unsupported or invalid command?

---

Thanks, btw for clarifying how I should connect an existing E-Stop button to the Arduino's hardware RESET pin rather than A0. I am working on a shield adapter for connecting Grbl to my sort of machine and I'm glad I became aware of this before getting too far.

chamnit, 2015-02-23:
@poelstra : Let me mull this all this for a bit.

@natevw : Soft limits with an "error:" response would be ok, but there are a couple problems with it. First, the main issue is that it's incompatible with arcs or any motion(canned cycles) that has multiple motions. You'd have to explicitly check that before moving on to the execution phase. Two, it's arguably better to enact an alarm for this situation, because some GUIs just ignore "error:"s. In this situation, if you continue the stream, very bad things can happen.

As for using the Arduino hard reset pin, let us know how it works for you. It's A solution, but not THE solution. That is what we're trying to figure out.

ashelly, 2015-02-24:
For what it's worth, here's the setup which is working for me.  I'm running a modified Grbl on a Mega. I have a normally-high E-Stop switch, which puts the driver chips into reset when it goes low, killing power to the motors.  I am monitoring that same signal on a spare pin, and when it goes low, I set the critical event alarm bit, just as if it was a hard limit.   
This seems to work perfectly to a) actually safe the motors,  b) notify the GUI that there is a problem, and c) prevent any further commands until both the E-Stop button is released and Grbl is reset.

If I didn't have the spare IO, I would probably re-purpose pin A0. I haven't found need for a dedicated hardware reset/abort line,  given the existence of the identical serial command and the Arduino board's reset button.

LotusPack, 2015-02-24:
Hi Sonny,

Speaking as someone who has been in CNC since the late 1970's!

It is a real shame your time is used to defend Grbl when there is no need
to.  The design is a near perfect fit for the requirement. 

All the functions you have created show a real solid knowledge of
engineering need for CNC using Grbl as the very key G-code interpreter
component of a suite of software and hardware needed for a complete system.

So I say; let us all let Sonny get on with the programming and stop
pestering him with trivial suggestions that he kindly takes hours to explain
and respond to.

Regards,

 

Lotus Pack

  _____  

 

<https://github.com/grbl/grbl/issues/604#issuecomment-75593893>  GitHub.
<https://github.com/notifications/beacon/AJvJ0AvGRalxXryclB7f9DIIgXAtOyOqks5
nu2ADgaJpZM4DikjT.gif>

poelstra, 2015-02-24:
@chamnit I've committed and pushed my proposals to https://github.com/poelstra/grbl/tree/wip/estop.

I've tested them using Chilipeppr, which accepts them without changes. Feedback on the abort lock is visible on its 'serial console widget' when it activates, but I didn't dare to spend more bytes on sending an "Abort unlocked" message when it deactivates.

I just noticed that you already also implemented some of it yourself, didn't expect that so fast! Thanks.

My implementations slightly differs from yours in that the abort lock option is 'transparent' when you don't have the abort pin connected, the hard limit edge triggering also handles the case when multiple switches are active at the same time (not a big deal for the hard limits, but the same trick may be useful for handling of the control pins).
I've also added the no-reset-on-critical-event option, which saves some bytes and should allow GUIs to still keep polling status (including limit pin states).

If you like, I'd be happy to provide a PR to your edge branch.

@ashelly Probably we had a similar idea then. ABORT_LOCK accomplishes almost the same.

@LotusPack I actually agree with you. It was never my intention for this thread to explode like this. I just expected to provide the 3 commits I just did, hoping to be of use to someone. Possibly later some other  'trivial suggestions', but only as "click to merge if you like them" PR's, indeed to let Sonny focus on the important stuff.

closed 2015-02-24 by poelstra

chamnit, 2015-02-25:
@poelstra : Thanks. I had a few minutes to review the code.
 - I'm starting to warm up to the idea of checking a pin before initializing upon power-up, but this doesn't capture a soft-reset very well. It depends on other state management for this to work effectively. However, at the moment, I still don't think we should be using the soft-reset pin for this, although it may save a pin. If I can't come up with an alternate solution, I may eventually give in to your demands.
- I don't think it's a good idea to automatically reset a critical alarm for the reasons I've stated already. Mainly it creates other problems, such as the stream not stopping on time and causing a lot of excessive feedback clutter when it doesn't. It's just as easy for a GUI to send a reset when it sees the critical alarm message "ALARM: xxx". If your concern is Grbl not reporting its status during a critical event alarm, it's really easy to have Grbl send these during this time. I think I will add this, so that the only time Grbl doesn't respond to status queries is when Arduino is actively disabled or turned off. This should make it easier for GUIs to know when to have the user check an e-stop or to see if Grbl is connected to the PC.

In general, these changes, outside your suggestions I added last night, are more patches to create behavior you desire. There is likely a better way to do this that does not involve GUIs needing to account for all these compile-time options. I think it'll be better to alter the protocol into a singular, standardized idea that can capture these different scenarios. This I will have to think about for a later version.

cri-s, 2015-02-25:
I'm was a bit absent for the discussion.
On Avr you could test the reset reason:
   unsigned char ResetSrc = MCUSR;   // save reset source
   MCUSR = 0x00;  // cleared for next reset detection

   ... // further initialisation

   if (ResetSrc & PORF)
      cputs("PWR RESET\n\r");
   if (ResetSrc & EXTRF)
      cputs("EXT RESET\n\r");
   if (ResetSrc & BORF)
      cputs("BOD RESET\n\r");
   if (ResetSrc & WDRF)
      cputs("WDT RESET\n\r");

This allows to test if it was SW-Reset (WDT), Hw-Reset, powerup or BOD
if enabled.

chamnit, 2015-02-25:
@cri-s : Interesting. Do you know if this reset source register is a standard feature on all MCUs? If so, it'd be a good way to have Grbl know what's going on to handle them correctly.

matthijskooijman, 2015-02-25:
@chamnit it seems this is a standard atmega features. A quick grep shows that almost every processor supported by avr-libc at least has a PORF, except these:

	matthijs@grubby:/usr/lib/avr/include/avr$ grep EXTRF --files-without-match $(grep \#include.*io io*.h --files-without-match)                                   
	io1200.h
	io2313.h
	io2323.h
	io43u32x.h
	io43u35x.h
	io4414.h
	io8515.h
	io8534.h
	io86r401.h
	iom161.h
	iom3000.h
	iotn11.h
	iotn12.h
	iotn22.h

Note that I excluded io*.h files that include other io*.h files, since
those will likely include the proper flags through that other file and
not define the flags themselves.

chamnit, 2015-02-25:
@matthijskooijman : Thanks, but I indirectly meant if ARM has it.

matthijskooijman, 2015-02-25:
Oh, right. No experience there :-)

cri-s, 2015-02-25:
Yes, this is a feature every MCU (Microcontroller) have.

Just tested it with STM32F103C8.

After power-on PINRSTF and PORRSTF  are set.
After soft reset PINRSTF and SFTRSTF  are set.

Test code:

// reset test:

if (RCC->CSR & RCC_CSR_PINRSTF)
    puts("pin reset\n");
if (RCC->CSR & RCC_CSR_PORRSTF)
    puts("POW reset\n");
if (RCC->CSR & RCC_CSR_SFTRSTF)
    puts("Soft reset\n");

// power on: make soft reset
if (RCC->CSR & RCC_CSR_PORRSTF)
{
    RCC->CSR |= RCC_CSR_RMVF;
    beep(800, 40);
    sleepms(4000);
    NVIC_SystemReset();
}

jaketeater, 2016-05-24:
This is an old discussion, but I had a question.  

What about using a relay so that when the EStop is pressed, one of the limit switches closes?  This would stop GRBL and inform a GUI of an alarm.

The text of the alarm would be incorrect ("ALARM: Hard limit"), but other than that, wouldn't all of the desired behavior be there?