grbl/grbl #608: Manual code changes needed on chrystal replacement (other frequency)?

morres83, 2015-02-28:
Hi,

I am currently implementing the GRBL-JOG of the German Make magazine (which was for GRBL 0.8) into the latest edge version of GRBL. I use the original Make pcb, but replaced the 16 MHz chrystal by a 18.432MHz one to eliminate the Baud rate errors.
As far as I can see, most timer things should work properly by the adjusted F_CPU (correct me if I am wrong).
Still, I am wondering if I have to adjust manually some things (I think of oscilloscope measured loop times and so on, if applicable)?

From my ocsilloscope I /think/ the step frequency could be a little higher than expected (not sure if its about exactly the 15% the uC runs faster now, but could be).

Thanks in advance,
morres

morres83, 2015-03-14:
Hi,
since I didn't get any answer for 2 weeks now, I am wondering if my question was not understood?

To be precise, I don't want to have the correct adjustments calculated for 18.432 MHz by anyone (I'll try this on my own). But maybe somebody has an idea, which timings are automatically correct by resetting F_CPU and where I have to do an adjustment by hand?

I know GRBL usually sticks to the standard Arduino, but maybe still someone can give a hint about this customization.

Thanks,
morres

P.S: Is anybody from Germany here who followed  (or even built) the GRBL-JOG project with the woodmill of the Make magazine (formerly c't Hacks)? I'd like to get in touch to exchange some experiences.

chamnit, 2015-03-14:
@morres83 : Sorry, you likely haven't got an answer, because there isn't one.I've never heard of the GRBL-JOG project. But, Grbl was written to be able to alter the clock frequency, but I've never tested it, or never had the need to. You're the first that I recall in the last four years that is attempting to. Theoretically it's possible to do, as you found by changing the F_CPU, but there is likely something else you need to change, like in the Makefile. I think you're on your own on this one, but let us know of your progress.

closed 2015-03-14 by chamnit

sid123q, 2017-02-27:
Hello. I am new in this community, but could i reopen this issue #608. As mentioned in issue #837 using  ch340 (and sometimes [unflashed](https://github.com/AlexHolden/Arduino/tree/master/hardware/arduino/avr/firmwares/atmegaxxu2/arduino-usbserial) atmega16u2) there is problem using 115200 boud rate. 
[image] So using 18.432MHz xtal there should be no problems with data transmission. 
I have personaly had problems both using ch340 and atmega16u2.

sid123q, 2017-02-27:
In fact if you use 18.432mhz xtal both last [Known-Issues](https://github.com/gnea/grbl/wiki/Known-Issues) would be gone. Using genuine and cheap clones with ch340. Or i am wrong?

109JB, 2017-02-28:
I doubt that you would solve the problems with the ch340.  The problem is in the firmware, which isn't available to modify. The 16U2 firmware is available and a modded firmware was produced to eliminate the problems with the 16U2.  Without the firmware for the ch340 available it can't be fixed. For a more detailed explanation of the problem see https://github.com/grbl/grbl/issues/845

sid123q, 2017-02-28:
@109JB ch340 is hardware based. No software changes is needed in ch340.
Simple math:
*16000000/115200=138,88889 (you can get an error, because there is slight chance to get UART signal in not right cycle) it has difference of cycle 2.1% +- 0.3% ch340 instability so it is more (in worse scenario 2.51%) than 2% atmega can handle. 
*16000000/250000=64 (yourself proved that work without error, no cycle diference)
*Someone else used  2Mb baud. So 16000000/2000000=8 (no proble)
*If we use 18.432MHz xtal we could use all standart baud rate. For example 18432000/115200=1600.
If you want to avoid error, you can use any baud that is xtal/baud=whole, or in theory with hither baud rates smaller value after comma (Residual don't know if it's right in English).

electrokean, 2017-02-28:
From my understanding the issue isn't purely baud rate error, but also something like USB packet implementation.

sid123q, 2017-02-28:
@109JB You can use for example boud of 100k with your made testing program and fake arduino with ch340, and you should get no error.

109JB, 2017-02-28:
I did get errors at all baud rates. Some had less, but all that I tested eventually showed errors.  You claim that 250000 baud should yield no errors, but in I stated this in one of my posts in #845 

> On my testing with the ch340g chip though it seems that running at 250000 results in fewer errors

Note that I said "fewer" errors, but there were errors. Even one error is too much in my opinion.  The errors encountered had to do with timing of packets received by the ch340 and status reports sent by the ch340 back to the host computer.  Bottom line is that it is something internal to the ch340 and can't be fixed.

sid123q, 2017-02-28:
@109JB You say

> I came up with the same conclusion as you. running a code with lots of short segments at high speed, or fast status requests, or a combination of these can result in more errors. On my testing with the ch340g chip though it seems that running at 250000 results in fewer errors whereas on my Mega board with the Atmel chip 250000 results in more errors. The errors I encountered on the ch340G @ 115200 is what started this whole deal to start with.

> Indeed, as I indicated in a previous post above, I can't get an error out of GRBL if I just stream G-code using send-response and turn off status requests. If I turn on the status requests I can get an occasional error. It is rather random, but it appears to me that slowing down the timed requests results in less errors, which would make sense.

when you get error using 250000 you heave been using **status requests** are you using 1start 8data 1stop bit, or using parity bits? 
You say that you can not get any errors if you use just sending g code. 
And as i understand you were using buggy version off @AlexHolden atmeaga16u2 software? After bug was removed  atmeaga16u2  version gets no errors, but not because there is no ERROR, but because it is same error(same difference), SAME xtal 16MHz for atmeaga16u2 and atmega328p.

this is my opinion. please correct me if i'm wrong.

Maybe you will be interested in [this ](http://wormfood.net/avrbaudcalc.php?bitrate=300%2C600%2C1200%2C2400%2C4800%2C9600%2C14.4k%2C19.2k%2C28.8k%2C38.4k%2C57.6k%2C76.8k%2C115.2k%2C230.4k%2C250k%2C.5m%2C1m&ubrr8bit=1&clock=18.432%2C16&databits=8)

sid123q, 2017-02-28:
Maybe CH340 gets more errors with 250k baud, because it uses [12MHz crystal](http://wormfood.net/avrbaudcalc.php?bitrate=300%2C600%2C1200%2C2400%2C4800%2C9600%2C14.4k%2C19.2k%2C28.8k%2C38.4k%2C57.6k%2C76.8k%2C115.2k%2C230.4k%2C250k%2C.5m%2C1m&ubrr8bit=1&clock=18.432%2C16%2C12&databits=8)  but it uses different architecture.

sid123q, 2017-02-28:
So, anyone could tell what i need to change in code to change crystal oscillator frequency?

sid123q, 2017-02-28:
@109JB could you share your testing program?

electrokean, 2017-02-28:
The FTDI FT232 used on most Arduino boards doesn't show this issue, and it doesn't even use a crystal.  That leads to the conclusion that the problem is most likely not baud rate error but a problem in the USB implementation.  I haven't looked into the 16u2 code to see what was changed to fix the issue.

In regards to using a different crystal with an Arduino, it is just a matter of editing the bootloader code and buring it, and creating a custom board definition for the Arduino compiler.  I've done this for a 20MHz crystal, and the code is [here](https://github.com/moorescloud/holidayduino) if it helps.

sid123q, 2017-02-28:
@electrokean and nothing else should be modified? No rise, fall times be affected?

AlexHolden, 2017-02-28:
With regard to the 16u2 changes: the original code was inefficient and buggy. At high transfer rates, the Arduino->PC buffer could overflow and corrupt the contents of the PC->Arduino buffer, typically resulting in part of a status response message being reflected back to grbl in the middle of a line of Gcode.

electrokean, 2017-02-28:
Thanks @AlexHolden - so nothing to do with baud rate errors, as I suspected.

sid123q, 2017-02-28:
@electrokean Yes. It nothing to do with baud rate error, as long as you use 16u2 with 328p, and same frequency oscillator.  As you can [see](http://wormfood.net/avrbaudcalc.php?bitrate=300%2C600%2C1200%2C2400%2C4800%2C9600%2C14.4k%2C19.2k%2C28.8k%2C38.4k%2C57.6k%2C76.8k%2C115.2k%2C230.4k%2C250k%2C.5m%2C1m&ubrr8bit=1&clock=16&databits=8) actual bit rate is about 117.647K for bouth 16u2 and 328p, so no problem. But if you use ch340  (acording [datasheet](https://cdn.sparkfun.com/datasheets/Dev/Arduino/Other/CH340DS1.PDF)) bit rate is 115.2k+-0.3% (from 114.8544 to 115.5456) on ch340 side and about 117.647K on 328p side. It's from 1.8% to 2.4% diference. recommended maximum difference,when U2X=1 and 8bits (no parity) is 1.5%.

Am i wrong?

chamnit, 2017-02-28:
All indications show that the Ch380 failure is related to internal buffer overflows when pushed very hard. Data chunks get lost, rather than individual bytes as you'd expect from timing errors.

electrokean, 2017-02-28:
@sid123q I see where you're going, but it seems so much simpler to just not use a CH340.  I guess if you really want to try prove the cause of the problem with the CH340 then it might be interesting.  As I said, you just need to build a custom ATmega boot loader (e.g. Optiboot) and Arduino board config file for the alternate crystal frequency, burn the bootloader, and solder in the crystal.  Not much effort.

When using an FTDI chip, you get pretty good baud rate accuracy - I'd say as good as the CH340.  And even though the FT232 doesn't use a crystal, it has an internal PLL that synchonises with the USB clock, and also allows sub-integer divisors.  And as far as I know no one has this problem on Arduinos with FTDI chips, nor do they have to adjust their baud rate to compensate for the ATmega baud rate division error.  As much as I dislike FTDI pricing and their approach to fake chips, they have produced some very useful and pretty reliable products.

So that is why I say that this is most likely a CH340 USB implementation bug, and not a baud rate error issue.  And as @chamnit says, the way the problem manifests also supports this.

109JB, 2017-02-28:
> @109JB could you share your testing program?

Sorry, but I don't have the code for the test program any longer.

I used 8 data bits, 1 stop bit and no parity.

> All indications show that the Ch380 failure is related to internal buffer overflows when pushed very hard. Data chunks get lost, rather than individual bytes as you'd expect from timing errors.

I think @chamnit comment above is a very likely possibility.  The bottom line is that when you push the ch340 chip too hard with status requests and fast g-code streaming that not all the data gets through to the 328P.

Incidentally, way back when I did that other testing I did several tests of just the ch340.

First was a loop-back test where the TX and RX pins from the CH340 to the 328P are shorted. This simply takes the data that would go to the 328P and shoves it right back into the ch340 looping it back to the host computer. Doing this I could not generate an error. However:

Another test is where I used 2 arduinos, one with grbl, and the other just using the USB->serial chip. The grbl board was a Nano with ch340, and the other was a Mega with 16U2, but the actual Mega did nothing. I had the RX pin of the Nano connected to the TX pin of the Mega. In this way, the 16U2 transmitted everything that got through the CH340 back to the host computer. Essentially just looping it back. Meanwhile the Nano was running grbl and was set to echo responses back to the host computer.  I wrote a test program that compared the 3 data sets, sent lines, grbl echoes, and data looped through the 16u2.  All 3 were compared and the results were that the corruption was happening within the ch340.  When an error occurred, the 16U2 data always matched the data from the ch340 to grbl, but it didn't match what was sent.  This confirmed that the corruption was happening inside the ch340.

109JB, 2017-02-28:
I guess for me, although I really like the nano platform and still use it a lot on other stuff, the problems with the CH340 and the fact that the 16U2 is "fixed" makes choosing a no-brainer for me.  I just don't trust the ch340 for my machining stuff. This is especially true when you can buy a UNO clone with a 16U2 for about $4 and avoid all those problems.