grbl/grbl #609: grbl on stm32f1

HUANGMENGDI, 2015-03-01:
hello there! i have port grbl for cotexM3 processer(without FPU@72mhz),and support 4axis ,then i've got a questions:how fast dose grbl run on it? f8000is OK? #my English is soso:).

Blue-Design, 2015-03-02:
Can you share your code,I am very interested in

HUANGMENGDI, 2015-03-05:
there are many bugs,likethen i fix then,i'll show you,

sebbra, 2015-03-05:
How have you done the processor-depending stuff like IO, timer and serial?
Do you use the STM32 peripheral library?
I am planning to use the STM32-Nucleo Boards with a grblshield-compatible shield.
I think we need some type of hardware abstraction layer, maybe take the avr depending stuff out of the grbl folder and make one folder for every architecture with extra makefiles.

HUANGMENGDI, 2015-03-05:
@sebbra : Add many new declare. Like:
#define X_dir_pin_On   GPIOB->BSRR.B.SETIO0=1  //Set x_dir_pin on 
 #define X_dir_pin_Off  GPIOB->BSRR.B.CLRIO0=1  //Set x_dir_pin off
serial do some change on IRQhandle. 
Timer use TIM6(timing_mode) and TIMx(one_plus_mode)
Donot use peripheral library,useless in grbl I think.
abstraction? if be sure as fast as it run,possible to do.