grbl/grbl #611: New 'Cycle-Start/Resume' pulse needed for each M4, M5 Spindle command

markbotics, 2015-03-02:
We've built a CNC like machine using GRBL 0.9c which is programmed into our Arduino UNO .

We have the 'Cycle Start/Resume' Uno pin connected to a pushbutton (green for us), and have set the 'Auto-Start' variable $14 to off ($14=0) (https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.8#14---auto-start-bool). This allows us to load an entire file of gcode command and have the machine not move until we press our green button, which is very nice.

What we want is for the machine to then execute the entire program after 'Cycle Start/Resume' has been pulsed low (except for pause or E-stop of course). 

Our problem is that it only works for programs with only G01, G02, G03 commands but not M4, M5 commands that we use to turn our spindle on/off. When the GRBL interpreter hits our M4 or M5 commands it waits for another green button press (low signal on 'Cycle Start/Resume' Uno pin).

Here is some gcode which runs fine, i.e. one green button press is enough to execute the entire program:

g90
f100000
g01 x0 y0 z0
g01 x150 y130 z0 
g01 x-150 y130 z0 
g01 x-150 y-130 z0
g01 x150 y-130 z0
g01 x150 y130 z0 
g01 x0 y0 z0
And here some gcode which stops after every M4 or M5 command:

g90
f100000
g01 x0 y0 z0
g01 x150 y130 z0 m4
g01 x-150 y130 z0 m5
g01 x-150 y-130 z0 m4
g01 x150 y-130 z0 m5
g01 x150 y130 z0 
g01 x0 y0 z0
It does not matter if the M4 and M5's are on separate lines.

Anyone know how to make the GRBL interpreter not wait for the spindle on/off (M4/M5) commands?

chamnit, 2015-03-02:
@markbotics : This has been fixed in Grbl v0.9i, edge branch.

closed 2015-03-02 by chamnit

markbotics, 2015-03-02:
Thanks for that, can you tell me where to find v0.9i?  I am new to Github.  I see only up to v0.9h at https://github.com/grbl/grbl-builds/tree/master/builds.

Thanks