grbl/grbl #612: Homing problem starting with machine at limit (with limit switch pressed)

markbotics, 2015-03-02:
As you probably know, the $h command is used for homing.  One sets some parameters (such as 'seek', 'feed' speeds and invert masks for direction) and the machine moves first in Z and then in X & Y until the limit switches mounted at the end of the machine's desire motion are reached, where it shuttles back and forth 3 times at the boundary of a switch, and then moves back away from the switches by another parameter amount. 

Our problem is that one of our axes is usually at the end of its travel after power-up.  This is because this is a vertical axis and after the power is shut off it moves (gently slides) down to the bottom.  And so unless someone moves it (undesired extra user interaction) the homing sequence has to be initiated with the vertical axis limit switch pressed at the start.

Unless we are doing something wrong, GRBL 0.9c does not like this situation and goes into the error mode requiring a reset.  

I can hypothesize one reason that it does not know which way it should move.  However, due to the invert mask bits for the 'seek' motion (moving towards the limit switches) it should know to back off in the opposite 'feed' direction until the switch become un-pressed.

For us, the limit switches are active (the machine is at an end point) with the default active low voltage.  Our limit switches are normally open and connect to logic GND, and the inputs have a pull-up resistor.

Thank you if you know what to do.  Hopefully we don't need to modify the source and reprogram.

chamnit, 2015-03-02:
@markbotics : The problem is that Grbl uses one limit pin to detect if a hard limit has been triggered. If you have two switches on both ends of travel, Grbl cannot tell which direction to move. If you have one limit switch (default assumption) per axis, it'll home even if it's triggered. I typically suggest OEMs use one limit switch for homing and then use soft limits to keep the machine within the machine volume.

closed 2015-03-02 by chamnit

markbotics, 2015-03-02:
Thank you for a quick answer.

I think the two hard limit switches per axis is not relevant and I shouldn't have mentioned it.  GRBL doesn't know that there are two limit switches connected in parallel.

Unless I'm doing something wrong, it doesn't home if triggered (what I assume you mean when the signal is active = low = limit switched pressed) when homing begins.  I find it gives an error and stops if I run $h while the limit switch is pressed.  

I assume it starts in 'seek' and immediately detects the limit switch, and then should try to go back in the 'feed' direction (opposite to seek) until the limit switch is no longer pressed (signal goes high for default polarity).  I notice it does this three times for each axis.  

Currently I can only run the homing sequence if someone holds the machine up so its vertical limit switch is not pressed.

markbotics, 2015-03-03:
Hello, I was wrong in my last message.  Grbl is indeed homing correctly, even with the machine booting up with one limit switch active (N.O. switch closed so connected to GND).  

The problem was that I needed pull-up resistors on the inputs.  The Uno internal pullups must not have been enough.  I needed not 10K, 4.7K, but 1K resistors pulled up to +5V to get clean performance.

timryder, 2015-03-04:
Inbuilt a custom board to run grbl and I too use 10k resistors to pull up the input for a clean Signal.

electrokean, 2015-03-04:
This is most likely electrical noise from the motors coupling into the limit switch cables.
Decreasing the pull up resistor value (i.e. providing a stronger pullup) will make the signal lower impedance, and reduce the effect.
I'd recommend also adding a 0.1uF 50V capacitor between the limit input and GND to filter high frequency noise.
Something like http://www.digikey.com/product-detail/en/K104K10X7RF5UH5/BC2665TB-ND/
I also put ferrites on my limit switch cables, near the Arduino (as a belt and braces approach).  I use two loops through a Wurth Electronik part 7427003
http://www.digikey.com/product-detail/en/74270034/732-1551-ND/

markbotics, 2015-03-05:
I confess I didn't experiment much with $26 (Homing debounce, ms) before putting on the pullup resistors.  With a storage scope it might be that the spikes were spurious and proper debouncing (checking a signal change is consistent one, or ideally several, times spaced several ms apart) would have fixed it without a resistor.  However, starting with a clean electrical signal is probably best.  Your mention @electrokean of adding the cap could have been an alternate solution that wouldn't have required such a small (1K) resistor if we look at it from a low pass filter point of view.