grbl/grbl #613: Homing best practices?

markbotics, 2015-03-02:
We are making a machine that we want to be as "turn-key" as possible.   Our machine needs to learn where it is after power up.  We aren't quite brave enough to put the homing code into the commands that can be run automatically after reset, and will likely have a GUI button to home (ideally there would be a 'Home' pin assigned on the microcontroller).

This is a general question, what is the best way to implement homing?  One specific sub-question would be, can we program them into a subroutine callable with a single command (not sure how to do that)?  So far our sequence (what we type into our GUI) is
$h
g92 x-180 y-260 z+205
f10000
g01 x0 y0 z0

For those that have implemented a homing system, beyond just knowing what sequence to enter each time (for those running their own personal creation), can you share some advice?

We are using 'GRBL Controller 3.6.1' as our Windows GUI.  We are using GRBL 0.9c in an Arduino Uno.

Thanks,
Mark

chamnit, 2015-03-02:
@markbotics : OEMs typically write or modify their own GUI to implement their specific homing requirements. Grbl is generic by nature to cover most machines. If you need it to do something else, I would look into writing your own GUI. See our Wiki on recommended protocol.

closed 2015-03-02 by chamnit

chamnit, 2015-03-02:
@markbotics : Re-reading your question on homing. Noticed that you use G92. Don't use this for a 'turn-key' machine. G92 deletes upon a reset or if its overwritten by the user. Use the G54-59 work coordinates systems. These are set by the G10 command. See the linuxcnc.org g-code descriptions on how they work. Grbl follows these very closely.

markbotics, 2015-03-02:
Thanks for that, I am now trying to learn coordinate systems.

I am looking at the linuxcnc.org tutorial you suggest, plus Wikipedia http://en.wikipedia.org/wiki/G-code#G50_position_register 

I'm assuming some G commands say "from now on x,y,z coords are in space named <--->" and that other G commands should set one of these coordinate systems, either zeroing them ("coordinate system named <---> is currently at 0,0,0") or setting them ("coordinate system <---> is now at coordinates X=542, Y=-200, Z=100").

It seems 'G...' can refer to commands and register names.

Do you know any simple clear text anywhere that summarizes the coordinate situation?  I imagine a few succinct sentences and examples would do it.

Apologies if this is not the forum for this discussion.

markbotics, 2015-03-02:
More on my homing question, which starts to extend into the general question of coordinate systems.

When it comes to multiple commands, you (Chamnit) suggested putting that into the GUI.  But I believe one can store a sequence of commands in non-volatile memory, I'd like to investigate this for reliability in case the serial communication or Windows does something funny during this homing operation.

I know in many CNC cases, the machine never moves except when stepper motor pulses are applied, also that in many applications people manually home their machine.  But our machine should be "turn key" for others to use without in depth training.  And in our machine, it can be moved once the power is off, indeed one axis is vertical and slumps down to a rest position when power is removed.  So some non-volatile machine position stored in the controller is not suitable for our specific application, every time the power is cycled the machine must be assumed to be in an completely unknown position.

On the coordinate system subject I am researching and posting in http://www.shapeoko.com/forum/viewtopic.php?f=3&t=5576 Ideally I would work in just one coordinate system, whose value is set once all limit switches are set in the homing sequence.

chamnit, 2015-03-02:
@markbotics : You are free to alter the Grbl's source code however you think it'd work better for your product. You just need to abide by the GPL licensing terms and open any changes you make.

markbotics, 2015-03-02:
I'm just trying to understand what can be done with the existing versions.  

We're eager to not re invent the wheel, especially since you've no doubt faced and thought through all these issues and done a better job than we would do.

Any simple description you can link to regarding my coordinate system question?  I don't know quite enough to formulate a specific question, other than I need to tell the machine "you are now at X=.., Y=..., Z=..." and have all G01, G02, G03 commands interpreted as absolution coords in this coordinate system.  Thanks.

twforeman, 2015-03-02:
I responded on the ShapeOko forum, but you can start by looking at this: http://www.shapeoko.com/wiki/index.php/G-Code#Using_the_Work_Coordinate_Systems

markbotics, 2015-03-02:
Thanks for that link @twforeman , that section was very good.  I responded on the ShapeOko forum (http://www.shapeoko.com/forum/viewtopic.php?f=3&t=5576) 

Just to tentatively answer my own second question about coordinate systems, what I needed to know is that the MCS (Machine Coord System) is set to 0,0,0 when the $h homing operation is complete and that one shouldn't try or want to change it.  Instead one should set the WCS (Work Coordinate System) to what this home position is in work coords with the 'G10' command.  There are 6 possible WCS's supported in Grbl of which I'm going to keep it simple and only work in the first (P1).  Therefore I enter 'G10 P1 ...'.  I believe, but aren't sure, that I then need to switch into that coord system with the 'G54' command.  Thereafter all G1,G2,G3 commands are in this P1 WCS.

And '?' instantly replies with a status message telling me the machine and work coord system <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>.  I will use this to confirm my understanding.

To answer my original question about homing, for my application I think it's correct to do the following.  I'm going to replace the original G-code sequence I wrote ($h, G92, ...) with:

$h
G10 P1 L20 X-180 Y-260 Z205
G54 (not sure if this is correct or necessary)
G90 (set absolute coords -should be default but will issue to be safe)
F10000 (set feed speed)
G1 X0 Y0 Z0 (move to desired origin in middle of workspace)

A question I have: do I need to call 'G54' to set WCS#1, or does the G10 command automatically move us into this coord system?

Thanks

markbotics, 2015-03-02:
I forgot to add, in my last comment I meant to say that my home position is at X=-180mm, Y=-260mm, and Z=205mm therefore after $h is complete I want the work coord system to be (-180,-260,205) therefore I send 'G10 P1 L20 X-180 Y-260 Z205', then possibly 'G54', then 'G90' to make sure I'm in absolute coord system, and then 'G1 X0 Y0 Z0' should move my machine to the WCS origin in the center of my workspace.

chamnit, 2015-03-02:
@markbotics : Please re-read and study the g-code standard on linuxcnc.org's website [link](http://linuxcnc.org/docs/html/), the section on G-code programming. They have extremely good explanations on how the g-code standard works, what are defaults, what is kept in persistent memory. A lot of these questions can be figured out by studying these pages and experimenting with Grbl to verify it does what it supposed to do. There are also plenty of other resources on CNC programming and machine behavior, which Grbl tries to follow closely, like Peter Smid's CNC Programming Handbook. 

If you are developing a CNC machine product, I highly recommend that you first spend the time to understand the g-code standard in its entirety and the standard operation of machine tools. This will help clear up all of these questions. I'll be happy to help with the Grbl specifics, but this question is about standard machine operation.

But I'll help you and answer your question:
- $H homes and finds machine zero. That's it sole purpose. Usually after a homing cycle, programs should use a G28 or G30 command to automatically move the machine to a pre-determined safe/start location. This is typically at the mid range of travel of X and Y and Z is all the way positive. The other G28/G30 (not used for post-homing) is usually used for moving to the tool length offset probe location. Work coordinates are usually only used for setting part zero or multiple part zeros if you are milling multiple identical parts in the machine volume. This is a common usage of these tool and are often written and described as used in this way.
- G28.1 and G30.1 commands stores the current location (if I remember right) to memory and will retrieve it when G28 or G30 are called. This is described in the website.
- G10 sets the work coordinate offsets relative to machine zero. It's persistent and kept in permanent memory. You only need to do this once. If you need to change it, send Grbl a new G10 command.
- There are a series of default G-code modes. These are outlined in the g-code standard on linuxcnc.org website and in standard CNC handbooks on g-code. These include G54, G17, G0, G90, G94, G21 and others. Grbl resets to these whenever Grbl is soft-reset or power cycles. It's redundant to send these at the beginning of a g-code program, but most CAM programs export a g-code header with these in it to guarantee the settings are correct. Most CNC handbooks will talk about the program header line. It's standard operating procedure to have a line like `G54 G17 G90 G94 G21` at the beginning of a g-code program.

markbotics, 2015-03-03:
Hi, some good info.

So I don't need to do 'G10 P1 L20 X-180 Y-260 Z205' more than once, since it will be stored in non-volatile memory.  I gather that it's best to do this during machine setup, when I also set the max speed, accel, direction invert flags, etc.  

And so I really only need to do $h each time the machine turns on, and get all G-code programs to include the standard G54 G17 G90 G94 G21 at the beginning.  Where G54 puts us into the P1 WCS, G17 makes arc drawing in the XY plane, G90 ensures all G movement commands are absolute coords, G94 sets the units for feed rate as units per minute and G21 sets us to metric millimetres.  

We're not making a line of products, but a single machine.  Yes, there is a large body of documents out there that we could read but we only really needed a simple clear set of instructions and minimal explanations.  I taught courses for a few years and know well the most efficient way to learn is to fully but concisely define the basics up front, ideally have first a complete working example.  For example does G54 mean you can enter 'G54' by itself, or follow it with more numbers as 'G01' for example.  You've done a lot of work, if you are the Grbl writer, and I posted these questions expecting other users to help and not take up your time.  However, of course, your help is appreciated.

chamnit, 2015-03-03:
@markbotics: True that you need only a subset of gcode commands but Grbl supports the minimum commands needed for a CNC machine tool. You'll need to learn all of the ones Grbl supports to create a good product. Gcode resources have plenty of great introductions on the basics. Please spend the time to read these because it's pretty clear that you need to. It'll take a lot of time and effort to explain everything.

markbotics, 2015-03-03:
I'm not creating a product, but a single machine for a single user.  You're doing a good job creating a g-code machine for the Arduino Uno.  But concepts that are obvious to seasoned CNC people are non-obvious to a newbie.  In an efficient world a person should't have to become an expert to do the basics.  A single page would have told me, and most users, all we need to know, but I couldn't really find it and have to cobble together knowledge from several pages.  And if I have these questions, so do others, and ideally you would want more people using Grbl. 

I'm verbose in my questions because I believe I'm helping others who will repeat my tracks later.

chamnit, 2015-03-03:
@markbotics : In an efficient world, yes, a single page would be helpful, but welcome to the world of open-source, where things are done when there is time and willing effort. 

However, I'm trying to give you the best advice I can. Learn the gcode standard and the codes supported by Grbl. The linuxcnc website I linked for you is one of the best resources available to us. I know the standard is confusing. It can be. It goes well beyond a single page to explain it. That's what I'm trying to get across   

Like I said before, explaining all the minute details can be answered by studying the standard and actually experimenting with Grbl. Even if you are building a single machine, not a product, you still need to learn it if you want something turn key. Otherwise you'll be flying blind the entire time, as you'll not understand what CAM programs are outputting and what you're telling the machine what to do.

markbotics, 2015-03-03:
Yes, all your replies are much appreciated, as is the no doubt countless hours of coding you did.  I did sit in front of our machine for some time trying things, and will go back to the machine tomorrow with new knowledge.  Thanks for everything.

My conclusion, as I understand it, to my original question is that '$h' is all the homing code I require, and the rest should either be done at setup (like G10 should be done when setting up the $ registers for max speed, accel, etc) or at the top of the program (eg. put 'G54 G17 G90 G94 G21' or similar at the start of every program).  

And the rest of my confusion was a fundamental misunderstanding of coordinate systems, section '5. Coordinate Systems' of http://linuxcnc.org/docs/html/common/User_Concepts.html was quite helpful.  My takeaway is that there are 7 coordinate systems supported in Grbl, MCS (Machine Coordinate System) which is reset to 0,0,0 with $h, and six WCS (Work Coordinate System)'s named G54 to G59, of which the default simplest case is to be in the G54 (first WCS).  The offset of G54 is set by G10 with argument P1, which is persistent (non-volatile) info.  And the G1, G2, G3 (and maybe G0) commands are only in one of the coord systems.  

I'm working in a more robotics than CNC application where we really just need to move along paths but need homing.

markbotics, 2015-03-05:
Thanks for your patience, and sorry for bugging you.  Now that I understand it, it seems clear but earlier I didn't quite get it.  Just to be thorough, here is my specific example, written out in the hope that someone else learning like me finds it useful.  Maybe this isn't the right place for it, being for coding issues, I also have almost the same text on the Shapeoko forum http://www.shapeoko.com/forum/viewtopic.php?f=3&t=5576

Here's what I did for my machine:

I first found the range for each axis for later use with soft limits, enter those 'max travel' numbers in the $130,$131,$132 registers (v0.9g).  This is the range (delta) of movements found by jogging that define a safe looking range.  These max travel numbers set what the MCS (Machine Coord. System) will be after homing.

Then, use the 'G10' command once, after homing, to set the WCS (Work Coord. System) relative offsets to MCS.  These are then stored persistently (non-volatile) in EEPROM memory.  Then turn on soft limits to refuse commands and go into reset should a command request motion outside the range of the home position to that in the $130,$131,$132 registers.  I was earlier mistaken thinking I could set the MCS range, and that the G92 command would be useful.  

Personally I found hard limits $21 (hard limits, bool) not to work for me, I had to keep it at 0.  I opened an issue on this https://github.com/grbl/grbl/issues/616.  It is possible that, like other issues, I am wrong, but so far I find it always triggers for any requested move (eg. G1 x0 y0 z0).

In this initial setup I set these registers for the range of motion:
  $130=560.000 (x max travel, mm)
  $131=720.000 (y max travel, mm)
  $132=400.000 (z max travel, mm)
Then I set these homing registers:
  $22=1 (homing cycle, bool)
  $23=3 (homing dir invert mask:00000011)
  $24=2500.000 (homing feed, mm/min)
  $25=2500.000 (homing seek, mm/min)
  $26=250 (homing debounce, msec)
  $27=20.000 (homing pull-off, mm)
Note I am homing to the bottom left (negative X,Y) but positive Z.  Note also I set the pull-off to 20mm (all axes from the one $27 register).
Then after a homing operation ($h) my MCS is at -540,-700,-20.  The MCS was (-560,-720,-0) (from max travel variables for negative homing moving axes X,Z and 0 for positive homing axis z) at the point when the last limit switch was found, before it did the homing pull-off.  After the homing is fully finished, it has backed off (moved positive for X,Y and negative for Z) 20mm in each such that the MCS is at (-540,-700,-20).  I see this by typing the '?' character and seeing <Idle,MPos:-539.993,-700.000,-19.996,WPos:-260.000,-340.000,285.000>.

Then, the first time, I would define what this MCS is in my desired WCS.  There are 6 WCS's supported, the default is the first (aka 'G54') WCS, so we use 'P1'.  I confess I don't know what 'L20' does.
G10 P1 L20 X-260 Y-340 Z285  
That sets the WCS (see 'WPos:-260.000,-340.000,285.000' part of ? reply above).
Note that the WCS is an offset of the MCS by (280,360,305).  I.e. a WCS point is MCS+(280,360,305)

Now if I supply a feedrate (eg. F5000), make sure I'm the first WCS (G54) and then go to my WCS origin with:
G1 X0 Y0 Z0
then my machine moves to the center of my work volume.  Then if I type '?'' I see 
<Idle,MPos:-279.985,-360.007,-304.994,WPos:0.008,-0.007,0.002>
Note the same relationship where  MCS+(280,360,305)=WCS     (-280,-360,-305) + (280,360,305) = (0,0,0)

A preamble (similar to that given by @chamnit on Github/grbl) to put at the top of our G-code programs is: (G54 makes sure we're in the first WCS, which is default unless you change it)
G54 G17 G90 G94 G21   (54=make sure we're in WCS#1, G17=XY arcs, G90=absolute coords, G94=feed rate =mm/min, G21=mm units).

Now if I turn on soft limits 
$20=1 (soft limits, bool)
then the machine will only let me move to points within the range 
range  X  -280mm to 279.9mm   (can enter G1 X-280 or G1 X279 as extremes)   
range  Y  -350mm to 359.9mm   (can enter G1 Y-350 or G1 y359 as extremes)
range  Z  - 95mm to 300.0mm   (can enter G1 Z-95 or G1 Z300 as extemes)
These are in the WCS#1 (G54) coord system.

Thank-you @chamnit for being patient and setting me straight, despite it not being your duty since you're already putting in massive time actually coding for us all to use.   Thanks to @twforeman for your links too to the coord system overview.

Miigon, 2018-03-23:
Saved me a lot of time.
Thx all you guys. Markbotics posted so many details although he has already solved his problem. And that really helps others (like me) who have similar problems.