grbl/grbl #614: Can anyone recommend a simple Windows GUI -to hand off to others

markbotics, 2015-03-02:
Hi,

We are building a machine around GRBL 0.9c on an Arduino Uno.  We are using 'GRBL Controller 3.6.1' as our Windows GUI.  It seems ok for development, but we need something simple to hand off the system to others who are not so interested to learn in depth. 

What we really want is for users to just simply turn on the machine and the nearby laptop, run some program whose shortcut we'll put on the desktop, see the machine isn't homed and push the 'home' button, then start/pause/stop the machine using a pendant we made.  Most of the time the G-CODE program will be the same.

Ideally we would have a simple Windows GUI that would have:
-a big 'Home' button to initiate a homing sequence ($h g92 ... f... g01 x0 y0 z0) 
-a clear depiction if the machine is locked or unlocked (this leads to a question of what serial command can be sent to determine the locked state) so they know whether to home or not  
-load G-CODE program  -ideally it would automatically have loaded a default program
-optional jog buttons

I hear there are many GRBL GUI's out there.  Before we re-invent the wheel, can anyone suggest a "dumbed down" simplified such GUI program for us?

Thanks

winder, 2015-03-02:
Universal Gcode Sender's current GUI is built with a graphical interface builder. If you install Netbeans you can configure the gui to look exactly the way you like with no coding.

chamnit, 2015-03-02:
@markbotics : UGS is one as @winder mentioned. There are several others list in the Wiki. Please read through this and use the most recent version of Grbl. The edge branch v0.9i, which will soon go to master, has gotten rid of the cycle start (green button). It was redundant when used with GUIs, since streaming starts the cycle automatically. Cycle start now only acts like a resume to disable holds and pauses.

closed 2015-03-02 by chamnit

markbotics, 2015-03-02:
The cycle start (green button for me) is not redundant for our application.  The reason being we only wish the user to go to the computer once when the power is turned on, and on rare occasions to change the program.

We want the user to be able to do everything with a pendant when working near the machine.  Green button = execute loaded program or resume from pause and E-stop
Yellow button = pause
Red button = E-stop -stop program and kill power

So if you've changed everything to not use the cycle start pin input, then you are restricting people to use a GUI on a computer.

chamnit, 2015-03-02:
You still need something to start the stream and that requires a GUI or something else. That something else could be your SD card server or whatever you are using to send data to Grbl. Or you can begin a job by issuing a feed hold and then the "green button" will behave like a cycle-start. There are tons of ways around this problem and there are too numerous reasons why Grbl was changed to this behavior to explain in detail here.

markbotics, 2015-03-02:
Ok, great.  Yes, I figured there were good reasons to change this.

Issuing a "feed hold" at the beginning could be a solution, except that the user needs to know to put this in their program.

I am looking for the 'feed hold' G-code command.  I see 'M0' as 'program pause' according to http://linuxcnc.org/docs/html/gcode/m-code.html.

chamnit, 2015-03-02:
@markbotics : Yes. M0 will work, but your interface to Grbl can do this automatically as well, whether its a GUI on a computer or a separate MCU. Keep in mind Grbl requires an interface.

markbotics, 2015-03-02:
I also see the potentially simpler '!' character as described in https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl#real-time-control-commands, which looks nice in that it's processed immediately and not after a set of instructions processed.  However, we may not use them since our users might be more familiar with convention G-code.

Yes, we know Grbl shouldn't exist as a standalone box, we just want the user to only have to initiate a homing sequence and load a program, and do the rest from a pendant.

chamnit, 2015-03-02:
@markbotics : The cycle start input pin uses the same code as the '!' realtime character. This is intended to be a software version of this. If you have a pendant , you can design the pendant to be the main controller to home, load a program, and send a '!' cycle start or begin streaming the program. This '!' resumes feed holds, M0 pauses, and exits the new safety door state, so you'll need it if you don't directly wire a switch into Grbl's cyclestart/resume pin.

gerritv, 2015-03-02:
Your pendant can be designed control the GUI/keystrokes as well as pins on Grbl.

markbotics, 2015-03-02:
@chamnit : you said cycle start='!', I think you meant '~'.  I see on https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl#real-time-control-commands "~ cycle start/resume, ! feed hold, and ^X soft-reset"

This conversation has moved away from the original GUI question.  I'll start a new thread (issue) titled "Request for input pin functions" where I'll also ask for direct pins for homing and cycle repeat (cycle repeat I'll also ask separately as "Looping using Cycle Start").

chamnit, 2015-03-02:
@markbotics : Yes, I meant '~' for cycle start, but the meaning is the same. All real-time commands are bound to the same routines as the input pin of the same name. This is to make Grbl controllable via the serial interface only. 

As for direct pins comment, a homing input pin is a pretty good idea. I think about this one. A cycle repeat is impossible, since Grbl doesn't store any g-code programs in memory. To repeat, you'll need to have your GUI stream the program again.

Please keep in mind that Grbl is an open-source project that I develop for in my free-time. If you want certain features that your product requires, please think about giving back to this project and try to learn the ins and outs of the program before asking a lot of questions.

markbotics, 2015-03-02:
@chamnit Thanks for addressing my misunderstanding on repeating the cycle, that makes sense that it does not repeat commands (unless there is some subroutine functionality I've heard mention of).

We appreciate your help.  But I did read all the relevant pages and experimented with my machine before asking questions.  While the answers were there, if I didn't absorb it the first time then others probably won't as well, and my questions serve to help others -as that with a good title others will find and read it before losing too much time (and possible machine damage).