grbl/grbl #616: Hard limits v0.9c

markbotics, 2015-03-03:
Hello, I am experiencing a situation where it goes instantly into alarm mode "ALARM: Hard/soft limit" whenever I try to do a move, once I turn hard limits on with $21=1.

The machine is definitely away from the limit switches.  And the limit switches are used for successful homing, which confirms I believe the digital inputs to the machine.  I can home ("$h") and the machine properly goes through the homing sequence, including the backing off by the amount in $27 ("Homing pull-off").  To be extra sure I move the machine clearly in the middle from all switches and enable $21=1 and it goes into alarm mode.  

A voltmeter confirms the three pins are at 5V.  All three limit inputs are definitely high, extra strong 1K pullup resistors on 'Limit X-axis',  'Limit Y-axis', and  'Limit Z-axis' pins.

I have v0.9c.

Thanks if you can help.

markbotics, 2015-03-05:
I mistakenly said v0.9c, it's v0.9g.

twforeman, 2015-03-05:
This is not a bug. You most likely need to add a filter capacitor to finish the RC filter to remove the motor noise. Also if you don't have shielded wires for your limit switches this can be an issue.

Other people are using limit switches (myself included) with no issues.

markbotics, 2015-03-07:
Hmmm.... homing works fine with no problems (after I added pullup resistors to the limit switch inputs).  They are active low signals.  But if I enable hard limits, it goes immediately into the alarm state, even with the MCS clearly within the max_travel limits.

markbotics, 2015-03-12:
Is the hard limits active high signals?  I would assume they are the same sense as is what is used for homing (active low for us).

markbotics, 2015-03-12:
Hi, can we re-open this?  We still have this issue where it goes immediately into "ALARM: Hard/soft limit" even with all three limit inputs clearly +5V.  With $21=1 (v0.9g) it never works.

markbotics, 2015-03-12:
Sorry... it is indeed open (just saw red "Closed" above)

rjclocks, 2015-03-12:
@markbotics You almost certainly have a 'noise' issue being generated by the wiring on your machine. This was causing me all sorts of problems with mine at first.
You can 'see' the problem if you have access to an oscilloscope.
There are two things you can do which will help: Shielded cables and a filter capacitor across the supply to your limit switches.
After implementing both of these, all my noise issues went away.

Dikabro, 2015-03-14:
I had same problem on grbl v0,8c, but I connected pull up resistors and then everything worked just fine. 
BUT yesterday I uploaded grbl v0,9g into my Arduino NANO and now it seems that I started having the same problem as markbotics. My system immediately goes into ALARM state because of Hard/soft limits.

Mgilbride, 2015-03-14:
Also look at your axes set up. There have been convetion changes that alone may be causing limit alarms right out of the box because your gcode is programmed for the previous convention. All axes are now negative by default and you may have programmed your part for positive axes. 

If you are having noise problems even after adding shielded cable, filtering caps and resisters to your control system, I strongly suggest 12 to 48 volt DC relays and run all of your limit signal lines at that voltage. Signal lines at that voltage are very immune to noise. I required these on my rig. And they eliminated all false triggers due to noise. They did not eliminate my moderately regular blunders. All problems with limit switches in my experience have been from noise, my miswiring, and not adjusting settings and programs to new developments in grbl.

chamnit, 2015-03-14:
@Dikabro : There have been no significant changes in Grbl v0.9 to be a causation for your problems. The hard limits work the same way as they did in v0.8 by using the pin change interrupt to detect a hard limit.

Dikabro, 2015-03-15:
Ok. So at first I will try to add some filtering caps and then will post my test results. :)

Protoneer, 2015-03-15:
Is there some kind of guide for adding filtering caps and best sizes to
use? (Sorry if this has been asked before)

gerritv, 2015-03-15:
http://www.eng.utah.edu/~cs5780/debouncing.pdf

twforeman, 2015-03-15:
@gerritv He's not dealing with switch bounce. He's dealing with noise on the line generated by the stepper motors.

What you really need is a pull up resistor (value is not super important - something under 1000 ohms) that goes from the pin on the arduino to the +5v supply and then a capacitor (about .1uf) from the pin to ground.

gerritv, 2015-03-15:
Which makes a filter not unlike described in the debounce article.
Without a scope he won't know what is really happening

chamnit, 2015-03-15:
@gerritv is right. It's a low-pass filter on the input pin, which is the same strategy use to both help remove noise and debounce a switch.

Protoneer, 2015-03-15:
Thanks @gerritv. Does the low value pull-up resistors improve the already
internally pulled-up pins?

Mgilbride, 2015-03-15:
Good read (what I've read so far). Really like the switch testing and other anecdotes. Thanks for the ref. I forget. Does grbl have software debouncing? Thought it did.

chamnit, 2015-03-15:
@Mgilbride : It does, but it doesn't help with noise. So, if you have a low-pass filter RC circuit wired in, it makes the software debounce redundant. Which is why it's only a compile option.

Dikabro, 2015-03-15:
Ok it helped! I added filtering capacitor from pin to ground and now my limits works correct. There is no  instant Alarm state anymore.

gerritv, 2015-03-15:
I don’ think being in parallel is a good thing. Use either one or the other.

 

e.g. 2 1k resistors in parallel result in 500ohm resistance.

 

Gerrit

markbotics, 2015-03-27:
My problem is not due to signal integrity problems on the limit switches.  It happens even when they are disconnected.   As soon as I turn on hard limits and then request a move it issues the alarm.

gerritv, 2015-03-27:
Without a scope on those pins you won't know....... Noise comes in via many paths. E.g. the spike form a spindle motor turning on can do interesting things when there are ground loops, open grounds, grounds at the wrong end of shielded cables, arcing across relay contacts, etc.

dgtlmoon, 2016-01-18:
But there would have to be a spike for more than ```$26 Homing Debounce ms``` right? If he set his $26 to like 5000, shouldnt the stepper run for 5 seconds before hitting the Hardlimit?

chamnit, 2016-01-18:
@dgtlmoon : Homing debounce doesn't apply to hard limits. It's simply a short enforced delay between parts of the homing cycle. Grbl doesn't enable software debouncing by default, because debouncing will not 100% help with electrical noise. There's always a chance that electrical noise can still trigger a hard limit. It's best to ensure that your electrical wiring is as noise free as possible. If this is impossible on your setup, I would consider just using homing and soft limits. This will generally do the exact same thing.

dgtlmoon, 2016-01-18:
@chamnit Hmm that's an interesting comment, so the documentation needs updating then? ```"
Whenever a switch triggers, some of them can have electrical/mechanical noise that actually 'bounce' the signal high and low for a few milliseconds before settling in. To solve this, you need to debounce the ```signal" or is soft switches still actually physical switches?

chamnit, 2016-01-18:
@dgtlmoon : No. This statement still applies to the homing cycle and making sure the switch isn't debouncing before moving on to the next part of the cycle.

dgtlmoon, 2016-01-18:
@chamnit Ok cool, so $26 only applies to homing, got it

ThomasBj, 2016-01-28:
I have experienced the same problem with the Alarm triggering.

With very short leads to the switches everything works great but with 1meter lead (shielded and grounded) the problem comes back.
I get this alarm, for example when i switch on or off some electrical appliance (LED lightbulb). I am definately not in an electrically noisy location.

So i tried experimenting with shifting from NO to NC switches, using all suggestions with stronger pull-up (or pull-down resistorns for the NC switch), but I still had the same problem. 
I reverted back to NO switches and continued to try using a RC low-pass filter with different RC values but still no change. I even tried 2 low-pass filters in series, but without any luck.
I was now prepared to start using relays or optical isolators instead as described in this article
http://www.eevblog.com/forum/projects/opto-isolators-for-cnc-limit-switch-noise/
 
BUT !!!
At the end of my experiments i happend to connect a 10K resistor in series  with the switch and all my problems were gone. (So its basically looking like a RC filter without the C).
I experimented with different resistor values and the alarm started triggering again at 330ohm . 1Kohm was ok.
I don't know the electrical physics behind this, as it is no longer working as a low-pass filter.
With the internal arduino pull-up it becomes a potential divider when the switch closes.
Hope this help.

skrutt, 2016-01-28:
That is wierd. The only thing i cal think of is that the noise is divided and becomes to small to trigger the input. It's altso a low pass filter but with a very small C. 
But that this would work but not the RC filter is really odd, where did you place the filter? and what values was used?
You can use this calculator to get some idea of how your filters work http://hyperphysics.phy-astr.gsu.edu/hbase/electric/capdis.html

kenansatir, 2017-11-02:
Please use RFI, EMI, EMCi Noise Filters on limit switch cables (near to board) like :
https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20171102041903&SearchText=+RFI+EMI+EMC+Noise+Filter

But try to use big ones... about 2cm diameter....

Also connect your boards GND pin to ground...

Best regards

pinatl, 2018-01-27:
Surprisingly for me it worked when I set the $5 to invert  (=1), in mode normal ==> 5V with a switch NC on each hard limit PIN with a pull down resistor (4.7K). 0V when the hard limit is activated. For me $5 works the opposite it should work. Anybody the same experience?