grbl/grbl #617: Compiling grbl edge 0.9i

campe89, 2015-03-04:
Hi all!
I have try to compile grbl edge 0.9i with arduino ide 1.0.5-r2 but there is this error:

core.a(main.cpp.o): In function `main':
C:\Program Files (x86)\Arduino\hardware\arduino\cores\arduino/main.cpp:11: undefined reference to `setup'
C:\Program Files (x86)\Arduino\hardware\arduino\cores\arduino/main.cpp:14: undefined reference to `loop'

Any solutions?

electrokean, 2015-03-04:
You shouldn't use the old build procedure anymore.
Instead, use the example sketch for library Grbl called grblUpload (installed with the library).

campe89, 2015-03-04:
I have used the grblupload.ini on the example folder.
With arduino ide 1.6.0 the error is:

Arduino:1.6.0 (Windows 7), Scheda:"Arduino Uno"

C:\Program Files (x86)\Arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10600 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard C:\Users\ACAMPE~1\AppData\Local\Temp\build873130048115589558.tmp\grblUpload.cpp -o C:\Users\ACAMPE~1\AppData\Local\Temp\build873130048115589558.tmp\grblUpload.cpp.o 

grblUpload.ino:28:18: fatal error: grbl.h: No such file or directory
compilation terminated.

electrokean, 2015-03-04:
Have you put all the files from https://github.com/grbl/grbl/tree/edge/grbl into a grbl folder under your Arduino libraries folder, and then restarted Arduino?

electrokean, 2015-03-04:
I did have a similar problem initially because my network uses redirected home folders, and the updated Cygwin based compiler in Arduino 1.6 doesn't handle UNC path name in include directives.  If you have something similar, you may need to change your Arduino sketchbook folder to a standard path (under File->Preferences).
But I don't see any include directive for the grbl library in that compiler command, so I don't think you've put the grbl library in the right place.

campe89, 2015-03-04:
I have:
- setted the arduino sketch folder to C:\Arduino
- downloaded the grbl-edge zip on my desktop
- added the grbl as library
- if I import the grbl library, does not display anything on the sketch
- if I try to compile the grblupload.ini the error is the same

electrokean, 2015-03-04:
If importing the grbl library does not display anything in the sketch, then you must not have the files in the right folder.  In C:\Arduino\libraries\Grbl do you see all the files like config.h and grbl.h, or are they in another subfolder (e.g. grbl) below that?
When you change files in the libraries folder, make sure you restart Arduino IDE.

chamnit, 2015-03-04:
@campe89 : I just downloaded the newest Arduino IDE release and tested it on my Mac. It works, but you'll need to close the IDE and re-open it after importing the library. Also, when adding the grbl library, make sure you select the 'grbl' folder that contains only the source code. Sorry I haven't had the time to update the Wiki and write instructions on this yet. But, I'll make sure I do when it gets pushed to master.

campe89, 2015-03-04:
Thanks @chamnit , now works!

closed 2015-03-04 by chamnit