grbl/grbl #618: Grbl Halting at G3 commands

RobDrizzle, 2015-03-05:
When my GRBL shield gets to G3 commands in the code below, is gives an error: Unsupported Statement and halts mid arc. I haven't found anything like this on line

( Profile2 )
S1000
G0 Z3.0
G0 X68.844 Y114.256
G0 Z1.0
G1 F5.0 Z-0.75
G1 F100.0 X75.562
G3 Y145.844 I0.0 J15.794
G1 X68.844
G3 X68.695 Y145.83 I0.0 J-0.794
G1 X53.084 Y142.844
G1 X-11.759
G3 X-12.552 Y142.05 I0.0 J-0.794
G1 Y118.05

chamnit, 2015-03-05:
@RobDrizzle : Weird. There isn't anything unusual in your g-code that would cause Grbl to report this particular error. If you're using Grbl v0.9, it shouldn't be doing that at all, since this error is only reported with a Grbl '$' command. What version of Grbl are you using? And what GUI?

RobDrizzle, 2015-03-05:
grbl 0.9 and the gui is the grblPanel.
I'm using Cambam as a Cam software with their post processor. Would coinverting arcs to lines be a better approach?

chamnit, 2015-03-05:
@RobDrizzle : Ran this g-code on the master v0.9g and edge v0.9i. No problems here. I'm guessing that this is a problem with grblPanel. It might be sending some Grbl '$' command in the background and causes this error. I would try UGS or something else to rule out the GUI.

gerritv, 2015-03-05:
I tested the code on current version of GrblPanel and it ran fine???

Gerrit

RobDrizzle, 2015-03-06:
I'm going to look at some other code because I find it hard to believe that this is the first time I had Arc commands in my codes. So I wonder why this code has issues

RobDrizzle, 2015-03-06:
The code I ran a few days ago was completely littered with G3 commands and didn't have a single issue. The code was generated with the same software, same post and was ran on the same machine using the same GUI...

Weird.

RobDrizzle, 2015-03-06:
I converted Arcs to lines and it still stops in the same two places. It gets to a point, stops with the error, I hit resume, it goes around the arc a bit more, then stops and drops out of the program... So confused.

chamnit, 2015-03-06:
@RobDrizzle : Very strange. Could you post your Grbl settings and the `$I` build info? Also is there any more to the g-code program? Could you post the next 20 lines, if so?

RobDrizzle, 2015-03-06:
Sure, I'll do that when I get home. The code posted is for a the first of 9 identical parts milled in a plate of carbon fiber. I removed the milling op for the first part and then re-posted the code to only include the remaining 8 parts and it have been running fine (arcs converted to lines). The only difference between the parts is position on the mill bed. It's so weird that it only happens to that one part so now I'm starting to wonder if the code file is corrupted during post. Honestly... I have no idea!

chamnit, 2015-03-29:
@RobDrizzle : Just checking in on where you were with this problem. I just installed an `echo` compile option into the most recent master build of Grbl. You enable it in config.h. This will cause Grbl to echo back the command that you sent it, right before it executes it. This should help with this problem, but I'm not sure how GUIs will interpret it.

aimeiz, 2019-10-08:
I am experiencing same issue.
My CNC machine Popular CNC3-3018PRO reports firmware Grbl 1.1f and when I send Gcode line:
G3 X30 Y3 I0 J3
It replays with error message:
Invalid Gcode ID:33

When I send complete code by usb or from offline controller from SD card, it stops on this command.

Maybe firmware is not full grbl 1.1f since controller is based on avr atmega 328 but not atmega 2560??
grbl should suport G3 code shouldn't it?

It is not arduino based, I suppose it could be programmed like arduino via usb port.