grbl/grbl #619: Invalid GCode ID: 33

timryder, 2015-03-06:
Hey all, 

I've built a new circuit board and new hardware for a machine to run GRBL and it's working pretty good but I'm using UGS and sending the test program from Shapeoko "HelloWorld.nc" and round about line 256 I keep getting the error "Invalid gcode ID: 33".

[image]

Anyone know what the heck is causing this?

Using 0.9g. 

P.S. if I move that first line which gets the error
-G2X3.1143Y1.1304I-0J-0.9783F40

If I move that anywhere else in the program just arbitrarily it errors out there too. 
I tested it by moving it to line number 100 and as soon as UGS got there it reported same error. 

Any help?

chamnit, 2015-03-06:
@timryder : I've been meaning to write a Wiki on these g-code errors, but it's in report.h and .c files, if you're curious. The ID 33 error is when the target of the arc is invalid. Meaning that from the current location with the programmed arc radius, the target does not lie on the arc. It's outside a certain error bound. 

There must be something slightly off earlier in the g-code program that is causing this error. Or it may be something with UGS. I know it some pre-parsing for Grbl and it may be truncating the precision of the values too much.

twforeman, 2015-03-06:
Just as a question, what precision does Grbl use internally?

chamnit, 2015-03-06:
@twforeman : It follows the linuxcnc.org description of what should fail. Described below.

When the arc is projected on the selected plane, the distance from the current point to the center differs from the distance from the end point to the center by more than (.05 inch/.5 mm) OR ((.0005 inch/.005mm) AND .1% of radius).

twforeman, 2015-03-06:
.050" Wow, that seems huge to me. :) Thanks.

timryder, 2015-03-06:
Ok so what does error 36 Unused words mean?

chamnit, 2015-03-06:
@timryder : About what it says. There is something in the g-code line (block) that isn't used by the commands present in the line or the current mode state. For example, if you have a G1 line command and you put an I arc offset word in there. The I isn't used by G1 and it will give you this error.

timryder, 2015-03-06:
Nailed it @chamnit  Thanks a ton :)

closed 2015-03-06 by timryder

timryder, 2015-03-09:
Side note: I'd have o agree with @tw

pedrotorres25, 2018-03-30:
I came here to find a solution to the ID33 error.  Very informative here as to what the error means.  My Issue is as follows.
I draft in CAD a 2d profile (Draftsight).  Save profile in DXF format.  Open file in DXF2GCODE adjust settings to my profile and export to a GCODE file (.ngc).  Then open ngc file in universal gcode sender and run it.  The error Id33 shows up and I search and come here for a solution.

With what I learned here and the possibility of truncating I did the exact same process as above only before I saved in cad to dxf I **regenerated or rebuild** profile in the cad software.
My problem was solved and it was the 2d cad software truncating just one of two arcs in my drawing.
I am verry happy with the previous people who took time to ask and answer the question.

gwag18, 2018-05-27:
What is it mean in bellow message in G-CODE?
error:invalid gcode ID:23 [for M300 S30]

pedrotorres25, 2018-05-27:
Erro 33 in gcode

This is from the program you draw circles or arc and curves.  That program
to save memory turns the cirlce or curves into a bunch of little strait
lines to save memory until you zoom back into it.  That is called
truncating.  The truncating will produce error 33 in gcode.

What software are  you using from drawing file to software to create tool
path?

I use Draftsight as CAD and then save as DXF file.  I then open DXF fiel in
Inkscape and then create toolpath to export as gcode file.  I have not had
a problem with id33.

Pedro

iyes87, 2019-09-05:
long decimals in your g-code, please your compress decimal digit 5 or 6 under coma.
or you can in website to compress your [gcode here](https://riaupart.com/index.php/gcode_decimal_converter)

ClarkDailey, 2019-12-07:
I was getting error 33 using GRBL and CamBam as the g-code generator.
The resolution was that the CamBam GRBL post processor was not generating enough decimal places.  The CamBam GRBL post processor has a setting called "Number Format" which I set to 0.000#. This generated enough decimal places to eliminate the error 33 message from the Arduino GRBL processor.

MeJasonT, 2019-12-07:
You might also want to check your CAD drawing, if you have a tiny fragment of a polyline floating about grbl will give you the same error - it sees the the line in error with the path its supposed to be following.  Grbl is not that clever at error checking so i have no idea why but it knows.
For example if i have designed a part as a 3D solid then exploded the part to give me a 2D drawing to import as a DXF into cambam, if i haven't removed all the under layers and a sneaky line is hiding.  when i come to select the lines and join them again in cambam in preparation for creating a profile, it will let you select the lines including the hidden line - error is in the making.  its also possible for you to have a tiny gap between two lines/shape and cambam will still produce a profile - say its in the middle of an arc, then, there be dragons.  If you are using a home built plasma table this error 33 thing has you thinking the worst, like communication crashes and stuff.  Having spent some time studying the error when it happened, i no longer blame grbl and have come to the conclusion its actually a solid bit of code.  The only errors i get now are human errors.

MeJasonT, 2019-12-07:
check your drawing for stray polylines or gaps in shapes where nodes are supposed to be connected.
issue has been answered above by @chamnit  and others.

shheshmat, 2022-11-26:
hi
I was getting error 33 using GRBL
S1000
G1F5
G17 G21 G91
G1 X0. Y0.
M3
G2 X10.5 Y10.5 I10.5 J0. F5
G2 X20.5 Y0. I0 J-10.5 F5     ******
G2 X10.5 Y-10.5 I-10.5 J0 F5   *****
G2 X0 Y0 I0 J10.5
M5

line ******