grbl/grbl #620: Acceleration rates causes barf...

timryder, 2015-03-06:
Ok so i'm doing some testing with my hardware and I've been experiencing some barf of Grbl when I perform lots of small linear motions with higher accel decel rates.  I have 120=600 110=1800

With these settings and lots of tiny fast motions changing directions grbl will just lock up and not respond. I send it a gcode and do not get any response.  I have to soft restart it.  BUT if i change the acceleration rates from 600 down to 500 it actually works just fine. 

Thoughts?

chamnit, 2015-03-06:
@timryder : I haven't had problems with that settings range. I have 600 mm/min^2 and 6000 mm/min max speed as a default for one of my machines. I always test machine settings with a g-code program that has a lot of small line segments in it. I suppose it depends on your Grbl settings, if the stream from the GUI is keeping up, and how small the line segments are you are sending. Please post your Grbl settings.

There are also some other tweaks in the config.h file that can help optimize Grbl to your system. If the accelerations are very high, you may want to increase the acceleration ticks per second (say 120-150 and your segment buffer size to 8 to compensate) and increase the serial baud rate to 250,000 to help increase throughput.

timryder, 2015-03-06:
#define DEFAULT_X_STEPS_PER_MM 640.0
  #define DEFAULT_Y_STEPS_PER_MM 800.0
  #define DEFAULT_Z_STEPS_PER_MM 640.0
  #define DEFAULT_X_MAX_RATE 1600.0 // mm/min
  #define DEFAULT_Y_MAX_RATE 1600.0 // mm/min
  #define DEFAULT_Z_MAX_RATE 1600.0 // mm/min
  #define DEFAULT_X_ACCELERATION (600.0*60*60) // 10*60*60 mm/min^2 = 10 mm/sec^2
  #define DEFAULT_Y_ACCELERATION (600.0*60*60) // 10*60*60 mm/min^2 = 10 mm/sec^2
  #define DEFAULT_Z_ACCELERATION (600.0*60*60) // 10*60*60 mm/min^2 = 10 mm/sec^2
  #define DEFAULT_X_MAX_TRAVEL 80.0 // mm
  #define DEFAULT_Y_MAX_TRAVEL 50.0 // mm
  #define DEFAULT_Z_MAX_TRAVEL 200.0 // mm
  #define DEFAULT_STEP_PULSE_MICROSECONDS 10
  #define DEFAULT_STEPPING_INVERT_MASK 0
  #define DEFAULT_DIRECTION_INVERT_MASK (1<<Y_AXIS)
  #define DEFAULT_STEPPER_IDLE_LOCK_TIME 0 // msec (0-254, 255 keeps steppers enabled)
  #define DEFAULT_STATUS_REPORT_MASK ((BITFLAG_RT_STATUS_MACHINE_POSITION)|(BITFLAG_RT_STATUS_WORK_POSITION))
  #define DEFAULT_JUNCTION_DEVIATION 0.02 // mm
  #define DEFAULT_ARC_TOLERANCE 0.002 // mm
  #define DEFAULT_REPORT_INCHES 1 // false
  #define DEFAULT_AUTO_START 1 // true
  #define DEFAULT_INVERT_ST_ENABLE 0 // false
  #define DEFAULT_INVERT_LIMIT_PINS 0 // false
  #define DEFAULT_SOFT_LIMIT_ENABLE 0 // false
  #define DEFAULT_HARD_LIMIT_ENABLE 0  // false
  #define DEFAULT_HOMING_ENABLE 1  // false
  #define DEFAULT_HOMING_DIR_MASK (1<<Y_AXIS) // move positive dir
  #define DEFAULT_HOMING_FEED_RATE 1200.0 // mm/min
  #define DEFAULT_HOMING_SEEK_RATE 500.0 // mm/min
  #define DEFAULT_HOMING_DEBOUNCE_DELAY 0 // msec (0-65k)
  #define DEFAULT_HOMING_PULLOFF 0 // mm

Do me a favor can you run this small snippet of Gcode on your system?

//

G20
G90
M9
F80.0000
G0 X1.4080 Y1.7509
M8
G4 P0.02
G2 X0.6178 Y1.1362 I-0.0500 J-0.7510
M9
G4 P0.02
G0 X0.1845 Y1.1884
M8
G4 P0.02
G1 X0.2306 Y1.2370
G1 X0.2331 Y1.1884
M9
G4 P0.02
G0 X0.2321 Y1.2079
M8
G4 P0.02
G1 X0.2029 Y1.2079
M9
G4 P0.02
G0 X0.2483 Y1.1884
M8
G4 P0.02
G1 X0.2702 Y1.2370
G1 X0.2824 Y1.2370
G1 X0.2837 Y1.2369
G1 X0.2849 Y1.2365
G1 X0.2860 Y1.2361
G1 X0.2869 Y1.2354
G1 X0.2876 Y1.2346
G1 X0.2882 Y1.2336
G1 X0.2885 Y1.2325
G1 X0.2886 Y1.2313
G1 X0.2885 Y1.2300
G1 X0.2882 Y1.2286
G1 X0.2877 Y1.2273
G1 X0.2870 Y1.2259
G1 X0.2861 Y1.2245
G1 X0.2850 Y1.2232
G1 X0.2838 Y1.2220
G1 X0.2824 Y1.2209
G1 X0.2810 Y1.2199
G1 X0.2796 Y1.2191
G1 X0.2780 Y1.2184
G1 X0.2765 Y1.2180
G1 X0.2750 Y1.2177
G1 X0.2736 Y1.2176
G1 X0.2752 Y1.2175
G1 X0.2767 Y1.2172
G1 X0.2781 Y1.2167
G1 X0.2793 Y1.2161
G1 X0.2803 Y1.2153
G1 X0.2813 Y1.2143
G1 X0.2820 Y1.2133
G1 X0.2825 Y1.2121
G1 X0.2829 Y1.2107
G1 X0.2830 Y1.2093
G1 X0.2829 Y1.2078
G1 X0.2827 Y1.2062
G1 X0.2823 Y1.2046
G1 X0.2816 Y1.2030
G1 X0.2808 Y1.2014
G1 X0.2798 Y1.1997
G1 X0.2786 Y1.1981
G1 X0.2773 Y1.1967
G1 X0.2759 Y1.1952
G1 X0.2743 Y1.1939
G1 X0.2727 Y1.1927
G1 X0.2710 Y1.1916
G1 X0.2692 Y1.1906
G1 X0.2675 Y1.1899
G1 X0.2657 Y1.1893
G1 X0.2639 Y1.1888
 //

chamnit, 2015-03-06:
@timryder : Thanks. I'll give it try when I get home this evening. Ping me tomorrow to remind me, if I forgot. Just from looking at this though, your settings don't seem unusual. Nor the g-code, although the movements are super small at 0.001" or so. What are you doing with your machine that requires this accuracy?

chamnit, 2015-03-07:
@timryder : Works fine. Check your `$$` settings to make sure that they updated to what you intended. Grbl doesn't automatically do this with the default settings.

J-Dunn, 2015-03-18:
The fact that this happens on short movements suggests it is when the trapezoid is reduced to a triangle. This is where it has to go from max +ve accel to max -ve accel in one step. 

I know there's very little space for nice smooth S-curve type acceleration but would some soft-start technique be possible?  I'm thinking of a short a/2 segment, this woud reduce a lot of mechanical stress too. 

At the very, very least it would seem desirable to have a minimum of one step trapezoid and never a triangle. That would change the reversal to a->0; 0-> -a instead of +a -> -a , ie half the shock. 

If there is a possible software glitch during these exteme changes is would probably improve the software too.

chamnit, 2015-03-18:
@J-Dunn : It works on my end, not sure what the status is on @timryder 's machine. I spent a lot of time with v0.9's stability and made sure to account for any potential short movement problems that you mentioned. Grbl should be robust for the expected range of motions that a CNC mill will encounter. There is a valid range that is partly due to numerical round-off of floating point and integer step value handling. That can't be avoided. There are also already a number of techniques that help round-off triangle velocity profile peaks and such. This is the reason why there is such a difference between Grbl v0.9 and v0.8 and why v0.9 can achieve such high rates through complex tool paths without losing steps. To me, it's solved. 

As for S-curves, look-ahead planning of S-curves is a lot more complex than people realize. It's one of the reasons why LinuxCNC still doesn't have it. TinyG states they do S-curve planning, but it works only within a small window of assumptions and tries to brute force the rest. But, don't worry, I'm working something special that I don't want to give away just yet.

J-Dunn, 2015-03-18:
OK, thanks for the explanations. If there are already methods to soften the edges that's great. I'm building a rather large CNC router machine and that was a concern with the kind of mass I'm going to be moving. 

In fact that was one of reasons I'd originally thought grbl was not sophisticated enough. Looks like with v0.9 that may not be the case. 

In fact, I suspect cosine curve would be the theoretical ideal rather than parabolic S-curve. That may be feasible to calculate with two or three terms of a Taylor series or, if not, using a look up table. There likely won't be room for that in Arduino though. 

I may well have a try with v0.9 and by the time I've got my hardware more completed, hopefully we will have more news about your "surprise".  Should be good.

timryder, 2015-03-18:
I'm still experiencing the issue, but I'm trying to pinpoint it elsewhere... I realized the Gcode section I gave you was the output of my software streamwriter up until it stopped... so I need to make another output which is complete and politely ask you to use it yourself again.  

Also @chamnit i'm curious what actual hardware are YOU using when you dev and debug? the Uno? what baud? and what sender or UI?

chamnit, 2015-03-18:
@J-Dunn : Mass shouldn't matter. There are huge, heavy 4'x8' wooden CNCs running Grbl. Acceleration is acceleration and S-curves arguably won't help with performance. S-curve motion profiles actually trade high-end performance with smoother transitions. But, what S-curves do help with is the reduction of exciting resonant modes. If your machine isn't very stiff and really fast, like a pick n place machine, S-curves profiles will definitely help. For a CNC mill, the machine should be stiff enough already to account for cutting forces such that resonant modes are a non-issue.

@timryder : I compile and upload with the Arduino IDE backend tools(same used in direct IDE compiling) to a stock Arduino Uno at 115200 baud. I typically use the Grbl Python streaming script to test g-code programs, as it's minimal, reliable, and to spec. I occasionally use GrblPanel or UGS for testing (bCNC also seems nice but haven't used it on a machine yet). I don't use a debugger or Atmel Studio. I've been told many times that I should, but I've always written code in a straight text editor and work the command line. I suppose old habits are hard to break.

chamnit, 2015-03-18:
@J-Dunn : As far as porting to something other than an Arduino Uno, you can immediately compile Grbl to an Arduino Mega2560 by changing the cpu map define in config.h. This will give you plenty of room to add stuff, but the Mega isn't any faster. Floating point math is really expensive on these 8-bit controllers and have to be minimized as much as possible. There are a few other ports of Grbl to ARM. One to a Teensy 3.1 and another on a Tiva Launchpad, but I can't recall how far they got. From my understanding, it's was fairly trivial to port.

J-Dunn, 2015-03-18:
chamnit , I agree about rigidity, it was mechanical stress and wear that I'd like to minimise. High rigidity plays against that: a bit of flex or toothed belts all helps to soften mechanical shocks.

J-Dunn, 2015-03-18:
chamnit says: "There are also already a number of techniques that help round-off triangle velocity profile peaks and such. "

stepper.c says:
/*    BLOCK VELOCITY PROFILE DEFINITION 
The planner block buffer is planned assuming constant acceleration velocity profiles and are ....

It seems that the comments need to be updated if that is not longer the case. 

The code is well documented but the AMASS thing looks like it needs a bit of head scratching to follow. Some write up of how this is now done would be a great help in following the code. ;)

chamnit, 2015-03-18:
@J-Dunn :  Grbl is still constant acceleration. The trapezoid/triangle peak rounding occurs in the step segment generation by its design. It doesn't do a true rounding since that would be an S-curve.

timryder, 2015-03-18:
I have a question unrelated to the thread... I decided to change my streaming code to allow for ? during the streaming of the gcode as well.. In doing so I noticed that when it barfs the status of GRBL goes to "Queue"...... Why? how do I prevent? Should I be sending my stream differently?

chamnit, 2015-03-18:
@timryder : Not sure without looking at your streaming code and understanding what it's doing. Try using the new v0.9i master. It gets rid of the QUEUE state for reasons such as this one.

timryder, 2015-03-18:
Does v0.9g,h,i still have the mapped pins for A0 A1 A2  Start,Hold,Abort? Are they pulled down to ground and a High input triggers them or reverse?  Also is the feature to enable the internal pull-ups for those pins specified somewhere or can would it be an issue to do it? I'm wondering if i'm getting Noise somehow on my custom board which might be causing the issue for me..... 

I will update to 0.9i though thanks!

chamnit, 2015-03-18:
@timryder : Everything is the same, except variable spindle is enabled by default (D11 and D12 swap). You can disable it in config.h. In v0.9i, all of the input pins are configurable when using some of the new config.h options, as high or low trigger and with the internal pull-ups enabled or disabled. By default, the internal pull-ups are enabled and assumes low as trigger for the control pins. You can flip it to a high trigger via config.h.

timryder, 2015-03-18:
@chamnit Thanks for this!  I'm actually not using a Z-axis in my system I'm using just X and Y. I'll alter the code to what I need it to work.

I'm getting all kinds of RUN, IDLE, QUEUE throughout my streaming process.. I may have cracked the shell on all my issues hopefully. I current have C0 C1 C2 just isolated from the rest of the circuit board, they aren't connected to ground but i'm thinking of using a 10k and pulling them down, or just disabling them in the firmware all together for my efforts. I'll have to kick it around.

cri-s, 2015-03-19:
For speed, how many speed you want ?
I'm running older code, never viewed at code after revision g, but
worst case for
stepper interrupt for 6 axis is 25.6 uS at 16Mhz for ramps configuration.
Adding 7.5uS for other interrupts, this is 30khz stepping frequences.
If there is no need to load buffers, then interrupt time is 10uS shorter.
Whit addition of two queue slots inside stepper.c, the stepping
frequence go over 100khz.
Increasing this slots adds frequences too.
The drawback is, that because of timing requirements, for shorter step
pulse length,
the step pulse need to be hardcoded inside interrupt and configured
using additional
config field.
If adding more slots, frequence increases.
Further, grbl compile nicely on stm32 too, this boost frequences
presumly to the limits of the
hw driver, as example 500khz for the A4988 stepstic driver using 1uS
for pulse width and
minimal 2uS interval for succesive steps. The real question is, do you
have motors for that
doing over 156 rotations per seconds  at 16x microstepping or 300 rps
at 8x microstepping.

timryder, 2015-03-19:
I updated my codebase to v0.9i and so far i've had great success eliminating my issues.  Although during an endurance test of pushing my system to the max speed and with various types of moves long and short, I encountered a weird behavior where the system I believe interpreted a decimal place in the wrong position or perhaps translated it as mm instead of in.  basically what I saw was the machine moving positive on the y to a VERY large position outside of my envelope.. as soon as it got there it continued to move back down towards the correct positioning but I powered off the motors and let the communication continue to run so it didn't hurt the machine and I could capture the output log. 

Here is a link to the transaction where I'm sending the commands and I record the exchange. This circumstance has happened to me twice so far this morning... again my protocol sends and waits for  an "OK" response before sending again. 

http://www.cmt-engineering.com/StrangeOutput.txt

Can you have a look for me and offer any thoughts?

ashelly, 2015-03-19:
So what's preventing the '?" from getting inserted in the middle of a G1 string, and causing the parser to reject all or part of the coordinates?  The `Receive: error: Invalid gcode ID:24` lines are suspicous.

timryder, 2015-03-19:
@ashelly The windows mscorlib allows you to send a Line at a time and as many threads as they want within the same application and send them.  If I use the "WriteLine" as opposed to just "Write" then it automatically handles the threads incoming requests and sends them in sequence never allows an insertion during the middle of an existing transmission.  FIFO stack.

chamnit, 2015-03-19:
@timryder : A error 24 is a g-code block that has two commands that use axis (XYZ) words. From your g-code, there is nothing other than G1's there. This leads me to believe that the stream is getting screwed up for some reason.

timryder, 2015-03-19:
@chamnit Would you imagine that it happens on my end? That is to say from the UI? Also what do you make of the machine running away outside of it's envelope when clearly I haven't issue a GCode whose value is anywhere near that high... if you look at the Machine status you'll see that it is counting way up to 140 inches on the Y axis. I never commanded anywhere near that.

For testing i'll temporarily disable the thread which does the status query. Do you guys have any sample code someone already implemented which allows for query of status during the streaming?

chamnit, 2015-03-19:
@timryder : It's likely that the stream is also corrupting the values. You can just place '?' characters in the g-code program to have Grbl report back without the extra thread.

timryder, 2015-03-19:
Would it work just fine if I append the string to have a "?"

chamnit, 2015-03-19:
Yes. And I've added an 'echo' option to my to-do list to help problem solve things like this.

timryder, 2015-03-19:
A compile time option of an Echo would be amazing!

LotusPack, 2015-03-19:
Think this is a question for Sonny, but going on the awareness of the code I see in these posts, perhaps anyone may be able to enlighten me. My question is will GRBL allow for a feed rate increase while machining? For example G01 X120,000 F120.0  will take a minute to complete if during that time the machine starts to shake and possibly run a risk on breaking a small tool, can I send a command to slow the federate down?

chamnit, 2015-03-19:
@LotusPack : Not at the moment, but I did write v0.9 in a way to allow feedrate overrides in realtime. There are some minor details I need to work out. This is the main feature to debut with v1.0.

timryder, 2015-03-25:
Gentlemen.  I'm very sorry for the headaches I've caused... i figured out FINALLY all of my issues were caused by a logging feature I use in my custom UI was eating up soo much of the CPU in the computer that it would cause it to barf... grbl is running great now and there's no issues :)

Thank you guys for your support!  Awesome project.

closed 2015-03-25 by timryder