grbl/grbl #621: Is there a way to force Grbl back into alarm mode?

markbotics, 2015-03-07:
Hi,

After Grbl has completed a homing sequence, a low signal on the A0 Reset/Abort* signal does not make the user re-home.  In our situation, the wire connected to Reset/Abort* also kills the power, which causes at least one axis to "droop" and move by gravity, and so the MCS is not correct.   

Unless I missed it, I think the only way to make the microcontroller go back into 'ALARM' state and require another homing sequence if if we wire to the microcontroller reset (sometimes on a push button on Uno board), but I;m not sure if that is appropriate.  But it would be nice if there existed some way to get it to know it has lost the MSC position after Reset/Abort* has been pulled low.

 in our machine we have an external E-stop switch wired to both Reset/Abort* and our own TTL logic flip-flop cct that kills the power in the event of emergency, and keeps it off until the user presses an 'unlock' button.  But the micro doesn't this happened except for the Reset/Abort*.

Any ideas or settings that could help?  
Thanks

chamnit, 2015-03-07:
Use the hard reset pin on the MCU for estop. Not the soft reset/abort pin. It's not intended for this use.

markbotics, 2015-03-12:
Thanks @chamnit , that's what we did.  Our E-stop flip-flop is now connected to hard reset pin, so when the flip-flop is in stopped state (visible with red LED) the MCU is in reset, and after user restarts (opposite button from E-stop) the MCU wakes up as if the power was just turned on, which is what we want.

closed 2015-05-17 by chamnit