grbl/grbl #622: Homing cycle not recognizing limits

corinrose, 2015-03-10:
I'm running using an Arduino Nano and this kit http://www.zentoolworks.com/product_info.php?products_id=133 and some motor drivers I found around for pretty cheap. 

Whenever I run the homing cycle, it (as always) starts with the Z axis but does not stop once it hits the limit switch. I've tested all of the limit switches by running GRBL outside of the homing cycle and hitting the switches, it always acknowledges the hit and asks for a reset, but during the homing cycle it just keeps going. I was having this problem on and off in the past but I recently rewired the limit switches so as to prevent interference and upon reuploading GRBL to the Arduino, and the homing cycle no longer works at all.

Here's my settings:
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=6 (dir port invert mask:00000110)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.020 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$14=1 (auto start, bool)
$20=0 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=2800.000 (x, step/mm)
$101=2800.000 (y, step/mm)
$102=2800.000 (z, step/mm)
$110=500.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=250.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=450.000 (x max travel, mm)
$131=450.000 (y max travel, mm)
$132=450.000 (z max travel, mm)

chamnit, 2015-03-10:
@corinrose : Make sure you don't have a version of Grbl that has variable spindle enabled. It's the default in the edge branch at the moment. Try recompiling Grbl with it disabled in the config.h file.

corinrose, 2015-03-13:
the #define VARIABLE_SPINDLE is commented out in the config file, so that's not the issue

chamnit, 2015-03-13:
@corinrose : Hmm, I was just working on the homing code last weekend and tested it. It was operating as it should have. There is a new compile-option that shows the limit switch state in the status reports. You can enable that to see if Grbl is reading the pins correctly. (It's a little user-unfriendly at the moment. I'll be fixing that before master release.)

corinrose, 2015-03-13:
alright ill check that out, thanks for the help!