grbl/grbl #623: Connect serial from another arduino as a controller

peluzza, 2015-03-11:
Hi! I'm wondering about make a human controller for my mill. I already know that Grbl can't do it, as it is a g-code interpreter. 

BUT my idea is to use another arduino as a proxy. If manual mode is on, 3 encoder pots controlls the mill, if manual mode is of, forwards serial to Grbl.

Could it be possible to do it in that way?

Thanks!

chamnit, 2015-03-11:
@peluzza : Absolutely! You simply need to connect the second Arduino's serial TX/RX lines to Grbl's RX/TX inputs. This has always been something I wanted to do, but never found time. The idea was to write a very simple .INO program in the Arduino IDE, using their libraries, to support an SD card, LCD, and some control pins. In this way you can make CNC setup without a computer.

peluzza, 2015-03-11:
Wow, then, it's time to put hands on keyboard and start to code ;)

I found an old thread about it here: 
http://letsmakerobots.com/content/can-one-arduino-resent-serial-com-another-arduino

If anyone finds/knows and already done project like this, please post it here. Thanks!

chamnit, 2015-03-11:
Also, this applies to others as well, please share the source code! I would love to make something like this a supported project.

Norwegian-Gadgetman, 2015-03-11:
Wouldn't a trackball or similar be a better option?

With pots you have what? 270degrees of adjustment, and a 5 - 10% error of positioning, which makes it rather difficult position accurately.
Also, it would require configuring to match the size of the machine.

With trackballs or other encoder-based tech(a bunch of sroll-wheels from computer mice?), you can scroll and have it move a set multiplier. Or you could have an additional 'sensitivity' control...
Also, it should transate easily into sending a continous stream of Reative coordinate system(G91) commands.

twforeman, 2015-03-11:
You should take a look at this option http://www.shapeoko.com/forum/viewtopic.php?f=7&t=5505

This intrigues me quite a bit and I may play with it (when I have no other projects... hah.)

peluzza, 2015-03-11:
@Norwegian-Gadgetman Why pots? Simple, rotary encoders usually has 20 steps and 360º without lock position, so are so much accurated than a track or joystick. Mine are EC11, which are momentary push button too, so my idea is to switch microsteps,step,revolutions by clicking, and simply rotate as much steps as I want, just like a G code does (A 3D Etch-A-Sketch!!!)

It would be so much usefull for an accurate set of custom zero positions (just think about a laser attached to it).

cri-s, 2015-03-11:
just soft limit don't work anymore, and some issues with coordinate offsets.
If instead pedant is handled by gui, then things are better, but
outside of the scope of grbl.

peluzza, 2015-03-11:
@cri-s I meant fine adjusting of zero position before start milling, I use it for fixing or improving already done parts. For me, fine tuning and manual control are a must have.

Norwegian-Gadgetman, 2015-03-11:
Peluzza, please don't call the EC11 a 'pot' as that's short for 'Potentiometer', whih is a variable resistor.

I assumed you meant the variable resistor such as you'd find used as a volume control in an amplifier. 

Mixing concepts just muddies the waters and wastes time.

peluzza, 2015-03-11:
@Norwegian-Gadgetman Well that was a simple shortcut after saying "encoder pots". Sorry if I confused you. I always meant to say rotary encoders with momentary push button.

JanJansen47, 2015-03-13:
Currently building an arduino Mega controlling an UNO (Grbl).  I have a working solution and want to extend it to  fully use the I/O buffer capability as described for Grbl.  I also developed a passthrough for the Universal gcode Sender.  Gcode files are stored, line by line, on SD. Just discovered I have stored multiple Gcodes in one "line" on SD spoiling the robustness of my solution.

tchepperz, 2015-03-16:
@chamnit 
Hi! I have something like that : I'm working on an embbeded cnc controller.
It's actually working with an arduino uno as grbl support, and a Mega to support LCD touchscreen which include SD card reader. It can generate gcode from user input for specifics purposes, save it in a gcode file on SD and stream it via serial from mega to grbl. Or I can just put gcode file on sd from 
computer, put it in SD reader an stream the file to grbl. All of this is actually working.
Just little issues to manage grbl returns.
I would be glad to share it, just let me clean up my code..
grbl is an amazing tool and make me save a lot of time, thank you.

JanJansen47, 2015-03-16:
My email address is: 

NL01871@gmail.com

Date: Mon, 16 Mar 2015 02:01:16 -0700

tchepperz, 2015-03-19:
Hi here is my code to stream a gcode file from an SD card via serial between two arduino.
If it can helps 
https://github.com/tchepperz/ArdSketch

peluzza, 2015-03-19:
Wow! Ir helps a lot! Thanks @tchepperz

chamnit, 2015-03-20:
@tchepperz : Nice! Do you off chance know if this will fit and run on a 328p Uno-type controller?

tchepperz, 2015-03-20:
@chamnit 
I think it won't, problem is that I wrote it for a Mega which has different serial ports. I use Serial to communicate with user via monitor, and Serial1 to communicate with grbl.
On Uno you just have one Serial port so there will be a compilation problem with calls to Serial1.

But with some modification it could run on Uno, without using serial monitor as an interface but just for 
grbl. Program could be modified to directly open and stream to grbl a specified file on SD. That will works.

If you want user interface, solution is an lcd touchscreen. On my mill I use a lcd interface and so, only one serial port for grbl. But I'm not sure that you have enough pin on Uno to manage lcd, touch, sd and serial.

JanJansen47, 2015-03-20:
Yes use the mega because of the memory requirements for SD and a graphics touch display.  Unfortunately cannot share code. I advice to have a better look at the grbl communications.  E.g read the universal gcode sender Java stuff.
Continue the good works.
Jan Jansen
Sagittalaan 22 5632al 
+31688806529

cri-s, 2015-03-20:
For compiling it on arduino uno, use this lib:
http://www.pjrc.com/teensy/td_libs_AltSoftSerial.html

After the include lines, add this two line:

#include <AltSoftSerial.h>
AltSoftSerial Serial1;

The resulting size is above 16k.

tchepperz, 2015-03-21:
@cri-s  Thanks 
AltSoftSerial seems to make it run with Uno. I couldn't test all the program because I don't have a SD reader for my uno but compilate without any problem.Just add lines as said cri-s and modify 
Serial1.begin(115200) to Serial1.begin(57600) .
Pin 9 become TX and Pin 8 RX for Serial1... @chamnit

cri-s, 2015-03-21:
The lib works without problems at 115200 , just use arduino version 1.0.1
or newer for compiling. The actual version is 1.6.1
What is needed to change is this line, as alrady described in the comments.
while(!SD.begin(53)){
And for records, the Sparkfun SD shield is incompatible with this solution
for
the arduino uno. As i know, it's the only incompatible sd shield.

benha, 2015-04-01:
This thread is somewhat on-point for a project I want to get off the ground so I thought I'd pile on rather than opening a new thread...

I want to build a controller for a rotary table to allow it to index positions around a circle.  (See the rather pricey "division master" to get an idea how this works.)

The gcode needed is extremely straightforward and, with the exception of the distance between stops, is the same every time.

Clearly I could use two arduinos - one that has a keypad and a display that performs UI tasks, calculates the required gcode and then streams it to a GRBL arduino over serial.

That just seems wasteful though.

Is there a straightforward way to perform both of these tasks from a single arduino?  In other words: it operates in "UI mode" until the gcode is built, and then stores the gcode, converts to "GRBL mode" and reads the gcode / moves the axis?

Sorry if this is an absurd question.  I'm very new to GRBL and arduino so I'm still figuring things out.

Best,
-Ben

chamnit, 2015-04-01:
@benha : It's possible to integrate everything on an Arduino Mega, not the Uno for flash and memory reasons. 

In general, I would suggest doing the second Arduino or RaspberryPi approach. Although it seems wasteful, you will not have to worry about keeping up to date with the main repo and can run Grbl on a $10 Arduino Micro (could be found even cheaper). All you need to do is update the communication protocol if it changes. 

You also don't have to worry about breaking something in the Grbl code. Even though it's written and commented well, you can easily do things that compromise the integrity and the performance of the code, if you don't know what you're doing.

benha, 2015-04-01:
Okay.  That's kind of what I expected, but I wanted to double check.

Thanks!

cri-s, 2015-04-01:
Keyboard could be connected to single io pin, it's not a problem, the
problem lies more on
the required pins for display, if not using i2c display.
Altrought, if you do want having only single axis, this gives you 6
pins, even more if you
don't use spindle, ... . So for that purposes it seems possible.
@Sonny_Jeon , As you have switched to arduino, it is possible to
officially include one function named idle as example, that are called
periodically when there is no planning buffer or
if some planning buffer is used, only if there is no motion, halt or error.
Including dummy function using __attribute__((weak)); allows to extend
the lib better.
The same should be done for init, and maybe instead of idle, usr_init,
usr_idle, usr_loop .
gc_execute_line() should call usr_loop() before returning, if no abort
state exist.
This allows to inject commands from i2c or sd card .

chamnit, 2015-04-01:
@cri-s : I'm not sure what you mean by switching to Arduino. Grbl has always run on the Arduino Uno. As for usr loops, like you ask, I have no intent of supporting anything like right now. I've got lots of other things to do first.

sundale, 2017-02-23:
I have tried, but the result is very odd: **I need some help!**

Grbl (v1.1e) is flashed on UNO(Serial), and connected to Leonardo(Serial1), using RX and TX pins.
And both board are connected to PC using USB, to check the communication status.
1) When I type $I command on Leonardo USB serial input, the Grbl version information is shown on UNO serial output window. But, Leonardo does not display the version info at all.
2) When I type $I on UNO serial input, no information is displayed on both serial output window.

It seems the two serial interlocking each other, or It works like one serial port in two windows.

Anyone can explain why it is so, and solution?

Leonardo code:

void setup() {
  Serial.begin(115200);
  while (!Serial) {}
  Serial1.begin(115200);
  while (!Serial1) {}
  Serial.println("<Arduino is ready>");
}

void loop() {
  char rc = 0;
  if (Serial1.available() > 0) {
    rc = Serial1.read();
	Serial.write(rc);
  }

  if (Serial.available() > 0) {
    rc = Serial.read();
    Serial1.write(rc);
  }
}
`

tchepperz, 2017-02-23:
I am not really in this anymore, but I can remember that if I tried to send grbl command to uno via usb  it doesn't work when serial rx tx were connected to my mega. So I was used to disconnect uno and mega serial (rx tx) to use grbl (uno) with usb. I couldn't explain precisely but I have red that usb works as serial communication on arduino, and uno have only one Serial possibility so you have to choose rx tx or usb but both can't work together. Leonardo or mega is different because there is many serial possibilities
Hope it helps..

sundale, 2017-02-23:
Thank you tchepperz.
I have tried what you have suggested. And the result is same - Leonardo does not receive data.
I have tried on beetle(small packaged Leonardo) board. It seems send and receive, again the result is not usable because half of the data is lost. This could be hardware related issue. Since grbl baud rate is fixed to 115200, I cannot test with lower baud rate. I will try with different combination of boards.

vMeph, 2017-02-24:
Here is a project of arduino using a second arduino to display a lcd, o hope it helps
https://github.com/grbl/grbl/issues/1125

sundale, 2017-02-25:
Thanks vMeph.
**Yes, it is helpful very much** that I have found the reason why my connection is not working, from the link. 
I have followed LCD on GRBL forum, and read the schematic that uses an analog IC for switching.
It will take time to get the IC, but sure that I can make it.

> The scheme is not very complicated. The IC CD4066 quad bilateral switch having only a gate used. This electronic switch is, when the MP does not need the TxD line on GRBL RxD, to isolate the line. Without this isolation, another transmitter into the USB port of the GP could no longer transmit. Indeed, if we look at the data transmissions diagram of a Nano (UNO is the same) USB data passes through a 1k. resistor and therefore are lower than ours and will, without insulation, crushed.

vMeph, 2017-02-25:
That will work the lcd with out the ic 4060 you just dont use the tx of the leonardo conected to the rx of the grbl arduino

b33zwaXer, 2018-08-23:
I am doing the same thing. I have GRBL uploaded onto an UNO and I have a MEGA controlling a LCD, Keypad, and a SD. I using the MEGA as a GRBL controller. I am having some trouble with the serial communication, maybe someone can shine some light. 

I have the mega Tx->Uno Rx and mega Rx -> Uno Rx. So they should be talking fine but, it seems like there is only one way communication.  I am able to send a Serial.println("$H") from the Mega to the Uno (over Pin 0,1 not USB), and GRBL will receive and run the command, but not send back the "ok" . But I look at the Uno's USB serial output I see that after the $H command has been sent, the GRBL acknowledges the command by Printing "ok" in my Arduino terminal, but the MEGA does not receive this ok over the pins??? 

What am I doing wrong.  Ideally I would have the Uno as a Master Device that can receive G-code from my Mega Controller, and can receive G-code from a PC when it is connected over USB.

Anyone out there doing this yet? Suggestions ? Tips?

algo1980, 2018-08-24:
Why do you wanna use it as a separate controller

b33zwaXer, 2018-08-24:
I ended up solving my problem.  I connected the mega serial3 rx tx to the uno/grbl serial Rx tx. Now I use the mega as the master controller by sending(bouncing) the read/write of mega serial3 to mega serial. Now I can use the keypad connected to the mega to send gcode to uno/grbl as simple stand alone system or I can connect the mega to a pc and send gcode “directly to” (bounced) to the uno.

algo1980, 2018-08-24:
it is stupid to use mega and uno together when you have so much to do with
mega alone.

b33zwaXer, 2018-08-24:
Why is it stupid? It makes things way less complicated, and simplicity is one thing I strive for. Why would I hack my way through smashing together GRBL, an  LCD, a  Keypad, and a SD card reader all into one mega. It is way easier to use an additional arduino as a dedicated grbl device. To me it seems like this is the intended use of grbl. It acts as a stand alone system, you send it g-code, it does what it is supposed to due.  Why would I obfuscate this well oiled machine?  You sound strongly opinionated about this, why should I put it all in the mega alone? or maybe I should just use a RPi!

algo1980, 2018-08-24:
It is stupid to engineer things this way.

>

chamnit, 2018-08-24:
@algo1980 : It's not kind to down people. It's not helpful to anyone. Please don't do it here.

b33zwaXer, 2018-08-24:
Hello sundale,

It is possible to do this without using the relay to interupt the TX line. I think the problem you are having is that you are connecting your USB to the grbl/arduino, which is tying up the TX line on Grbl_Serial.  If you treat the Grbl as a slave to the a master controller( the leo in your case), you can talk directly to grbl through the master. I have the same code as you showed above. The Master controller just bounces the info between master_serial and master_serial1(which is connected to GRBL_serial). Doing this, I can use the master controller to send G-code(stand alone), or I can plug in a PC and use a g-code sender if I like. Hope this helps.

b33zwaXer, 2018-08-24:
Algo, Why? Give reasons please. I am not a software engineer. I do this for fun.

algo1980, 2018-08-24:
My dear if you are doing it for fun please carry on, it will add to your
learning experience. Using two mcu's can add more cost to the system. Mega
is good and capable of doing all tasks but you will need some additions to
the code. How many switches you want to connect with your controller?

koogar, 2019-10-21:
> That will work the lcd with out the ic 4060 you just dont use the tx of the leonardo conected to the rx of the grbl arduino, the ic is only needed in case you wanted to send data from leonardo to the grbl arduino, in this case the lcd will work sense Leonardo is only reciving data.

I used a  1N4148 small signal diode in series on the GRBL RX line then you get bidirectional comms
No IC required

asankakumara, 2019-11-05:
How to control grbl using UART connection with Arduino Uno

opemipoadigun, 2020-12-02:
> I am doing the same thing. I have GRBL uploaded onto an UNO and I have a MEGA controlling a LCD, Keypad, and a SD. I using the MEGA as a GRBL controller. I am having some trouble with the serial [...]

hey would you mind sharing the code you used on your mega to do this?
I have been having some problems trying to get the grbl to respond when I send Serial.print("$H"). However when I type in the command on the serial monitor of the controller arduino the grbl responds

b33zwaXer, 2020-12-02:
In the end I dropped the mega and used universal gcode sender UGS on a Rpi. It worked well!

opemipoadigun, 2020-12-02:
> In the end I dropped the mega and used universal gcode sender UGS on a Rpi. It worked well!

Ohh I see I was just wondering what you wrote to make the grbl commands work when you type in serial.println(“$H”)

buzzshot, 2021-05-13:
Not sure if this helps but i did come across this website.
https://github.com/ChrisAnd1998/GRBL-Pendant-with-LCD

ecncshopInd, 2021-08-03:
Iam using grbl for my CNC. 
I want the spindle to switch off, if serial communication between my PC and Uno stops.
( Iam querying position data with "?" command from PC to grbl every 200 ms. 
So If I disconnect the serial port communication or my PC CNC app crashes, this serial data line traffic will stop).
I want to detect this with another PLC and switch OFF the spindle etc. 
Can I just monitor the Tx pin for this traffice (pulse edges ) without affecting serial communication ? 
Any one tried this ?

ecncshopInd, 2021-10-17:
Iam using grbl for my CNC.
I want the spindle to switch off, if serial communication between my PC and Uno stops.
( Iam querying position data with "?" command from PC to grbl every 200 ms.
So If I disconnect the serial port communication or my PC CNC app crashes, this serial data line traffic will stop).
I want to detect this with another PLC and switch OFF the spindle etc.
Can I just monitor the Tx pin for this traffice (pulse edges ) without affecting serial communication ?
Any one tried this ? Or is there any other way to achieve this ?  Note: Iam talking of a scenario, where USB cable is still physically connected and powering the Uno. Only serial port I have disconnected in software.

ecncshopInd, 2021-10-17:
From PLC I plan to give a DOOR signal to Uno on A1 pin, so spindle will be switched off by Uno itself. Or I can put a PLC controlled relay in series to the Spindle command from Uno etc. But I need to first detect reliably that serial communication has stopped.

MeJasonT, 2021-10-17:
> From PLC I plan to give a DOOR signal to Uno on A1 pin, so spindle will be switched off by Uno itself. Or I can put a PLC controlled relay in series to the Spindle command from Uno etc. But I need to first detect reliably that serial communication has stopped.

This site is an old outdated forum - you need to use Gnea/grbl 
As im here ill try to help you.
The best way to incorporate a safety door operation is to use the feed hold input pin.
there is a comment in the code that can be uncommented to make it a live function withing the code.
The safe door code retracts the tool in Z axis then stops the spindle.  Its stops exactly where it is in code and waits for a cycle start to resume its operation.  its one of the niftiest functions in grbl in my opinion and it works a treat.  Its already in the code.

MeJasonT, 2021-10-17:
The serial comms thing is a nightmare.  I did manage to sniff the serial data to control an external vfd controlled spindle from grbl.  If you connect a rx and gnd wire from another serial device you can listen in on the commands being transmitted to grbl from a pc.  By parsing the received data you can find elements of the command string you are looking for and use the listening device to do somthing with that data.  For instance you could send messages to shop floor software to monitor machine activity (just like the big boys do lol).  It is an added level of complexity which you will find is just a fancy project at the end of the day.  Money is made by the spindle turning and chips flying.  Having gone down the rabbit hole of messing about with controllers i try to now convince people to just trust the controller to do what it does best - make chips.