grbl/grbl #624: Spindle control question

109JB, 2015-03-14:
I have been thinking about the issue of lack of spindle synchronization as it relates to lathe operation and I think I have an idea for smaller lathes t least.  I have a 9 x 20 lathe that I also have a surplus DC motor that I intend to use for a spindle drive. If I were to add an encoder to the motor and drive it using a servo driver such as the Gecko G320X, GRBL would see the motor as simply another step/direction motor like a stepper. All of the encoder and servo feedback control would be internal to the Gecko driver. The motor could then be hooked to the spindle with a timing belt drive so that the motor position could be related to spindle position. Since GRBL is 3-axis only, the normal lathe X and Z axes would take up two of the available axes, leaving the third available to control the spindle motor through the Gecko driver. Since the motor would be seen as just another axis, then coordinated motion would be possible, allowing lathe threading. The issue is how to implement this within GRBL.

For example for a lathe, lets say you set up the spindle as the Y-axis with a distance of 1 inch corresponding to one revolution of the spindle. A G91 command of G1 z-1 y10 would give a 10 thread per inch type of move.  But this isn't the way that lathe threading is normally programmed and doesn't account for non-threading use where you would want the spindle to run continuous at a set rpm. 

This could also be applied to mill spindle synchronization in a similar fashion allowing rigid tapping but would require a fourth "axis".

I think the use of a servo for the spindle drive could provide a way to incorporate spindle synchronization in a fashion that is possibly easier than having an encoder and index pulse mounted to the spindle that GRBL would have to monitor and adjust the other axes for. The servo drive on the spindle would adjust to keep up rather than having to have GRBL adjust the other axes due to spindle speed fluctuation.

I think it should be possible to have the GUI interface handle most of the heavy work with GRBL just keeping track of the spindle position by keeping track of the number of step pulses sent to the driver.. The gui could then convert any type of canned threading cycle into a series of G1 coordinated motion commands. For example. for this G33 threading code snippet:

G90 (absolute distance mode)
G0 X1 Z0.1 (rapid to position)
S100 M3 (start spindle turning)
G33 Z-2 K0.125 (move Z axis to -2 at a rate to equal 0.125 per revolution)
G0 X1.25 (rapid move tool away from work)
Z0.1 (rapid move to starting Z position)
M2 (end program)

The gui could easily calculate that the Z axis has to move a total of 2.1 inches and with a K value of 0.125, there would need to be 2.1/0.125=16.8 revolutions of the spindle. To get 300 rpm out of the spindle if set up for 1 unit=1revolution, the feedrate sent to GRBL would have to take into account the Z axis travel. So, the feedrate sent to GRBL would be (300/16.8)*(16.8^2_2.1^2)^0.5=302.33 IPM and the thread would be cut. GRBL would just see a G1 F302 Z-2 Y16.8 where Y is the spindle servo. I used G33 because it is simple to illustrate.for this you would have to provide a thread relief cut at the end of the travel since the spindle would stop in this example.

The only thing I am not sure of is how you would keep the spindle turning say for non-threading commands.  Maybe a simple binary variable that grbl uses to know if the intent at the time is syncronized motion or not. Or implement a G33 within GRBL that is simply the same as a G01 except that using it tells GRBL to convert M3/M4 SXXX commands into syncronized motion.

Just thought I would throw this out there as a possibly easier way to implement spindle synch. Much of the above is simply rambling while thinking about this. Would something like this be possible?

chamnit, 2015-03-14:
@109JB : The post is long, so I may have missed a few specific points, but yes, using another motor is definitely a way to do spindle sync. The idea isn't new, but the main problem with this concept is that you have to detach your normal spindle and connect the servo/stepper every time you need to do a threading operation. IMO, it's better to have a spindle encoder that provides Grbl the spindle angular position and rate. Grbl would only need to begin motion on time. Plus, your spindle motor is typically much more powerful and should provide enough torque to do the threading operations over an external motor.

109JB, 2015-03-14:
I am not talking about having 2 motors on the spindle but rather a single servo motor. Using the motor in 2 modes. The first mode as a servo for syncronized motion. At all other times as just a motor, but using the servo driver to still control it. Grbl would just need to know to drive it as a servo syncronized to the other axes, or to just send it a constant pulse stream appropriate for the rpm when it is jus being used as a motor.

chamnit, 2015-03-14:
@109JB : Ah ok, you want to have a separate closed-loop MCU for the motor spindle with the input being Grbl's step pulses. That is doable. If I recall, this is the approach most professional CNC controllers use on their larger machines. Although more expensive, this keeps the main CNC controller independent of the various servos, where each servo controller is tuned specifically for the motor it is responsible for. 

Closed-loop control can be really tricky, because you have to numerically tune the feedback loop for both performance and avoiding numerical harmonics. It's always a balancing act. Asking Grbl to handle this is too much on an AVR, but I'm certain a separate Arduino Nano can be programmed to do this.

Long story short, yes you can do this. And yes it's probably the way to go with servo/spindle sync support for precision things like rigid tapping. As for normal threading with a threading tool, a spindle encoder should be enough for this. All you need to do is start the motion on time.

109JB, 2015-03-14:
The closed loop control would be totally handled by the Gecko Servo driver. This is a commercially available driver that closes the servo loop within the driver. Grbl would see the spindle servo as just another step driven motor like the other axes

tklus, 2016-07-12:
has anything been done with this? I am getting a small CNC lathe that requires a windows 98 PC. I want to get away from the old PC and run it with GRBL. I know that the current lathe has an encoder on the spindle as it currently can thread with the old windows 98 PC.

Thanks!
Tim

109JB, 2016-07-13:
Nope.  I have been busy with other things and haven't had a chance to even look at this.  Too many projects. Too little time.

HuubBuis, 2018-04-02:
I have put a Nema23 (2 Nm) stepper on the spindle. Using a lever, i can switch between stepper or original spindle motor. This works fine for threading, but when threading larger threads, due to the "small" torque of the stepper, the cuts needs to be small.
This week i did a M33 thread on 42CrMo4 steel for making a lathe collet holder. I used "constant volume cut", started with 0.18 mm and ended with 0.015 mm steps. It could cut probably a bit deeper, but i did not want to risc ruin the tread because i had no material to try again.
If i would do it again, i would go for a larger stepper (4Nm) but as long as the treading is cnc controlled, who cares to wait a bit longer.

markotime, 2021-04-13:
Continuous spindle operation: For about 5 dollars, one can get a module generating pulses at a programmable rate with programmable duty cycle.  A DPDT switch would select either grbl or module control, using step and direction leads.  The direction would normally be hard wired for "forward" motion. Assuming your stepper is robust enough to provide enough torque at 
the desired speeds (remember that torque drops off rapidly with speed!), no need for a second motor.  Also, 24 or 36volts on the stepper driver will provide MUCH more rapid buildup of current in the stepper coils, and allow higher torque at speed.
As an aside, I ran into a problem with the Z (longitudinal/carriage) axis pausing and re-energizing during my first tests.  For now, I think it's a thermal cutout issue, so today's mission is to tweak the current downward a bit, and install a fan.  Also, I was running
all 3 microstepping jumpers where full-stepping will provide better than .1 mm resolution, but will likely use 1/2 step for smooth
ness.  Also, at 1/32 microstepping, the instantaneous current TOTAL at any time can be almost 1.5 times that for non-microstepping.

markotime, 2021-04-17:
More spindle control thoughts: XY-LPWM
This is a 5 dollar module with programmable (pushbuttons) rate and duty cycle suitable as a variable speed
signal to control a stepper motor driver. It also has tx/rx serial capability, with a 4-character ASCII word to set each of the (rate,duty-cycle) parameters.
My thought is to feed GRBL Gcode for X and Y (carriage and cross-slide) axes while using the pulse generator
to drive the lathe spindle. (Something like a 6600 cnc driver taking step signals from either the generator or
from a GRBL/CNC shield, with a simple switch to select which.)
For threading, the Z axis (spindle) is by nature synchronized to the X axis for each longitudinal pass. For non-
threading cuts, the independent spindle is preferable.
I would like to add a GRBL function, perhaps using an M-code, to ENABLE the spindle stepper driver,
and to send serial signals to the pulse gen. to specify RPM. For threading, I have my spindle set for 1 turn=1mm, so speed
is proportional to mm/min. Obviously, some tricks are needed to reflect acceleration times, but the basic
Gcode pass can be easily programmed.
I'm NOT capable of creating the code to add the serial command outputs to GRBL, so would appreciate some
help or advice on programming or pointing out problem areas. thanks so much!

HuubBuis, 2021-04-18:
You can use the coolant or mist pin to control (enable) the spindle stepper.

If it is just for threading, why don't you use [grbl-L-mega](https://github.com/MetalWorkerTools/grbl-L-Mega) or [grblHAL](https://github.com/terjeio/grblHAL)

markotime, 2021-04-18:
thank you, I'll give them a try /mark