grbl/grbl #625: Just 1 endstop pin

Snebajer1, 2015-03-14:
On my machine, the three limit switches are wired together, so i can't tell, which endpoint is triggered.
Q1: Will this affect homing abilities ?
Q2: Di i just edit cpu_map.h defining all three endstops to the same pin, which would the be common to all three axis ?

Thanks !

chamnit, 2015-03-14:
@Snebajer1 : Grbl was written to allow using just one limit pin on multiple axes, but I haven't tested it in a while. Theoretically, all you need to do is updated the cpu_map.h, as you already have, and then update the homing cycles in config.h to only home one axis at a time.

Snebajer1, 2015-03-14:
Thanks !
Really quick response.

Snebajer1, 2015-03-16:
Still works beatifully !

closed 2015-03-16 by Snebajer1