grbl/grbl #627: no text file ?

waricap, 2015-03-19:
I followed the instructions
https://github.com/grbl/grbl/wiki/Compiling-Grbl
"4. Open the GrblUpload Arduino example.
Click the File down-down menu, navigate to Examples->Grbl, and select GrblUpload."

There is no code in the file. 
https://github.com/grbl/grbl/blob/master/grbl/examples/grblUpload/grblUpload.ino
Where is the correct file (with code)?
Thanks!

chamnit, 2015-03-19:
There should only be one include. No code.

closed 2015-03-19 by chamnit

waricap, 2015-03-19:
Sketch  grblUpload.ino shows error during compilation because there is no code. 
What should I do next?

waricap, 2015-03-19:
instruction
5. Compile and upload Grbl to your Arduino.
Connect your Arduino Uno to your computer......

chamnit, 2015-03-19:
@waricap : Make sure that you imported the grbl library correctly. You may have this problem if you select the entire repo folder (has a build and doc folder in it), rather than the 'Grbl' folder that contains only the source code.

waricap, 2015-03-19:
https://github.com/grbl/grbl/blob/master/grbl/examples/grblUpload/grblUpload.ino

link this file ?