grbl/grbl #628: config.h support for pin for status light?

DasWookie, 2015-03-20:
This may be possible to do somewhere that I'm just not seeing, but I've done a cursory look to no avail... how can one hook up a status light to grbl for a green/yellow/red light tower?

I was expecting to be able to set a pin which could then have voltages defined (or multiple pins perhaps, not that we have many spare!) to then set various status voltages which the light tower could then use to display the machines current state... be it green - running, yellow - cycle completed or awaiting interaction (tool change, hold, door, idle, whatnot), and red (alarm, limit, big-badda-boom).

cri-s, 2015-03-20:
yes, i have it, i search it out.

cri-s, 2015-03-20:
Code is this, it need to be inserted after
void protocol_execute_runtime()
{
  uint8_t rt_exec = sys.execute; // Copy to avoid calling volatile
multiple times

This is the code:

#ifdef signal_tower
  static uint8_t state=-1;
  if(state!=sys.state) {
    if(state==-1) { // init
          signal_tower(init);
    }	
    state=sys.state;

    if	(state==STATE_ALARM) signal_tower(red);
    else if(state==STATE_CYCLE) signal_tower(green);
    else signal_tower(yellow);
  }
#endif

And the define for signal_tower to be placed on cpu_map.h,
i use lm324 to decode the pin outputs, but if you use pin, just place
DDR to init and
output pin high/low for the individual color.

#define SIGNAL_TOWER_BIT 3
#define SIGNAL_TOWER_PORT PORTB
#define SIGNAL_TOWER_DDR DDRB
#define signal_tower(x)	signal_tower_##x()
#define signal_tower_green()	(\
	SIGNAL_TOWER_DDR  |= (1<<SIGNAL_TOWER_BIT),	\
	SIGNAL_TOWER_PORT &=~(1<<SIGNAL_TOWER_BIT)	\
	)
#define signal_tower_red()	(\
	SIGNAL_TOWER_DDR  |= (1<<SIGNAL_TOWER_BIT),	\
	SIGNAL_TOWER_PORT |= (1<<SIGNAL_TOWER_BIT)	\
	)
#define signal_tower_yellow()	(\
	SIGNAL_TOWER_DDR  &=~(1<<SIGNAL_TOWER_BIT),	\
	SIGNAL_TOWER_PORT &=~(1<<SIGNAL_TOWER_BIT)	\
	)
#define signal_tower_blue()	(\
	SIGNAL_TOWER_DDR  &=~(1<<SIGNAL_TOWER_BIT),	\
	SIGNAL_TOWER_PORT |= (1<<SIGNAL_TOWER_BIT)	\
	)
#define signal_tower_init() signal_tower_red()

2015-03-20 11:17 GMT+01:00, Cri S <phone.cri@gmail.com>:

chamnit, 2015-03-20:
@DasWookie : There is no feature like in Grbl. It's intentionally minimalistic to avoid software bloat, so it can concentrate on whats important. It's fairly simple to add if you want. As @cri-s stated, you can set the LEDs in the protocol_realtime_execute(), where most of the state machine changes occur.

DasWookie, 2015-03-20:
I get software bloat concerns! There are so many users, each with our own particular desires. It just seemed like a status light was one of those things that folks might desire tho considering so many professional CNCs have them and many changes in Grbl seem to have been made to try and have parity there... an example being the door feature.

My other idea besides an op-amp was to use a shift register to control lights, tho that would require the use of 3 rather than a single pin out to control multiple lights/status message. My expectation was to see something akin to support for it in cpu_map.h and/or config.h, and corresponding code in report.c for a tower... or possibly in some sort of signal_tower.h and signal_tower.c set of files.

I'm not adverse to coding up my own implementation and patching my branch for subsequent releases of Grbl... my hope was that it could be supported in the mainline is all.

chamnit, 2015-03-23:
@DasWookie : The safety door feature is something that OEMs need for shipping machines to the EU, so it's a critical thing. LEDs on the other hand is not a critical feature. True, it's fairly inert and small to implement, but the standard build of Grbl on an Arduino Uno does not have the pins to support this. I'll add it to the list of things to do, but there is no guarantee that it'll make it in.

closed 2015-03-23 by chamnit

RabbitMountain, 2021-08-03:
I know this is from 2015, but I am also looking for this functionality. In my case I have designed a new screw-based shield for the arduino uno and would very much like the functionality of a working alarm led indicator. Considering many commercial CNC machines have such a status light I am also slightly surprised this has not at least been added as an option.

On my new board I have wired up an LED to Analog pin 4 since it seems to be listed as unused in the documentation and can be configured as an output.

I can understand that this may not be a critical feature and that the uno has limited program space on the 328p, however I would not consider this bloat. However with such limited space I guess anything could be considered bloat at this point.

Still I hope this will be added in the upstream development and could be enabled with a quick comment / uncomment.

The above code suggestions may provide enough that I can hack something together that will work, but I am much more familiar with aruino c code within the IDE. I am not even sure what the value of A4 will need to be for the cpu_map.h at this point. If anyone has any more code suggestions please let me know. In my case I only need one amber LED to light when Alarms are triggered instead of a whole led tower with 3 lights.

Here is my shield for anyone interested:
https://github.com/RabbitMountain/GRBL-CNDY-SHIELD

RabbitMountain, 2021-08-05:
@cri-s , based on your code example above I seem to have partially working LED code for Analog 4. However the LED does not turn off when there is no alarm state active. Can you help? I'm a bit out of my element with this port manipulation code. I'm not much of a programmer. Thanks for any help you can provide.  @chamnit, any help you could provide would also be welcome.

Here is my partially working code:

void protocol_exec_rt_system()
{

////////////////////////
// ALARM LED CODE: 
#ifdef signal_light
static uint8_t state=-1;
if(state!=sys.state) {
if(state==-1) { // init
signal_light(init);
}
state=sys.state;

if  (state==STATE_ALARM) signal_light(on);
else if(state==STATE_CYCLE) signal_light(off);
else signal_light(off);
}
#endif
////////////////////////

  uint8_t rt_exec; // Temp variable to avoid calling volatile multiple times.
  rt_exec = sys_rt_exec_alarm; // Copy volatile sys_rt_exec_alarm.
  if (rt_exec) { // Enter only if any bit flag is true

and:

    // Define flood and mist coolant enable output pins.
    #define COOLANT_FLOOD_DDR   DDRC
    #define COOLANT_FLOOD_PORT  PORTC
    #define COOLANT_FLOOD_BIT   3  // Uno Analog Pin 3
    //#define COOLANT_MIST_DDR   DDRC
    //#define COOLANT_MIST_PORT  PORTC
    //#define COOLANT_MIST_BIT   4  // Uno Analog Pin 4

    //////////////////    
    // Define ALARM LED OUTPUT
    #define SIGNAL_LIGHT_DDR   DDRC
    #define SIGNAL_LIGHT_PORT  PORTC
    #define SIGNAL_LIGHT_BIT   4  // Uno Analog Pin 4
    
   #define signal_light(on) (SIGNAL_LIGHT_DDR |= (1<<SIGNAL_LIGHT_BIT),SIGNAL_LIGHT_PORT |= (1<<SIGNAL_LIGHT_BIT))
#define signal_light_init() signal_light(off) (SIGNAL_LIGHT_DDR |= (1<<SIGNAL_LIGHT_BIT),SIGNAL_LIGHT_PORT |= (0<<SIGNAL_LIGHT_BIT))
    //////////////////

RabbitMountain, 2021-08-05:
Tinkering with this more, I seem to have come to the conclusion that there is bug in this code either in the portion for the protocol.c file or the cpu_map.h file. I can get an LED to turn on A4 with this code when the off state is not defined in cpu_map.h, but when i go to define the off state the code seems to default to whatever state the LED has been programmed last (which in this case is off).

This results in the LED failing to change when states change. I can either get the LED to constantly light whether we are in an alarm state or constantly off regardless of state, but not to change as intended. I feel like I may inadvertently be programming the LED to stay on forever and not be able to change. Perhaps something akin to using an = instead of an ==. But as stated earlier I am not familiar with this form of C programming, only with simplified arduino coding through the abstraction of the IDE and various libraries.

Any help would be appreciated.

void protocol_exec_rt_system()
{

  uint8_t rt_exec; // Temp variable to avoid calling volatile multiple times.
  
    ////////////////////////
// ALARM LED CODE: 
#ifdef signal_light
static uint8_t state=-1;
if(state!=sys.state) {
signal_light(init);
if(state==-1) { // init
}
state=sys.state;

if (state==STATE_ALARM) signal_light(on);
else if (state==STATE_CYCLE) signal_light(off);
else signal_light(off);
}
#endif
////////////////////////
  
  rt_exec = sys_rt_exec_alarm; // Copy volatile sys_rt_exec_alarm. 

and

 // Define flood and mist coolant enable output pins.
    #define COOLANT_FLOOD_DDR   DDRC
    #define COOLANT_FLOOD_PORT  PORTC
    #define COOLANT_FLOOD_BIT   3  // Uno Analog Pin 3
    //#define COOLANT_MIST_DDR   DDRC
    //#define COOLANT_MIST_PORT  PORTC
    //#define COOLANT_MIST_BIT   4  // Uno Analog Pin 4

    //////////////////    
    // Define ALARM LED OUTPUT
    #define SIGNAL_LIGHT_DDR   DDRC
    #define SIGNAL_LIGHT_PORT  PORTC
    #define SIGNAL_LIGHT_BIT   4  // Uno Analog Pin 4
     
    #define signal_light(x) signal_light_##x()
//    #define signal_light(on) (SIGNAL_LIGHT_DDR |= (1<<SIGNAL_LIGHT_BIT), SIGNAL_LIGHT_PORT &= (1<<SIGNAL_LIGHT_BIT), SIGNAL_LIGHT_PORT |= (1<<SIGNAL_LIGHT_BIT) )
//    #define signal_light_init() signal_light(off)
    #define signal_light_init() signal_light(off)
    #define signal_light(on) (SIGNAL_LIGHT_DDR |= (1<<SIGNAL_LIGHT_BIT), SIGNAL_LIGHT_PORT |= (1<<SIGNAL_LIGHT_BIT) )
//    #define signal_light(off)
    //////////////////

RabbitMountain, 2021-08-08:
After much tinkering with the code i seem to have solved my own issue. In the end I had to get rid of the macro that was programmed in the example code. It also didn't hurt to learn a tiny bit about bit setting in C / C# programming. I'm still not a great programmer by any means, but I'm slightly better now. In the end the KISS method helped to simplify the code to where I both understood what the code was doing within the state machine AND worked as expected.

So... go me! Yay! :)

here is the working simplified code:

for cpu_map.h file:

// Define flood and mist coolant enable output pins.
#define COOLANT_FLOOD_DDR DDRC
#define COOLANT_FLOOD_PORT PORTC
#define COOLANT_FLOOD_BIT 3 // Uno Analog Pin 3
//#define COOLANT_MIST_DDR DDRC
//#define COOLANT_MIST_PORT PORTC
//#define COOLANT_MIST_BIT 4 // Uno Analog Pin 4

//////////////////
// Define ALARM LED OUTPUT
#define SIGNAL_LIGHT_DDR DDRC
#define SIGNAL_LIGHT_PORT PORTC
#define SIGNAL_LIGHT_BIT 4 // Uno Analog Pin 4

#define signal_light_init signal_light_off
#define signal_light_on (SIGNAL_LIGHT_DDR |= SIGNAL_LIGHT_PORT |= (1<<SIGNAL_LIGHT_BIT) )
#define signal_light_off (SIGNAL_LIGHT_DDR |= SIGNAL_LIGHT_PORT &= ~(1<<SIGNAL_LIGHT_BIT) )
//////////////////

and for protocol.c file:

// Executes run-time commands, when required. This function primarily operates as Grbl's state
// machine and controls the various real-time features Grbl has to offer.
// NOTE: Do not alter this unless you know exactly what you are doing!
void protocol_exec_rt_system()
{

uint8_t rt_exec; // Temp variable to avoid calling volatile multiple times.
rt_exec = sys_rt_exec_alarm; // Copy volatile sys_rt_exec_alarm.

///////////////////////
// Define ALARM LED OUTPUT
signal_light_init; //init LED in off state
if (sys.state==STATE_ALARM) {signal_light_on;}
else if (sys.state!=STATE_ALARM) {signal_light_off;}
// else {signal_light_off;}
///////////////////////