grbl/grbl #629: Homing Problem

wheniwas19, 2015-03-20:
I've built a homebrew 3-axis plotter out of foam-core and hardware I had laying around. GRBL. Holy shit. SO happy this exists.

Overcome a lot of stupid "why did I do that?" errors and finally have a successful homing sequence. 

I hate to call this an <i>issue</i> because this is not a fault with GRBL, but my understanding of functionality, maybe. My pull-off moves directly into the negative. My understanding of the homing sequence was to use the limit switches to find 0,0,0, then pull-off by the amount in the settings. In my case, that's 1 mm. Once each axis locates 0, it then moves <i>further</i> by the pull-off amount leaving me in the negative. While the homing reports success, I am now over-limit and any attempt to use gcode, is "thwarted" by being on my switches.

I could use some advice. Not sure how to make this NOT happen. 
My settings:
    $0=10 (step pulse, usec)
    $1=25 (step idle delay, msec)
    $2=0 (step port invert mask:00000000)
    $3=6 (dir port invert mask:00000110)
    $4=0 (step enable invert, bool)
    $5=0 (limit pins invert, bool)
    $6=0 (probe pin invert, bool)
    $10=3 (status report mask:00000011)
    $11=0.020 (junction deviation, mm)
    $12=0.002 (arc tolerance, mm)
    $13=0 (report inches, bool)
    $14=1 (auto start, bool)
    $20=0 (soft limits, bool)
    $21=1 (hard limits, bool)
    $22=1 (homing cycle, bool)
    $23=0 (homing dir invert mask:00000000)
    $24=25.000 (homing feed, mm/min)
    $25=500.000 (homing seek, mm/min)
    $26=250 (homing debounce, msec)
    $27=1.000 (homing pull-off, mm)
    $100=80.000 (x, step/mm)
    $101=80.000 (y, step/mm)
    $102=98.522 (z, step/mm)
    $110=500.000 (x max rate, mm/min)
    $111=500.000 (y max rate, mm/min)
    $112=500.000 (z max rate, mm/min)
    $120=100.000 (x accel, mm/sec^2)
    $121=100.000 (y accel, mm/sec^2)
    $122=100.000 (z accel, mm/sec^2)
    $130=300.000 (x max travel, mm)
    $131=360.000 (y max travel, mm)
    $132=140.000 (z max travel, mm)

Once homing is run, you can see positioning.
    <Idle,MPos:-1.000,-1.000,-1.005,WPos:-1.000,-1.000,-1.005>

Any advice?

Just for sh*ts and gigles, here's my build. I wasn't kidding. Foam core.

[image]

chamnit, 2015-03-20:
@wheniwas19 : This is typically what happens when the limit switches are inverted. Meaning, Grbl assumes normally-open switches and you've wired normally closed. Set `$5=1` to invert them in software. Otherwise, this can happen if your cartesian coordinate system is not setup correctly.

wheniwas19, 2015-03-20:
@chamnit hm, they're set up normally-open. the blue LEDs are connected to the switches and turn off when homing occurs.

I'm not sure what you mean by my coordinate system is not set up correctly.

chamnit, 2015-03-20:
@wheniwas19 : Here's a [link](http://www.linuxcnc.org/docs/2.6/html/common/User_Concepts.html#_machine_configurations) to LinuxCNC's webpage on coordinate systems. You may have to alter the direction invert mask to make sure a positive motion moves in the correct direction.

With the current Grbl v0.9i released earlier this week, you can enable limit state reporting in config.h and recompile/uplaod Grbl. This will cause Grbl to send back the limit states in the '?' status reports. This should be useful in helping you debug your limit switch, making sure Grbl is reading the pins as you think they should. There is more detail in the config.h file on how it works.

wheniwas19, 2015-03-20:
@chamnit thanks, sonny. i'll check that link out. i'm obviously slightly out of my depth at the moment.

wheniwas19, 2015-03-20:
@chamnit sorry. i just noticed in my original post that my positioning statement didn't paste. or i neglected to paste it.

    <Idle,MPos:-1.000,-1.000,-1.005,WPos:-1.000,-1.000,-1.005>

this is what i get when hitting <b>?</b> so once the 1 mm pull-off occurs it pulls closer to the limit switch. should this be the result?

chamnit, 2015-03-20:
@wheniwas19 : This is correct, in terms of where Grbl thinks it is. Machine coordinates should be in all negative space, where zero is home. You also may have your homing switch located in the wrong side of the machine. Grbl searches in the positive direction by default. You can alter this with the homing dir invert setting.

chamnit, 2015-03-20:
@wheniwas19 : First make sure you axes are moving correctly. Type `G91 G0 X1`, such that the X moves 1mm in the positive X direction. On your machine that means the toolhead moves to the right. If it moves left, invert the direction port setting. With `G91 G0 Y1`, your Y gantry should move away from you. 

Once that is set, the limit switches should be located on the right side of the X-axis and the back side of the Y-axis. If not, invert the homing direction setting for the ones that are setup this way.

wheniwas19, 2015-03-20:
@chamnit interesting. the directions are correct. limit switches and homing switches are not the same? i'd run my limit switches in parallel to control both limit on all axes.

[image]
[image]

chamnit, 2015-03-20:
@wheniwas19 : According to your picture, X+ should move "Right", Y+ should move "Top", and [0,0] should be located in the "Top-Right" corner. Homing should engage the limit switches at the "Top-Right" corner.

wheniwas19, 2015-03-20:
imagine my "machine" is turned 90° clockwise, which it is. the picture perhaps confuses the issue. i'd turned it to better fit in my workspace. Top & Right are just that. the X carriage is at 0,0 as it is moved along the Y by the motor at the bottom.

if i jog X by 1 the X carriage moves to the right, which makes sense.

Unless you're saying all of gcode happens in the negative space. i just don't get it. limits and home are two separate things, no?

chamnit, 2015-03-20:
Yes, all gcode works in negative "machine" space. This is traditional. You can set offsets to work in positive "work" space by setting the work coordinate systems with the G10L2 or G10L20 command. Please read up on work coordinate systems. LinuxCNC has good documents online about these commands.

wheniwas19, 2015-03-21:
I'll have a read of LinuxCNC and see what I make of it. I swapped the wires on the steppers for X and Y and they now home to top right. Z still homes to top. The pull-off still moves into the switches though. Need to figure that out. The pull-off should move away from the switches, no?

LKG-Studio, 2015-03-21:
Hi,

I think i've exactly the same problem as you.
Here is a drawing of my machine's workspace : 
[image]

and the parameters of my CNC (i'm using GRBL web) : 
SEND: $$
RESP: $0=10 (step pulse, usec)
RESP: $1=25 (step idle delay, msec)
RESP: $2=0 (step port invert mask:00000000)
RESP: $3=4 (dir port invert mask:00000100)
RESP: $4=0 (step enable invert, bool)
RESP: $5=0 (limit pins invert, bool)
RESP: $6=0 (probe pin invert, bool)
RESP: $10=15 (status report mask:00001111)
RESP: $11=0.020 (junction deviation, mm)
RESP: $12=0.002 (arc tolerance, mm)
RESP: $13=0 (report inches, bool)
RESP: $20=0 (soft limits, bool)
RESP: $21=1 (hard limits, bool)
RESP: $22=1 (homing cycle, bool)
RESP: $23=3 (homing dir invert mask:00000011)
RESP: $24=25.000 (homing feed, mm/min)
RESP: $25=50.000 (homing seek, mm/min)
RESP: $26=25 (homing debounce, msec)
RESP: $27=1.000 (homing pull-off, mm)
RESP: $100=1280.000 (x, step/mm)
RESP: $101=1280.000 (y, step/mm)
RESP: $102=1280.000 (z, step/mm)
RESP: $110=720.000 (x max rate, mm/min)
RESP: $111=720.000 (y max rate, mm/min)
RESP: $112=720.000 (z max rate, mm/min)
RESP: $120=20.000 (x accel, mm/sec^2)
RESP: $121=20.000 (y accel, mm/sec^2)
RESP: $122=20.000 (z accel, mm/sec^2)
RESP: $130=326.000 (x max travel, mm)
RESP: $131=301.000 (y max travel, mm)
RESP: $132=43.000 (z max travel, mm)

I think i've well configured the invert masks for homing and directions i.e. everything is moving in the good direction, but when the homing is done, the positions X and Y are negative :
[image]

After homing, the position should be (1,1,-1) and it should be able to go to (326, 301, -43)..
if I get it right..

I think i have inverted some parameters or i've missed something.
Can you help me ? 

Thanks

chamnit, 2015-03-21:
Grbl places the entire machine volume in negative space after homing, regardless of how you invert the homing directions. This is traditional on CNC machines. It's a common misunderstanding due to 3d printers doing it different way. You can disable this in the config.h to zero all coordinates wherever you home to. The setting is called `HOMING_FORCE_SET_ORIGIN`. Just uncomment this and recompile and upload.

In your case, @LKG-Studio, the origin is actually located at the upper right corner and where you home to is the [-326,-301,0] point. You should be able to move from -326 to 0, -301 to 0, and -43 to 0. Grbl is working correctly.

@wheniwas19 : Please enable the `REPORT_LIMIT_PIN_STATE` option in config.h by uncommenting the line and recompiling and uploading Grbl. This will cause Grbl to report by via the '?' status reports the state of the limit pins. This will help you verify if your switches are truly wired as normally-open. In the past, there was a user who had everything wired correctly and eventually discovered that his switches were bad.

LKG-Studio, 2015-03-21:
Ok, all is clear now, thank you very much.

wheniwas19, 2015-03-21:
@chamnit i hate to bother you with a wiring issue but here i am having eliminated half my limit switches and changed them over to N/O (or at least they should be.) 

when i click the switch, the LED i have attached to the circuit lights up. this is working as intended, no?

however, on attaching the switch to the pin location, i get nothing in the L readout.
    `<Alarm,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000,Lim:>`

I am making some rudimentary mistake but i am close to solving this. not sure what i am missing. 

[image]

JacobChrist, 2015-03-21:
The connection to the arduino is always to ground, therefor never
changing.  Leaving the circuit unchanged, just move the connection to the
arduino from gnd to the C on the switch.  This way when the switch is open
it has a path to ground, but when closed it's tied directly to +V.  Not the
greatest but it will probably work better than what you have.

Jacob

wheniwas19, 2015-03-21:
Ok. I rewired my limit switches (again). Everything homes correctly. This little Frankenstein's monster will be a thing of beauty but it's fun to see something made of foam core move around.

Odd thing, the Lim: never showed up. Not sure why. The limit switches for sure are doing what they are supposed to as the axes all stop and bounce-off as they are meant to. Anyway. Onward and upward!

closed 2015-03-21 by wheniwas19

wslewis, 2016-12-25:
Hi.. despite having read all the posts here I'm no closer to solving the homing issue. Here are my settings:

[image]

when I do a manual jog the limit switches activate (which tells me they're correctly wired) and yet when I enter $H to home it completely ignores the x and y limit switches even though the z limit switch is working fine. What am I doing wrong? I want 0,0,0 to be on the closest left point of the machine with z at top.