grbl/grbl #630: Update problem

wheniwas19, 2015-03-20:
I updated GRBL to 0.9i and uploaded to my UNO. besides <b>?</b> nothing works.

chamnit, 2015-03-20:
@wheniwas19 : If the state says HOLD or DOOR, you've somehow engaged the feed hold or safety door feature. You have to cycle start to exit, which is set by sending the '~' character. Please read the wiki for details.

closed 2015-03-20 by chamnit

wheniwas19, 2015-03-20:
This is the output on connecting to my Arduino.

    Grbl 0.9i ['$' for help]
    ['$H'|'$X' to unlock]

if i type "?", i get. this i understand it has happened before. and generally means i need to home, correct? 

there is nothing else i can do. there is no state i can access. i cannot open "help"

I assure I have spent hours on the wiki. the only reason i am contacting is i am stuck. i don't mean to frustrate you but what seems obvious to you is just beyond my grasp.

chamnit, 2015-03-20:
Make sure your Z-limit switch is not connected to D11. This has now moved to D12 to access the hardware PWM. This suppose this could cause a weirdness if you had things wired incorrectly.

wheniwas19, 2015-03-20:
yes, i had noticed that change as per the pinout on https://github.com/grbl/grbl/wiki/Connecting-Grbl and updated that pin accordingly.

wheniwas19, 2015-03-20:
@chamnit strange. i updated Arduino prior to uploading the latest GRBL build. Serial Monitor appears to be the culprit in this instance. I connected via CoolTerm and it's working as intended. Apologies for the mix-up. I couldn't upload the latest build with Arduino 1.0.6 and therefore updated to 1.6.1 

[image]

at least now i'm back to working out my homing issues. thanks for your patience.