grbl/grbl #631: request for wipe buffer command

lordmundi, 2015-03-23:
I implemented a jogging interface for GRBL using the Contour ShuttleXpress jog dial.  It works well - I used the method where I sent small motions over and over when folks are rotating the shuttle wheel.  When I release the wheel, I'd like for it to stop immediately, and as you know, sending the feed hold and then the reset doesn't really work well and maintain stuff in memory. 

But, if I just had the "%" wipe command, I think all would be fine.  I know folks have talked about a full fledged jogging mode in grbl, which seems like it would take significant resources on a limited chip, but if I just had the ability to wipe any commands in the buffer, I really think this would be a great jog interface without requiring a jog mode in GRBL.  It looks like someone in Issue #178 managed to implement their own "%" wipe command.  

Any chance this could get worked in?  What if I bought you pizza?... with bacon on it?

timryder, 2015-03-23:
LoL awesome!

chamnit, 2015-03-23:
@lordmundi : '%' is a reserved character in the g-code protocol, but that isn't to say that another character could be used, like a control character. That said, something like this is buried in my list of things to do. There are ways around this problem, like feeding Grbl short line segments and only buffering a fraction of a second of motions. 

Also, when jogging, I would argue that using a reset isn't a bad thing. In a traditional workflow, you generally aren't doing anything that deletes anything important when jogging and setting up your job. Typically, you jog to find part zero, set the work coordinates (not the evil G92 that everyone still uses), reset to ensure all of the modes are back to defaults, and start the job. 

Anyway, I'll keep this in mind with the v1.0 release, but I've got bigger fish to fry at the moment.

lordmundi, 2015-03-23:
aren't there coordinate systems and things you might lose?  i'm not that
familiar with it.

i only suggested "%" because I think that is what the tinyG uses to wipe.
I think I'm able to cope with all of the buffering it is doing normally if
I just had some way to know once I come out of a feed hold it isn't going
to do more queued up actions.

FG

chamnit, 2015-03-23:
@lordmundi : No. All g-code parameters, such as work coordinate offsets, are retained permanently in the EEPROM. Only G92 offsets are lost upon reset per the g-code standard. This is why you should always use work coordinates, rather than G92. In professional environments, G92 is typically avoided like the plague and is often the source of crashes by newbies.

lordmundi, 2015-03-23:
<hangs his head in shame>.... but... I.... I use G92.

gerritv, 2015-03-23:
@lordmundi Try out GrblPanel, it was built to highlight and use Grbl's features, such as work offsets. The wiki there has some explanations of how to use the offsets.
(I am trying to get it working under Mono as well for the Linux crowd)
Gerrit

chamnit, 2015-03-23:
@lordmundi : LOL! No worries. G92 existed in Grbl well before I started developing for it, and it's been proliferated by the community by various forks, parallel CNC/3d printer firmwares, etc. It's been four years since work coordinate systems has been introduced to Grbl, and people still don't use it correctly. I've been pushing hard to fix it, but it's incredibly difficult to change how people do things once they learn one way.

gerritv, 2015-03-23:
The problem is that GUI's don't make them visible or accessible :-)

cri-s, 2015-03-23:
why not use $X for that ?

bensuffolk, 2015-06-12:
@chamnit Can you point me to any good articles on workflow using work coordinate offsets instead of G92. I ask as I am just starting out in the hobby CNC stuff, and if I'm going to learn something it might as well be the right way. The problem is all the tutorials and what not out there like you say points to G92

Re the reset as part of jogging being a good thing. I can see that, but I guess a reset at the end is whats good, if you have to keep having the reset on every jog it wastes time, and stops everything being quite so smooth from a user experience. So I'd be up for seeing a wipe buffer command as well.

chamnit, 2015-06-12:
@bensuffolk : I"m going to close this issue. There is likely going to be no wipe buffer command, as this can lead to misuse due to lack of understanding the implications of doing so. The only reason people have requested this is in regard to continuous jogging, which can be solved in other ways.

closed 2015-06-12 by chamnit

bensuffolk, 2015-06-12:
@chamnit No worries, I can implement it another way in my interface. 

Just for anybody searching this later, I have found a good write up on work co-ordinate systems here that might be useful, http://www.shapeoko.com/wiki/index.php/G-Code#Using_the_Work_Coordinate_Systems

lordmundi, 2015-06-12:
I'd like to object, if anyone is interested in hearing it. I would argue
that clearing the buffer is a very basic feature.  Trying to anticipate
that everyone's need for it is related to jogging might be in error.  I
would argue that a well defined interface for developers should include the
ability to wipe the buffer.  Whether folks use it for jogging, or something
else, is really up to them.  But if I'm making an interface to a buffer,
loading it up, statusing it, and clearing it are fundamental operations in
my opinion.

chamnit, 2015-06-12:
@lordmundi : I know where you are coming from. Wiping the buffer is easy to add. I thought about installing it long ago. But when you do wipe the buffer, you have to know where in the buffer you are wiping from, what the gcode state is from the last command sent is the problem, and the implications the wipe has to the state.  You have to know everything about the inner workings to have robust control of the state. To me, that isn't a very good tool. 

If you can come up with a use case that does not involve jogging (solve able via a proper jogging interface) and that can't be resolved by the straight gcode standard or some other robust method, I'll think about reopening this.