grbl/grbl #633: Question about new spindle PWM

michaelfranzl, 2015-03-24:
Cysign asked in issue https://github.com/grbl/grbl/issues/325 :

> And at what rate might the pwm be adjusted?

I've got the same question. If I had this Gcode:

    G01 X0.00 Y0.00
    X0 Y0 S0
    X0 Y1 S100

chamnit, 2015-03-24:
@michaelfranzl : It's linearly adjusted with configurable bounds via the S parameter. Those bounds are set in the config.h file.

chamnit, 2015-03-24:
@michaelfranzl : Forgot to add. There isn't a laser mode for Grbl right now, but it's very soon to be added. There are a couple of good forks that do support real-time lasers, if you want to use them. By default, Grbl comes to a stop during a spindle speed change, as it should for CNC mills.

michaelfranzl, 2015-03-24:
Thank you for the clarification, and yes I'm on the lookout for realtime laser PWM in GRBL. The problem is that this 'upstream' grbl/grbl has rearranged many files and refactored code, so none of the available realtime laser GRBL forks can be brought up-to-date right now (at least automatically). I'll see if I can dig into other forks to actually understand what's needed.

closed 2015-03-24 by michaelfranzl

Mad-Onion, 2016-01-22:
Sorry for pulling this old issue up again, but @chamnit, is there a kind of timeline for the release / implementation of the laser-mode?

Best regards
Tobias

nmz787, 2016-02-01:
I second what @Mad-Onion said... is this so-called "real-time" spindle-speed variation part of main-branch grbl yet? If not, what is the most recent branch which includes this functionality?

chamnit, 2016-02-01:
@nmz787 : The code has been completed, but is not posted publicly anywhere. It's currently being bench tested by myself, Inventables, and a select few colleagues. When it's closer to an actual beta release, I'll post it. I prefer not to let it all out into the wild now and have to manage the alpha-status bug reports.

splitn2, 2016-02-01:
Can't wait for this!!!

nmz787, 2016-03-02:
So I am now confused... the 0.9 release does change PWM duty cycle with the "S" g-code instruction... so what is this ticket about? How synchronized the duty-cycle change is with XYZ stepper movement?

chamnit, 2016-03-02:
@nmz787 : Grbl assumes a machine is using a spindle, rather than a laser. So it correctly stops motion to alter the spindle speed before resuming. Lasers need to change "spindle speed" almost every motion and on-the-fly, particularly when engraving. Stops will cause burn marks. 

Real-time overrides to the spindle speed uses the same code needed by a laser mode to alter "speed" while running without stopping.

nmz787, 2016-03-02:
@chamnit Ok, thanks, I am targetting a laser cutter so this is great information. I was only using an oscilloscope to probe the PWM pins leading to the laser controller... I was not also checking the motor outputs, or I guess I would have seen this interruption.

Marley1953, 2016-09-28:
HI There Just looking in at this conversation about the GRBL running a laser. I am not an electronics or programmer type but I would like to convert my plotter I built to using a laser to do the marking.  I would like to also know when the laser control is available to use or is there a way to turn the laser on and off right now and at different power settings.  any help would be great.
Marcel

chamnit, 2016-09-28:
@Marley1953 : Grbl v1.1 (skipping v1.0 due to existing use) will be out shortly. It's imminent and already out to major GUI developers. I had to tweak how Grbl communicates to cram more data in the status reports and make things easier on developers in the future. If everything goes to smoothly, it'll likely be released as a public testing beta next week sometime. 

FWIW, I don't imagine the beta testing to be very long. The new real-time overrides in v1.1 were written almost a year ago and have been thoroughly tested already. I had to address the comm issues and come up with a new jogging mode (for low-latency joysticks) before release.

Marley1953, 2016-09-28:
Thank you for the quick reply on this chamnit