grbl/grbl #634: Disabling internal pull-up resistor for limit switches

Jibeji, 2015-03-24:
Hi,
I'd like to control a CNC which already have pull-up resistors for limit switches.
How can I disable Arduino's internal pull-up resistors, without having a normal low operation with $5?

Scott216, 2015-03-24:
Having both internal and external pull-ups won't cause any problems.  You can just leave the internal pull-up enabled.

chamnit, 2015-03-24:
@Jibeji : You can disable them in the config.h file. Uncomment the corresponding option, re-compile, and re-upload Grbl.

closed 2015-03-24 by chamnit

samitray, 2015-04-21:
It appears from the documentation  that the nature of the pin changes if we disable pullup, the default normally high ( low on switch activated) would become normally low (high on switch activated). Switch types (NO/NC) additionally confuse matters.

So in the above case set $5=1 to invert the behavior again because disabling pullups inverts the default behavior to normal - low and inverting $5 will set it back to normal high.