grbl/grbl #635: Limit wiring when using a breakout board

109JB, 2015-03-24:
I finally got my Arduino and have GRBL installed and am getting ready to hook it to my milling machine. I previously had my machine running with LinuxCNC on a PC through parallel port to a breakout board. I would like to keep the breakout board in place and simply connect the Arduino to it through the DB25 connector, allowing me to try GRBL without disturbing the LinuxCNC setup. Motor connections are no problem but I am not sure what I need to do in regard to the limits. The breakout board senses the limit switch and sends a +5V signal to the appropriate DB25 pin back to the PC. If I connect the ground pins of the DB25 to the GND of the Arduino, and then connect the limit pin of the DB25 to the appropriate pin on the Arduino, is this all I need to do? I don't think so since this would have the pin high when on the limit switch. Or, do I need to invert the Arduino pin and include a pull-down resistor?Thanks in advance. I'm not too good when it comes to this type of thing. In case anyone is curious, here is a link to the breakout board I am using.

http://www.ebay.com/itm/Board-Interface-Adapter-For-Stepper-Motor-USB-DB25-Cable-5-Axis-CNC-Breakout-/141267691912?pt=LH_DefaultDomain_0&hash=item20e435e988

gerritv, 2015-03-24:
Looks like an interesting board.
The limit pins will need to be inverted in the options. Is the limit switch output low when inactive? If so then you don't need a pull down.

Your bigger issue is with the motor control signal. It works opposite to what the typical 25pin breakout board needs. Even if you use the other side of the relay, you get a blip of the spindle motor when Grbl starts up. I haven't fixed this yet on my set up, will have to add an inverter at minimum.

Gerrit

109JB, 2015-03-24:
Thank you for the fast response. It is much appreciated.

The limit pin on the breakout board is indeed low (0V) when not on the switch and high (5V) when the limit is triggered. Just to be sure I have it right on the limit issue I should only have to do the following:

1 - hook the GND of the DB25 to the GND of the arduino
2 - hook the appropriate limit pin of the DB25 to the appropriate limit pin on the arduino
3 - Invert the arduino pin in the GRBL settings

Is this correct. Just don't want to fry the arduino that took 4 weeks to arrive.

As for the motor control on the breakout board, I currently don't use it. Right now I am manually controlling the spindle, so this won't be a problem.  I am only going to wire the features that I am using which right now are the 3 stepper motors, and 4 limits. I am currently using the A-axis limit on the breakout board for my tool probe plate.

Thanks Again.

109JB, 2015-03-25:
Got it working. So far nothing fried.

closed 2015-03-25 by 109JB